Pho3niX90/solis_modbus #13: Issue to get it working with my datalogger S3-WIFI-ST.

ccj2740, 2023-12-10:
Thanks for this nice project. I am getting data from Solis Cloud with S3-WIFI-ST datalogger atm.
This data is delayed with 1-3min , so not useful for automation and charging my car.

I really want to get realtime dataflow , so this is why your solis_modbus is very interesting.

I have installed it already , but I get an error when I try connect to the IP ?

Here is some screenshot:
[image]
[image]
[image]
[image]

ccj2740, 2023-12-10:
Here is a picture of the datalogger manual.
Let me know if I can do more :).

[image]

0rangutan, 2023-12-12:
The Solis Data Logger can only accept one connection at a time.  So using just the logger, you can use this integration OR Solis Cloud but not both.

If you want to use both simultaneously you will need to do a little DIY, eg. https://github.com/alienatedsec/solis-ha-modbus-cloud

ccj2740, 2023-12-14:
> The Solis Data Logger can only accept one connection at a time. So using just the logger, you can use this integration OR Solis Cloud but not both.
> 
> If you want to use both simultaneously you will need to do a little DIY, eg. https://github.com/alienatedsec/solis-ha-modbus-cloud

Ahh ok , I need the Solis cloud because I have battery with time schedule for charge/discharge settings.
Guess I have to accept the 1-2min delay from Solis Cloud as I am not geek enough to do the DIY :(...

0rangutan, 2023-12-14:
Actually, with the latest release of this integration you can now configure charge and discharge times too.
So if there's nothing else you need Solis Cloud for, you can set this up as the only connection to your logger and enjoy!

ccj2740, 2023-12-14:
> Actually, with the latest release of this integration you can now configure charge and discharge times too. So if there's nothing else you need Solis Cloud for, you can set this up as the only connection to your logger and enjoy!

Very interesting, but if I take it out of Solis cloud will I loose the possibility for firmware updates from Solis. Also I have some other configuration with SOC levels. However is it easy to just take it out and setup solis_modbus ? (sorry for my newbie questions)

Pho3niX90, 2023-12-14:
> Very interesting, but if I take it out of Solis cloud will I loose the possibility for firmware updates from Solis. Also I have some other configuration with SOC levels. However is it easy to just [...]

Firmware updates are still possible. I think this also depends on the specific support that is assisting in the firmware update. 

Example. I have requested 2 firmware updates in the last month. Once for the inverter, and on e for the datalogger. For the inverter, it was just pushed from their side without issues, for the datalogger, they requested it to be connected to their api.
For the latter, I just disabled the modbus, gave it 5mins to sync, and they issued the update.

ccj2740, 2023-12-15:
@Pho3niX90 : Ahh then it make sense to switch my datalogger to LAN configuration instead. Is it easy to disconnect from Solis cloud and set it up to your solid_modbus ? Is there a guide for this ?

kiloptero, 2023-12-28:
hii! maybe a stupid question.. you can use this integration and when only you need the firmware update disconnect the integration for a few minutes?

Pho3niX90, 2023-12-28:
> hii! maybe a stupid question.. you can use this integration and when only you need the firmware update disconnect the integration for a few minutes?

Yes correct.

kiloptero, 2023-12-28:
cant wait to install! i will share it
thanks a looot!!

kiloptero, 2024-02-16:
another question.. ill have two dataloggers.. its possible to use 1 with solis cloud and other for ha.. just "swaping" when i needed? ill use 99% of the time with ha.. just solis for updates..

Pho3niX90, 2024-02-25:
> another question.. ill have two dataloggers.. its possible to use 1 with solis cloud and other for ha.. just "swaping" when i needed? ill use 99% of the time with ha.. just solis for updates..

How will these data loggers be connected? 

So from the plugins side, it pulls the information from the datalogger, vs the datalogger pushing to it. So if one of your dataloggers has an open and unused modbus port, it should work fine.