Pho3niX90/solis_modbus #20: PV Power anomaly

aavanzyl, 2023-12-15:
Hi

So I have been running the plugin for a while now and with the last update, I noticed a change in the PV energy.

Here on the graph, you can see its a definite change after the update. 
[image]

The total seems to be closer to the actual. 
[image]

Pho3niX90, 2023-12-15:
Will investigate this today. Is the issue only for the individual string values?

Pho3niX90, 2023-12-15:
could you check `Solis PV Voltage 1` and `Solis PV Current 1`, as one of them would be the offending values.

Pho3niX90, 2023-12-15:
Can you confirm if `1.2.1` fixes the issue?

aavanzyl, 2023-12-15:
The Solis PV Voltage 1 & Solis PV Current 1 both match exactly the values directly from the inverter itself. I use the bluetooth connection to connect to the inverter while HA uses the Wifi.

Just looking at immediate values, the inverter indicates 1490,41W and the 14904,1W. It's as if it's just multiplied by the factor of 10. 

Below is a quick screen grab to illustrate the values, they might be taken 10 seconds apart, so the values do fluctuate, but are always very close to each other, so not worry about difference, since the app might only be reading at an different interval. 

Solis Home Assistant             |  Solis Cloud App (Bluetooth)
:-------------------------:|:-------------------------:
[image]  | [image]

aavanzyl, 2023-12-15:
> Can you confirm if `1.2.1` fixes the issue?

Checking...

aavanzyl, 2023-12-15:
Working as expected, thank you :) I owe you a beer.

PR #21 (2023-12-15): Bugfix/issue 20
fixes #20
closed 2023-12-15 by aavanzyl