Pho3niX90/solis_modbus #22: Bugs with sensors for Battery Charge / Discharge

0rangutan, 2023-12-15:
Possible issues with battery related sensors:
1. Solis Battery Force-charge Power Limitation reads as 50W in the integration but 500W in Solis Cloud.  So I think out by factor of 10 somewhere?
2. Solis Battery Power is the sensor for the power currently going into the battery to charge it so perhaps this could be re-named to indicate the charging, eg. Solis Battery Charge Power?
3. I can't find an equivalent sensor for power currently being **discharged** from the battery (eg. Solis Battery Discharge Power) - Have I missed this somewhere?

Pho3niX90, 2023-12-15:
What do you mean by point 3? 
The charge sensor would be `sensor.solis_battery_charge_power` and discharge `sensor.solis_battery_discharge_power`

0rangutan, 2023-12-15:
Apologies for any confusion.
Are you saying that I should use sensor.solis_battery_charge_power and sensor.solis_battery_discharge_power ?
Neither are present in my instance of your integration.  Should they be?

Point 3 was requesting that a Battery Discharge sensor be added.

Is the problem that they are missing from my setup or from your integration?

Pho3niX90, 2023-12-15:
For the other points

1. This is definitely a bug, and will address this. The input `number.` is the correct value.
2. `Battery power` is a signed 32, however, solis just returns this as a positive, direction is determined by `Solis Battery Current Direction`.  if  `Solis Battery Current Direction` == 0, then it's charging, if it's 1, then it's discharging.

Pho3niX90, 2023-12-15:
I was mistaking about `sensor.solis_battery_charge_power` and `sensor.solis_battery_discharge_power`. It seems I had this as a template sensor in my configuration. I will defnitely add these sensors.

0rangutan, 2023-12-15:
Ah, great - that all makes sense now.
Yes please - adding these as sensors in the integration would be very helpful.

Pho3niX90, 2023-12-15:
So the 2 sensors have been added in `1.3.0`. I am still investigating how to properly handle point 1

closed 2023-12-18 by Pho3niX90

0rangutan, 2023-12-18:
All working well now, many thanks

PR #24 (2023-12-18): Bugfix/issue 22
Refactor variable naming and replace "decimal_places" with "multiplier"

Update the code to enhance readability by improving the variable names and replacing "decimal_places" with "multiplier". This modification increases the maintainability of the code and provides a clearer representation of the role of each property in the module.

Fixes #22
commit 462ff4dd (2023-12-18): Refactor variable naming and replace "decimal_places" with "multiplier"

Update the code to enhance readability by improving the variable names and replacing "decimal_places" with "multiplier". This modification increases the maintainability of the code and provides a clearer representation of the role of each property in the module.

Fixes #22