Pho3niX90/solis_modbus #26: its is possible to see the list of entities that this integration show to ha? thanks a lot!!

kiloptero, 2023-12-28:
**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]

**Describe the solution you'd like**
A clear and concise description of what you want to happen.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context or screenshots about the feature request here.

Pho3niX90, 2023-12-28:
The current list can be found here: https://solis-modbus.readthedocs.io/en/latest/sensors.html

kiloptero, 2023-12-29:
Solis S6 Pro 6kW Advanced Hybrid Inverter
its is possible to share the sku model of the inverter?

Pho3niX90, 2023-12-30:
The sku would be
S6-EH1P6K-L-PRO

But the code should work for these as well:
S6-EH1P3K-L-PRO
S6-EH1P3.6K-L-PRO
S6-EH1P5K-L-PRO
S6-EH1P6K-L-PRO
S6-EH1P8K-L-PRO

I will be happy to add support for others as well. The code should theoretically already be compatible with all solis inverters.

kiloptero, 2023-12-30:
thanks for the time to supporting us!

Andres Roepke Del Solar.
Ingeniero Civil Industrial
Ingeniero Electronico
Mobile 88196932

closed 2024-01-03 by Pho3niX90