Pho3niX90/solis_modbus #27: Slowdown of home assistant

neox387, 2023-12-30:
Since installation my home assistant web pages can sometimes take up to 10 seconds to load

It seems to happen when solis_modbus is polling
I've tested increasing the following from 5 to 10 and this indeed cause less frequent hangs
POLL_INTERVAL_SECONDS  in  solis_modbus/const.py 

In the logs I found this a few times:

2023-12-30 23:47:30.180 DEBUG (MainThread) [custom_components.solis_modbus.sensor] register_key = 43172, value = 0
2023-12-30 23:48:22.821 ERROR (MainThread) [homeassistant] Error doing job: Exception in callback _run_async_call_action(<HomeAssistant RUNNING>, <Job track ti...7fb73cd3c9a0>>) at /usr/src/homeassistant/homeassistant/helpers/event.py:1450
Traceback (most recent call last):
  File "/usr/local/lib/python3.11/asyncio/events.py", line 80, in _run
    self._context.run(self._callback, *self._args)
  File "/usr/src/homeassistant/homeassistant/helpers/event.py", line 1454, in _run_async_call_action
    hass.async_run_hass_job(job, time_tracker_utcnow())
  File "/usr/src/homeassistant/homeassistant/core.py", line 671, in async_run_hass_job
    hassjob.target(*args)
  File "/usr/src/homeassistant/homeassistant/helpers/event.py", line 1532, in interval_listener
    hass.async_run_hass_job(job, now)
  File "/usr/src/homeassistant/homeassistant/core.py", line 671, in async_run_hass_job
    hassjob.target(*args)
  File "/config/custom_components/solis_modbus/sensor.py", line 903, in async_update
    values = controller.read_input_register(start_register, count)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/solis_modbus/modbus_controller.py", line 31, in read_input_register
    raise ValueError(f"Failed to read Modbus register ({register}): {result}")
ValueError: Failed to read Modbus register (33132): Modbus Error: [Input/Output] Modbus Error: [Invalid Message] No response received, expected at least 8 bytes (0 received)
2023-12-30 23:48:37.856 DEBUG (MainThread) [custom_components.solis_modbus.modbus_controller] register value, register = 33000, result = [243, 178, 67, 1, 12593, 12358, 13106, 12849, 17200, 12848, 12339, 12336, 0, 0, 0, 0, 0, 0, 0, 0]

Pho3niX90, 2024-01-01:
Could you give me some insights? What hardware are you using? Have you built charts with the data? What does the hardwares utilization look like before and after installing (if available)

Pho3niX90, 2024-01-01:
Another thing, it seems you have debug logging enabled, this repo has a lot of debugging logs, so turning this on will spam the logs, and likely slow down devices that write to SD cards, so my first suggestion would be to disable debug logging, and see if this resolves the issue.

neox387, 2024-01-03:
I had debug loggin enabled from the integration page of modbus to hopefully see more what was wrong its not always on.

Hardware wise I should not have any bottlenecks 19 integrations running no problem everything snappy loading within a second of clicking, enabling this integration and load times can go up to 10 seconds.

cpu utilization have not seen it go over 4%, ram at 7-12%

Fresh Home Assistant OS 11.2 vdi (virtualbox)
CPU 2vcores from a Intel N5105 (tested with 4 cores)
RAM 4GB ddr4-2933
Storage:  32GB on a raid 5 nvme gen 4 ssds

Pho3niX90, 2024-01-03:
Okay it's a peculiar issue. Do you know how to modify the integrations files? Inside the const file you can change the interval of scans, if this works I will prioritize #15

neox387, 2024-01-03:
Ok i did try POLL_INTERVAL_SECONDS = 10 before i made the issue but at the time it did not seem to help

Testing again
10 seconds the value errors are 90% gone did have another ping service with a warning tough
15 after a long time i did get another new one
ValueError: Failed to read Modbus register (43010): Modbus Error: [Input/Output] No Response received from the remote slave/Unable to decode response

Using 30 now because this way its less noticable, can still reproduce every 30 seconds doing a reload or interacting with home assistant it will hang for a few seconds.

Not sure how you get the data from the logger is it not 1 call or is it big that it takes a while, or is it my older datastick donno,

but maybe what you said in #15 about not updating all the field every times make a difference.

Pho3niX90, 2024-01-04:
So there are about 5 calls done, to avoid multiple calls. The menory used should be low since it's all U16 values. 

NO response received will be noticed from time to time, this can be due to a timeout.

Could you do a ping test from your HA hardware to your inverter to see how stable it is? It could be a timeout issue if perhaps larger than 100ms (need to verify)

neox387, 2024-01-04:
Ping test where 2 times the polling happend and notice page loading for a few seconds but the ping test shows nothing strange and keeps pinging trough he hangs.

33 packets transmitted, 33 packets received, 0% packet loss
round-trip min/avg/max = 0.819/1.362/1.841 ms

Tought I eliminated most errors with higher polling rate but and running over 24 hours not but sadly not:

> Error doing job: Exception in callback _run_async_call_action(<HomeAssistant RUNNING>, <Job track ti...7fbf671f9ee0>>) at /usr/src/homeassistant/homeassistant/helpers/event.py:1450
11:28:35 PM – (ERROR) Solis Modbus (custom integration) - message first occurred at January 3, 2024 at 10:33:25 PM and shows up 127 times
Error doing job: Exception in callback _run_async_call_action(<HomeAssistant RUNNING>, <Job track ti...7fbf68fbbf60>>) at /usr/src/homeassistant/homeassistant/helpers/event.py:1450
11:20:55 PM – (ERROR) Solis Modbus (custom integration) - message first occurred at January 3, 2024 at 10:32:19 PM and shows up 171 times>

neox387, 2024-01-04:
I'll test this weekend on my desktop with the VMware image to see if the same thing happens.

neox387, 2024-01-05:
Test on desktop with latest 11.3 vmware image with only this integration and can reproduce slowdown every 15 seconds

If others don't have this I suspect it's because of a different datalogger

Pho3niX90, 2024-01-05:
Are you able to do a short video of the slowdown?

Even if the datalogger is to blame, it's not suppose to slow down your system, so will have a look over the weekend to see what might be causing this.

neox387, 2024-01-06:
refreshing the page every second you see it at 14 and 30, just before the next polling at 44 I click on system info.

https://github.com/Pho3niX90/solis_modbus/assets/6033262/b361a498-7ab9-4726-bd52-ec8fc1dad068

Pho3niX90, 2024-03-09:
@neox387 it's been a while since you opened this issue, and not sure if you are still using the integration. 

If you are, could you please see if version `1.4.0-beta` fixes the issue for you?

cgilis, 2024-03-22:
Hi, 

I've the same issue, my home assistant is slower with this plugin installed.
HA: 2024.3.3

Pho3niX90, 2024-03-23:
> Hi, 
> 
> I've the same issue, my home assistant is slower with this plugin installed.
> HA: 2024.3.3

Which version are you running of the plugin?

cgilis, 2024-03-23:
The latest one, 1.4.1 on HA Yellow.
But I think it's ok now, I removed it and added it again.

Pho3niX90, 2024-03-23:
> The latest one, 1.4.1 on HA Yellow. But I think it's ok now, I removed it and added it again.

Okay please keep me updated if there is slowdown again

cgilis, 2024-03-23:
Sometimes it feels like HA takes longer to load or react but it can just be me of course :-) 

is there a way to disable fetching data I don’t need? So maybe make it faster?

cgilis, 2024-03-24:
@Pho3niX90 it feels like it is the same as in the video of @neox387, so it sometimes takes longer to load a page.
I did not have this before this plugin.

Pho3niX90, 2024-03-24:
> @Pho3niX90 it feels like it is the same as in the video of @neox387, so it sometimes takes longer to load a page.
> I did not have this before this plugin. 

Okay will investigate a solution today.

cgilis, 2024-03-24:
Wow, nice.
Do you need something from me?

cgilis, 2024-03-24:
[image]

cgilis, 2024-03-24:
@Pho3niX90  BTW I have this Solis inverter: Solis RAI 3kW.
The only control functions that are working are those in the screenshot below. It's enough to control the battery.

[image]

cgilis, 2024-03-26:
@Pho3niX90 if you need me to test things, let me know, happy to help!

cgilis, 2024-03-29:
I removed the plugin and HA was fast/normal again.
I've installed the plugin on a separate Raspberry Pi and it gives the same issue.

Pho3niX90, 2024-04-02:
@cgilis sorry for the late reply, was a busy week. 
I created a separate issue here #43 
I am wondering if the missing registers aren't the culprit of the slowdown, if we can continue the discussion there?

cgilis, 2024-04-02:
No no there are no missing registers, everything is visible, but I can only control these in the screenshot.
The other ones have no effect on the Solis

Pho3niX90, 2024-04-03:
fixed in https://github.com/Pho3niX90/solis_modbus/releases/tag/1.4.4

PR #37 (2024-03-12): Refactor async_update to use asyncio
The `async_update` method in the solis_modbus sensor has been refactored to utilize asyncio for entity updates. Instead of using simple loops to call the update method on each entity, asyncio.gather and asyncio.to_thread are used to make the update calls asynchronously. This change should improve the efficiency of the update method, reducing the time taken and resources used.

fixes: #27 
fixes #38 

pre-release: https://github.com/Pho3niX90/solis_modbus/releases/tag/1.4.0-beta-02
closed 2024-04-03 by Pho3niX90