Pho3niX90/solis_modbus #28: Charge / Discharge Time Slots - Error displaying new selection

bangorha, 2024-01-02:
Home Assistant Solis Modbus allows me to change the start/end times for my various charge/discharge slots - for info, I am only using Charging Time Slot 1 and Discharging Time Slot 1, but after refreshing the webpage it reverts to the previously selected times. 

[image]

[image]

[image]

Even though the previous times are displayed, the new times have been successfully written to the Inverter Registers as shown in the Solis Remote Control screenshot, and the system operates as I intended, but for some reason only the original times are displayed after a refresh.

Pho3niX90, 2024-01-02:
So if I am understanding the issue correctly. The change was written successfully, however, the state of the change wasn't persisted in HA?

bangorha, 2024-01-03:
> So if I am understanding the issue correctly. The change was written successfully, however, the state of the change wasn't persisted in HA?

Yes that's correct - the change happens instantaneously, as I am able to observe the relative increase in import / export as soon as I apply the figures.

An immediate check on Solis Cloud shows the correct new times, but when I refresh the web browser, the HA Dashboard times have reverted to what they were previously.

For now I have removed the solis-modbus integration and am trialling the solax integration - the issue is not prevalent when using the Solax integration.

Pho3niX90, 2024-01-03:
Okay perfect, thanks for letting me know. Will add a fix in the next day

Pho3niX90, 2024-01-03:
This is now fixed in `1.3.2`

commit d3a5a35e (2024-01-03): Improve performance of Solis Modbus sensor

Removed unnecessary polling counter and scan interval in the Solis Modbus sensor. Also rearranged the order of operations in the time setting process to further optimize the code. The async write operation for the home automation state is now executed immediately after setting the time value. fixes #28
PR #29 (2024-01-03): Improve performance of Solis Modbus sensor
Removed unnecessary polling counter and scan interval in the Solis Modbus sensor. Also rearranged the order of operations in the time setting process to further optimize the code. The async write operation for the home automation state is now executed immediately after setting the time value. fixes #28
closed 2024-01-03 by Pho3niX90