Pho3niX90/solis_modbus #30: Add NTP time sync

0rangutan, 2024-01-31:
The Solis Inverter clock can drift out of sync leading to problems with time charges etc.
The Solis Cloud UI allows syncing the time on the inverter to match PC time.
Is there any way to add an NTP sync or similar to the integration to automate this regulalry?

Pho3niX90, 2024-02-16:
I suppose this is possible. I will investigate this

Pho3niX90, 2024-03-09:
I'm currently conducting an investigation to determine the frequency of time drift in the system. My goal is to develop a solution that updates the time only when necessary, minimizing unnecessary writes to the inverter. 

My solis is only a couple months old, and my time drift hasn't out of sync as yet. 

How often have you seen this happen, and to what degree? (seconds, minutes)

0rangutan, 2024-03-11:
Checking today, my inverter has drifted forwards by 2m50s since the reset on 31st January.
So ca. 4.25 seconds per day or just over 2 minutes per month.

Pho3niX90, 2024-03-11:
That seems rather extreme, could you perhaps compare it with https://time.is/ so that we use the same baseline?

My drift, in comparison since my last reset is only 2 seconds, and was reset with the last update around 3 weeks ago

0rangutan, 2024-03-11:
That's exactly what I used!
I guess the clock accuracy may be quite variable between inverters.

Pho3niX90, 2024-03-12:
So it will be in the `/config/logs` path

The format will be as follows:
inverter time {hours}:{minutes}:{seconds}. drift = {d_hours}:{d_minutes}:{d_seconds}

Above 60 seconds, should print a critical log, 30 would print a warning, 10 info, and anything below that wouldn't print anything at all.

0rangutan, 2024-03-12:
I can't find any entries about drift - can I check that I am following the right process?
I have updated to 1.4.0b2, enabled Debug Logging on the integration, restarted HA, Disabled Debug Logging, opened the downloaded file.
I have also checked the main home-assistant.log file.
Thanks

Pho3niX90, 2024-03-12:
It might be that there isn't an actual drift above 10 seconds. 

If you want some more in depth logging to see the current drift, you can edit the sensor.py file, on line 763, you can change the 10 to 0 
https://github.com/Pho3niX90/solis_modbus/blob/1.4.0-beta-02/custom_components/solis_modbus/sensor.py#L763 

If you prefer not to edit the python files, you can install this version: https://github.com/Pho3niX90/solis_modbus/releases/tag/1.4.0-beta-03

I have added the debug logs, which wasn't enabled in 02

0rangutan, 2024-03-12:
Thanks - that works.  Up to 3 seconds drift now after a reset yesterday:

2024-03-12 15:47:12.765 DEBUG (SyncWorker_18) [custom_components.solis_modbus.sensor] inverter time 15:47:15. drift = 0:0:-3

Pho3niX90, 2024-03-12:
Okay, this version will do the time syncing: https://github.com/Pho3niX90/solis_modbus/releases/tag/1.4.1-beta-01

Currently, it's set to only sync when the time has drifted more than 5 seconds.

0rangutan, 2024-03-12:
Many thanks - installing now and it will likely drift over 5 seconds tomorrow so will feed back then.

0rangutan, 2024-03-13:
The time was in sync when I checked this morning, thanks!
Is the sync recorded in a log when this happens?

Pho3niX90, 2024-03-13:
> The time was in sync when I checked this morning, thanks! Is the sync recorded in a log when this happens?

Yes it will state that its going to modify it.

0rangutan, 2024-03-13:
Thanks - can you confirm which log and what text?  Would just like to confirm that it is present in mine.

Pho3niX90, 2024-03-13:
So, it will be available in the general logs in the front end. It will print the same as the other logs mentioned previously. However, it will just include that it will attempt a sync.

Pho3niX90, 2024-03-13:
Will look like this. It will be an info log.

inverter time {hours}:{minutes}:{seconds}. drift = {d_hours}:{d_minutes}:{d_seconds}, adjusting

The last part stating "adjusting" is when it syncs.

0rangutan, 2024-03-14:
Thanks - that just kicked in and worked now:
2024-03-14 19:34:11.702 INFO (SyncWorker_26) [custom_components.solis_modbus.sensor] inverter time 19:34:17. drift = 0:0:-6, adjusting

From the debug log, it looks like the drift value is checked every 15 seconds.  I think this could be much less frequent, eg. hourly?  Or do you pick the values up because you poll all the registers every 15 seconds anyway?

Pho3niX90, 2024-03-15:
> Thanks - that just kicked in and worked now:
> 2024-03-14 19:34:11.702 INFO (SyncWorker_26) [custom_components.solis_modbus.sensor] inverter time 19:34:17. drift = 0:0:-6, adjusting
>
> From the debug [...]

Yeah, it just piggy backs of the clock registers, which is fetched along with others in the same range in a single request.

0rangutan, 2024-03-15:
Great thanks - I'd say this feature is done in that case.

PR #40 (2024-03-16): Feature/clock adjustment on drift
fixes: #30
closed 2024-03-16 by Pho3niX90