Pho3niX90/solis_modbus #41: Question: Coming from homeassistant-solax-modbus

wawyed, 2024-03-16:
Is there any benefit for me to use this integration if I'm using the https://github.com/wills106/homeassistant-solax-modbus already?

Just a genuine question to see if it's worth doing so.

0rangutan, 2024-03-16:
Yes, if you have a Solis inverter - you may prefer to run this instead.
This integration is Solis focussed and has a better implementation of charge/discharge times, plus now syncs the time on your inverter if it drifts.
You can run both for a while and see which you prefer.

wawyed, 2024-03-16:
Hmm not really I can run both at the same time? Surely only one can access the modbus at the same time?

0rangutan, 2024-03-16:
It will work fine for read access.  Just one TCP/IP connection is supported by the data logger but as both integrations come from the single HA instance IP it isn't a problem.

kiloptero, 2024-03-16:
its is possible to use two dataloggers? one for solis cloud and other for
solis modbus?

dont trust solis api

Andres Roepke Del Solar.
Ingeniero Civil Industrial
Ingeniero Electronico
Mobile 88196932

0rangutan, 2024-03-16:
Not directly.
Instead, use one data logger and split it using Waveshare devices:
https://github.com/alienatedsec/solis-ha-modbus-cloud
Then one talks to Solos Cloud and the other to HA.
This is my setup following the above:
https://github.com/alienatedsec/solis-ha-modbus-cloud/discussions/24

kiloptero, 2024-03-16:
> alienatedsec/solis-ha-modbus-cloud#24

im not able to create that... but change the datalogger will be easy
i already have 2 s2 and s3 wifi
the idea is to use s2 lan for ha and s3 for solid cloud

Andres Roepke Del Solar.
Ingeniero Civil Industrial
Ingeniero Electronico
Mobile 88196932

Pho3niX90, 2024-03-17:
> Is there any benefit for me to use this integration if I'm using the https://github.com/wills106/homeassistant-solax-modbus already?
> 
> Just a genuine question to see if it's worth doing so.

To shortly answer your question. 

It comes down to what you are after. This was created to have fast data, so that automations aren't delayed, it's specifically focused only on solis inverters. 

I am obviously biased, but wills was the first one I used before writing this integration.

Pho3niX90, 2024-03-17:
> im not able to create that... but change the datalogger will be easy
> i already have 2 s2 and s3 wifi
> the idea is to use s2 lan for ha and s3 for solid cloud
> 
> Andres Roepke Del Solar.
> Ingeniero Civil Industrial
> Ingeniero Electronico
> Mobile 88196932
> 
> 
> 
> > Not directly.
> > Instead, use one data logger and split it using Waveshare devices:
> > https://github.com/alienatedsec/solis-ha-modbus-cloud
> > Then one talks to Solos Cloud and the other to HA.
> > This is my setup following the above:
> > alienatedsec/solis-ha-modbus-cloud#24

Haven't attempted this myself, but since you already have the hardware, you can definitely try it.

wawyed, 2024-03-17:
> To shortly answer your question.
>
> It comes down to what you are after. This was created to have fast data, so that automations aren't delayed, it's specifically focused only on solis inverters.
>
> I [...]

What was missing from the other one that made you create this one?

Pho3niX90, 2024-03-17:
I cant recall, but was mostly performance and granularity of the data related, but you can see the current registers here -> https://solis-modbus.readthedocs.io/