Pho3niX90/solis_modbus #43: @Pho3niX90  BTW I have this Solis inverter: Solis RAI 3kW.

Pho3niX90, 2024-03-24:
@Pho3niX90  BTW I have this Solis inverter: Solis RAI 3kW.
The only control functions that are working are those in the screenshot below. It's enough to control the battery.

[image]

_Originally posted by @cgilis in https://github.com/Pho3niX90/solis_modbus/issues/27#issuecomment-2016818375_

Pho3niX90, 2024-04-02:
Could you perhaps post your visible registers here that you are able to see? @cgillis

The idea, is to create a version compatible with your inverter, and then see if it solves the slowdown. One question I never asked in #27 was what inverter they were using. 

This will aid in finding the slowdown location, and making more compatible with your inverter.

cgilis, 2024-04-02:
[image]

cgilis, 2024-04-02:
[image]

cgilis, 2024-04-02:
The first screenshot are the only ones that I can control. The other actions has no effect on my Solis

Pho3niX90, 2024-04-02:
Aah I understand better now. 

Do you see any errors or warnings in your logs?

cgilis, 2024-04-02:
[home-assistant_2024-04-02T17-22-31.374Z.log](https://github.com/Pho3niX90/solis_modbus/files/14841575/home-assistant_2024-04-02T17-22-31.374Z.log)

Sorry is this enough? :)

Pho3niX90, 2024-04-02:
That does seem to show a general connection issues. 

Are you able to do a ping test from your raspberry pi to the solis? Just to see how stable the connection is? I will have to do some reading into the HA python code to see why this might be causing delays. My assumption is that this is related to IO to the SD card (assuming you are using one)

cgilis, 2024-04-02:
On the Pi:
[image]

On a macbook:
[image]

Pho3niX90, 2024-04-02:
Ok there are some ups and downs, some dramatic and some not so much of an issue. It might be that the more dramatic ones closer to a second might eventually add up, and cause some queueing. 

Is the slowdown (of ha) from the start, or only after a while?

cgilis, 2024-04-02:
It is from the start, when the plugin is loaded.

Pho3niX90, 2024-04-02:
> btw, the connection is good! ![CleanShot 2024-04-02 at 19 48 50@2x](https://private-user-images.githubusercontent.com/62599539/318902750-7ee48399-1bf1-40f5-b414-d57f6113a35e.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTIwODA2NDgsIm5iZiI6MTcxMjA4MDM0OCwicGF0aCI6Ii82MjU5OTUzOS8zMTg5MDI3NTAtN2VlNDgzOTktMWJmMS00MGY1LWI0MTQtZDU3ZjYxMTNhMzVlLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA0MDIlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwNDAyVDE3NTIyOFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWU4ZjdhMTkxMTA3ZDIwMjJmOGYwMmFkYWVlNDk4OGZiNmVmYjBhYjMwZWUwOTVjNGYzNjkzZDk2ZDFjODZhMTkmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.PDomwGLGB2hDPkwSp9vNSi0rlF4iI1xBmEM8QUkq6pY)

Got to love unifi! my entire system us unifi

Pho3niX90, 2024-04-02:
> It is from the start, when the plugin is loaded.

Okay let me do some investigation, something is blocking the mainthread in the code. Will try and diagnose, and startup a raspberry pi

cgilis, 2024-04-02:
Nice! Thx man!

PS I also had it on my Home Assistant Yellow, but yeah that's running on a RP Compute Module :-)
I can if you want, try it on a virtual machine on a Synology NAS?

Pho3niX90, 2024-04-02:
Could you see if https://github.com/Pho3niX90/solis_modbus/tree/feature/speedup made a difference, if any?

cgilis, 2024-04-02:
Sorry for my stupid question, but i can't find it in the list 

[image]

Pho3niX90, 2024-04-02:
Just released 1.4.3

cgilis, 2024-04-02:
It feels like no difference here :-s 

[home-assistant_2024-04-02T19-28-16.689Z.log](https://github.com/Pho3niX90/solis_modbus/files/14842817/home-assistant_2024-04-02T19-28-16.689Z.log)

Pho3niX90, 2024-04-02:
Those logs just contains info that the rpi is undervolted :) 

I will dig some more

cgilis, 2024-04-03:
Hi @Pho3niX90 

I transferred a backup to a new Raspberry Pi 5, but it is still the same issue.
[home-assistant_2024-04-03T14-19-25.367Z.log](https://github.com/Pho3niX90/solis_modbus/files/14853617/home-assistant_2024-04-03T14-19-25.367Z.log)

Pho3niX90, 2024-04-03:
@cgilis I made some more changes, please see if 1.4.4-beta-1 made any difference?

cgilis, 2024-04-03:
It feels better, but it's still there.. sorry man :-) 

[home-assistant_2024-04-03T15-25-17.234Z.log](https://github.com/Pho3niX90/solis_modbus/files/14854628/home-assistant_2024-04-03T15-25-17.234Z.log)

Pho3niX90, 2024-04-03:
going to do another release now

Pho3niX90, 2024-04-03:
if you can try 1.4.4-beta-2

cgilis, 2024-04-03:
Feels again a bit better, but after 12 refreshes the loading is still there.. strange
[home-assistant_2024-04-03T15-43-59.967Z.log](https://github.com/Pho3niX90/solis_modbus/files/14854821/home-assistant_2024-04-03T15-43-59.967Z.log)

Pho3niX90, 2024-04-03:
okay let me investigate some more

Pho3niX90, 2024-04-03:
@cgilis once you have time, please test https://github.com/Pho3niX90/solis_modbus/releases/tag/1.4.4-beta-3
I believe this fixes the issue. I was able to reproduce it my side

cgilis, 2024-04-03:
Nice, Will try it right now.

cgilis, 2024-04-03:
> @cgilis once you have time, please test https://github.com/Pho3niX90/solis_modbus/releases/tag/1.4.4-beta-3 I believe this fixes the issue. I was able to reproduce it my side

🥳. Yes it’s working!! Can I buy you a beer? 🍺

Pho3niX90, 2024-04-03:
That's good news! Thanks so much for helping to find the issue. Quickly going to tidy things up, will then release the real 1.4.4

cgilis, 2024-04-03:
Is it ik that I remove my log files in this thread?

Pho3niX90, 2024-04-03:
Yes you can remove them. No longer needed

cgilis, 2024-04-03:
Small question about Solis RAI 3kW support. Do you think the only things I can control more? (screenshot on top)

Pho3niX90, 2024-04-03:
Going to check if I can find some documentation on that specific model. But I am sure there should be more controls available. 

The items that aren't working, do they work via the app?

cgilis, 2024-04-03:
The app is not usable anymore as from the installation of the plugin, but that is OK. This plugin is better 😂

but I think nothing was controllable before in the app..

cgilis, 2024-04-03:
> That's good news! Thanks so much for helping to find the issue. Quickly going to tidy things up, will then release the real 1.4.4

Confirmed working on 2024.4 🙃

Pho3niX90, 2024-04-03:
Yeah the modbus can only handle a single connection at a time. So once HA connects with the inverter, it breaks the cloud connectivity. To enable it again, you need to disable the solis ha plugin. 

The documentation I am currently finding states it's the same registrars, it might mean that certain switches needs to be off on the RAI before others take effect. Will dig some more

cgilis, 2024-04-04:
Hey @Pho3niX90  Since the update , these controls became unavailable 😅

[image]
[image]

Pho3niX90, 2024-04-04:
> Hey @Pho3niX90 Since the update , these controls became unavailable 😅
> 
> ![IMG_2284](https://private-user-images.githubusercontent.com/62599539/319456194-d99c9fdf-8c46-4ac1-be00-cc453d3d4662.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTIyMjkzMDcsIm5iZiI6MTcxMjIyOTAwNywicGF0aCI6Ii82MjU5OTUzOS8zMTk0NTYxOTQtZDk5YzlmZGYtOGM0Ni00YWMxLWJlMDAtY2M0NTNkM2Q0NjYyLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA0MDQlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwNDA0VDExMTAwN1omWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTkwMDQ0NTNiYjU3OThhZjI0NWMxZWZkMGEzODgxOTA2YTAyZmNlODY3Njg1OTdlZDA1ODNhNWM3NTc3OWZmOTgmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.IUwJXlvJkFTRn_kteIm3uDUbciPUi2WY6FnfV3-gsKo) ![IMG_2285](https://private-user-images.githubusercontent.com/62599539/319456202-671f777b-a721-4738-a98d-3797a7e5cc33.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTIyMjkzMDcsIm5iZiI6MTcxMjIyOTAwNywicGF0aCI6Ii82MjU5OTUzOS8zMTk0NTYyMDItNjcxZjc3N2ItYTcyMS00NzM4LWE5OGQtMzc5N2E3ZTVjYzMzLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA0MDQlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwNDA0VDExMTAwN1omWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTY1YzAzMmE2MTIyMjBjMDg3MzY3OWFhMDllNTAxNjA4MjUzMDQ2ZWIzMTc2OGVjN2JjZmJlMzY0MTJhNDVhODYmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.75cnJhfzSHPPP_yV7G2k0X4VyjmC6Or-W6nJmygCyHk)

release 1.4.5 should fix this

cgilis, 2024-04-04:
> release 1.4.5 should fix this

Perfect!! 😍 this works

cgilis, 2024-04-04:
Hey @Pho3niX90 found another bug. 
cant set te charging current. 

[image]
[image]

Pho3niX90, 2024-04-05:
the latest version should fix that as well

cgilis, 2024-04-05:
> the latest version should fix that as well

Will test it tonight. Thx man!

cgilis, 2024-04-08:
Sorry for my delayed response, but it is working with the latest version! Thx @Pho3niX90

commit dfed96a3 (2024-04-03): Update version and clean error handling in Modbus operations

The version number in the system has been updated from 1.4.3 to 1.4.4 in both manifest.json and const.py. Additionally, error handling for modbus operations within modbus_controller.py has been updated to streamline failure responses, eliminating unnecessary error check steps and focusing on exception-based error handling.

fixes #43 and #27
commit a98b2ded (2024-04-03): Feature/speedup (#47)

* Refactor update methods and adjust poll intervals

Changed the 'async_update' methods to just 'update' in sensor.py, number.py, and switch.py files. Additionally, the poll intervals have been adjusted. These modifications aim to enhance the efficiency and timing of data updates.

* Update async operations in sensor.py

The async operations for updating sensor entities have been refactored and encapsulated in a new function `get_modbus_updates`. This improves the asynchronous handling of updates, making the code more efficient and organized.

* Switch to async Modbus op [... cut]
commit da00d3c2 (2024-04-04): Feature/speedup (#48)

* - bump version

* - bump version

* Refactor update methods and adjust poll intervals

Changed the 'async_update' methods to just 'update' in sensor.py, number.py, and switch.py files. Additionally, the poll intervals have been adjusted. These modifications aim to enhance the efficiency and timing of data updates.

* Update async operations in sensor.py

The async operations for updating sensor entities have been refactored and encapsulated in a new function `get_modbus_updates`. This improves the asynchronous handling of updates, making the code more efficient and org [... cut]
closed 2024-09-11 by Pho3niX90