Pho3niX90/solis_modbus #45: Is it normal that after the installation SolisCloud is offline?

cgilis, 2024-03-25:
**Describe the bug**
On soliscloud the installation is offline

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.
[image]

0rangutan, 2024-03-25:
Yes if you are running both the HA Integration and Solis Cloud from a single Data Logger then this is expected behaviour.
The logger can only manage a single TCP/IP connection and so Solid Cloud will drop off like this.
To run both simultaneously, you need a setup similar to this one:  https://github.com/alienatedsec/solis-ha-modbus-cloud
Its a bit of work but runs nicely - I am using Pho3niX90's integration plus Solis Cloud in parallel: https://github.com/alienatedsec/solis-ha-modbus-cloud/discussions/24

cgilis, 2024-03-25:
Ok thx! @0rangutan

closed 2024-03-25 by cgilis