Pho3niX90/solis_modbus #49: Not working, TCP port required

HectorJRL, 2024-04-04:
Hi there! 
I'm trying to make this work with my Solis inverter, but I'm stuck at the beginning of the process.
After writing the inverter IP (doble checked that it's the right one), and NOT changing the tcp port by defect (502), it gives me an "unknown error ocurred".

[image]

I guess it's a simple mistake, but I can't find the way to figure out the proper tcp port.

Any help will be really appreciated.

Pho3niX90, 2024-04-05:
the ip should not contain http://, just the ip
192.168.1.65

closed 2024-04-06 by Pho3niX90

HectorJRL, 2024-04-17:
Sorry for the delay. I had to finish the solar panel instalation on the roof itself, and that took all my free time.

Same result removing the http, sadly...