Pho3niX90/solis_modbus #53: HA 2024.5 plugin not working

cgilis, 2024-05-02:
**Describe the bug**
Since the update to 2024.5 the plugin is not working anymore.

```
This error originated from a custom integration.

Logger: pymodbus.logging
Source: custom_components/solis_modbus/modbus_controller.py:24
integration: Solis Modbus (documentation, issues)
First occurred: May 1, 2024 at 21:47:29 (4 occurrences)
Last logged: May 1, 2024 at 22:01:27

Failed to connect
```

```
This error originated from a custom integration.

Logger: custom_components.solis_modbus.modbus_controller
Source: custom_components/solis_modbus/modbus_controller.py:26
integration: Solis Modbus (documentation, issues)
First occurred: May 1, 2024 at 21:47:29 (4 occurrences)
Last logged: May 1, 2024 at 22:01:27

Failed to connect to Modbus device. Will retry, failures = 1
```

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: [e.g. iOS]
 - Browser [e.g. chrome, safari]
 - Version [e.g. 22]

**Smartphone (please complete the following information):**
 - Device: [e.g. iPhone6]
 - OS: [e.g. iOS8.1]
 - Browser [e.g. stock browser, safari]
 - Version [e.g. 22]

**Additional context**
Add any other context about the problem here.

yurix2019, 2024-05-02:
I’ve also had this issue. No sensors reporting, no numeric state changes accepted.

Pho3niX90, 2024-05-03:
Investigating

0rangutan, 2024-05-03:
Same issue here

Pho3niX90, 2024-05-03:
@0rangutan @cgilis @yurix2019 could you confirm if release 1.4.7 solves the issue for you

cgilis, 2024-05-03:
> @0rangutan @cgilis @yurix2019 could you confirm if release 1.4.7 solves the issue for you

Confirmed working on **2024.4.4** and **2024.5**

Thx @Pho3niX90 for the fast fix! 🦸‍♂️

yurix2019, 2024-05-03:
[image]
Thanks for hard work. I’m on 2024.6 and most things are working for me with the exception of the charge/discharge amps and times (see screen grab).

Pho3niX90, 2024-05-03:
> [image]
> 
> Thanks for hard work. I’m on 2024.5 and most things are working for me with the exception of the charge/discharge amps and times (see screen grab).

I created a separate issue for this, as it seems it might have been broken in the last release as well?

0rangutan, 2024-05-03:
Fixed for me here, thanks @Pho3niX90

yurix2019, 2024-05-03:
For me it was more or less working on the core release before 2024.5. There were some anomalies with the charge and discharge amps defaulting to 50 amps, but I had work around automations to reset to zero when I wasn’t charging or discharging.

Pho3niX90, 2024-05-03:
@yurix2019 1.4.8 should have the amps fixed

yurix2019, 2024-05-03:
Thanks so much. It’s now working well.

This integration is useful and it saves me using the Solis Cloud app (not the greatest) or having to climb into my loft twice a day, so thanks again.

closed 2024-05-04 by Pho3niX90

cgilis, 2024-05-06:
Hey @Pho3niX90 i think it’s back after 2024.5.1 🥲

Pho3niX90, 2024-05-06:
> Hey @Pho3niX90 i think it’s back after 2024.5.1 🥲

I am running core 2024.5.1 without issues

cgilis, 2024-05-06:
This is in the logs:

```
Logger: homeassistant.config_entries
Source: config_entries.py:575
First occurred: 07:53:55 (2 occurrences)
Last logged: 10:08:41

Error setting up entry Solis: 192.168.1.*** for solis_modbus
Traceback (most recent call last):
  File "/config/custom_components/solis_modbus/modbus_controller.py", line 25, in connect
    raise _LOGGER.warning(
TypeError: exceptions must derive from BaseException

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/config_entries.py", line 575, in async_setup
    result = await component.async_setup_entry(hass, self)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/solis_modbus/__init__.py", line 65, in async_setup_entry
    await controller.connect()
  File "/config/custom_components/solis_modbus/modbus_controller.py", line 32, in connect
    raise _LOGGER.error(f"Failed to connect to Modbus device. Will retry")
TypeError: exceptions must derive from BaseException
```

Pho3niX90, 2024-05-06:
> Traceback (most recent call last):

Ware you trying to setup a new connection? Or was this just on restart?

cgilis, 2024-05-06:
> > Traceback (most recent call last):
> 
> Ware you trying to setup a new connection? Or was this just on restart?

Just on restart after I did the update to 2024.5.1.
Do I try to remove it and add the solis again?

PR #54 (2024-05-03): Feature/2024.05
fixes #53