Pho3niX90/solis_modbus #58: Support for regular string inverters

debietim, 2024-05-03:
Hello

I am using this integrations for my Solis RHI 10kW Hybrid inverter and it works very good!
I have also 2 regular inverters without battery. I't would be great if I can control this inverters also like I do with the Hybrid one. 

But today I get an error when I try to setup one of these inverters.

Is there an option to support also non-hybrid inverters?

regards

Pho3niX90, 2024-05-04:
Which errors are you getting?

debietim, 2024-05-04:
I can add them (by IP address) but then I gat a "failed" message...
[image]

Pho3niX90, 2024-05-04:
Which port are you using for them? Also, are there any other modbus software connecting to them? Lastly, do you know which wifi dongle is used with them?

debietim, 2024-05-04:
I am using port 502 (was standard filled in)… i am using the same wifi dongle as on my hybrid inverter… (S2-WL-ST)
No other modbus connection, only soliscloud…

0rangutan, 2024-05-04:
Ah, you can't use this integration and Solis Cloud simultaneously - one or the other.
Or set up an RS485 gateway - see the threads re. Waveshare

debietim, 2024-05-04:
I know, but when i installed my hybrid with this integration, it automatically stopped on soliscloud… but now it don’t even start…

debietim, 2024-05-14:
I have next to my Solis RHI 10kW also al Solis S5-GR3P8K string inverter. Now it would be very nice if I can control the Power Limit (value between 0 and 110%) via home assistant. Now I can only change this value via the SolisCloud platform. But when I use this integrations the device is not recognised... any chance that it can be supported in the future? I use the same datalogger as I use on the RHI 10kW inverter.

fastsonic, 2024-10-30:
everything 0
[image]

Pho3niX90, 2024-10-30:
#84 related

Pho3niX90, 2024-10-30:
There is a test version https://github.com/Pho3niX90/solis_modbus/releases/tag/string-test

debietim, 2024-11-04:
Hello, I installed the test-version but was forced to roll back.
The battery shows constant discharge of 139W.
When tried to add a Solis String Inverter, this did not work. It connected, but no values (complete none) came thru..

Pho3niX90, 2024-11-06:
Unfortunately the documentation I have found regarding the string inverters are unreliable to blindly code it. I would either need access to a system where I can do testing, or  someone can continue from https://github.com/Pho3niX90/solis_modbus/pull/86 to make a PR

fastsonic, 2024-11-07:
i can open a Port from the Elfin so you can connect to the Inverter over Modbus TCP !? 
but a little bit scared about security... is there a other Tip !?

Pho3niX90, 2025-02-03:
@fastsonic @debietim @christian-albrecht22 
The above includes registrars of the official documentation. However, some info was missing, such as the conversion rate. So this is a basic implementation, to test the critical sensors, if all is well, I will implement the rest

omenukas, 2025-02-20:
Hello,
I have an inverter s5-gr3p12k without batteries. 
I have made a modbus connection according to this description https://github.com/alienatedsec/solis-ha-modbus-cloud
The SolisCloud side is working fine.
If I manually add some known registry sensors to the .yaml, it shows them as well.
For example:
  - name: modbus
    timeout: 1
    type: tcp
    host: 192.168.XX.XX
    port: 502
    sensors:
      - name: Solis power
        device_class: power
        data_type: uint32
        slave: 1
        address: 3004
        input_type: input
        unit_of_measurement: W
        state_class: measurement
        scan_interval: 20
        
      - name: Solis today energy
        device_class: energy
        data_type: uint16
        slave: 1
        address: 3014
        input_type: input
        unit_of_measurement: kWh
        state_class: total_increasing
        scan_interval: 60
        scale: 0.1
        precision: 1     
      - name: Solis total energy
        device_class: energy
        data_type: uint32
        slave: 1
        address: 3008
        input_type: input
        unit_of_measurement: kWh
        state_class: total_increasing
        scan_interval: 300

[image]

However, I do not know how to extract smartmeter grid data.
A smartmeter Arcel DTSD1352C is connected to the inverter via the meter port. The data about the voltages, currents and power of all three phases of the grid from this smartmeter are all displayed in SoliCloud. I don't know how to get this data into HA.
Therefore, I tried to install this Solis modbus integration, but with not one option, all sensors are either inactive or with zero values.
Is there any chance that I can see grid energy consumption with this integration?

Pho3niX90, 2025-02-20:
> Hello, I have an inverter s5-gr3p12k without batteries. I have made a modbus connection according to this description https://github.com/alienatedsec/solis-ha-modbus-cloud The SolisCloud side is [...]

These registers doesn't seem to correspond with the solis pv/grid doc.
 
See here:

[image]

[image]

[image]

If you have confirmed the values you are getting are in fact accurate, where did you get the registrars from?

omenukas, 2025-02-20:
from this document:

[RS485_MODBUS Communication Protocol_Solis Inverters_1.pdf](https://github.com/user-attachments/files/18888382/RS485_MODBUS.Communication.Protocol_Solis.Inverters_1.pdf)

just need to evaluate:

[image]

Pho3niX90, 2025-02-20:
> from this document:
> 
> [RS485_MODBUS Communication Protocol_Solis Inverters_1.pdf](https://github.com/user-attachments/files/18888382/RS485_MODBUS.Communication.Protocol_Solis.Inverters_1.pdf)
> 
> just need to evaluate:
> 
> [image]

Thanks for pointing that out, I have missed that part completely. 

I cannot seem to find any reference in regards to the smart port for grid/string inverters

omenukas, 2025-02-20:
And I have already found some parameters I need from smartmeter:

      - name: S5 meter voltage A
        device_class: voltage
        slave: 1
        address: 3250
        input_type: input
        data_type: uint16
        scale: 0.1
        unit_of_measurement: "V"
        scan_interval: 10
        precision: 1
        
      - name: S5 meter A phase current
        device_class: current
        data_type: uint16
        slave: 1
        address: 3251
        input_type: input
        unit_of_measurement: A
        state_class: measurement
        scan_interval: 10
        scale: 0.01
        precision: 2

      - name: S5 meter import energy total
        device_class: energy
        state_class: total_increasing
        data_type: uint32
        slave: 1
        address: 3282
        input_type: input
        unit_of_measurement: kWh
        scan_interval: 300
        scale: 0.01
        precision: 2
        
      - name: S5 meter export energy total
        device_class: energy
        state_class: total_increasing
        data_type: uint32
        slave: 1
        address: 3284
        input_type: input
        unit_of_measurement: kWh
        scan_interval: 300
        scale: 0.01
        precision: 2

Similarly for phases B and C.

However, I do not find the most important for me - active power in each phase.
Maybe these parameters could be what I need, but I don't know how to describe a sensor with this "device type - S32":

[image]

omenukas, 2025-02-20:
I already found how to find the power consumption:

      - name: S5 meter active power A
        device_class: power
        data_type: int32
        slave: 1
        address: 3256
        input_type: input
        unit_of_measurement: kW
        state_class: measurement
        scan_interval: 10
        scale: 0.001
        precision: 3

closed 2025-02-24 by Pho3niX90