Pho3niX90/solis_modbus #60: Wanted to install a new installation

cgilis, 2024-05-16:
**Describe the bug**
Hi, 

I have a new installation I want to add this plugin to.
But I always get this error:

[image]

Here is the error log:
```
This error originated from a custom integration.

Logger: custom_components.solis_modbus.modbus_controller
Source: custom_components/solis_modbus/modbus_controller.py:32
integration: solis_modbus (documentation, issues)
First occurred: 17:32:05 (14 occurrences)
Last logged: 17:36:53

Failed to connect to Modbus device. Will retry
```

```
This error originated from a custom integration.

Logger: custom_components.solis_modbus.modbus_controller
Source: custom_components/solis_modbus/modbus_controller.py:25
integration: solis_modbus (documentation, issues)
First occurred: 17:32:05 (14 occurrences)
Last logged: 17:36:53

Failed to connect to Modbus device. Will retry, failures = 1
```

```
This error originated from a custom integration.

Logger: pymodbus.logging
Source: custom_components/solis_modbus/modbus_controller.py:23
integration: solis_modbus (documentation, issues)
First occurred: 17:32:05 (14 occurrences)
Last logged: 17:36:53

Failed to connect [Errno 111] Connect call failed ('192.168.0.128', 502)
Failed to connect [Errno 111] Connect call failed ('192.168.0.101', 502)
Failed to connect [Errno -2] Name does not resolve
Failed to connect [Errno 111] Connect call failed ('192.168.0.191', 502)
```
[home-assistant_2024-05-16T15-44-35.549Z.log](https://github.com/Pho3niX90/solis_modbus/files/15337677/home-assistant_2024-05-16T15-44-35.549Z.log)

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: [e.g. iOS]
 - Browser [e.g. chrome, safari]
 - Version [e.g. 22]

**Smartphone (please complete the following information):**
 - Device: [e.g. iPhone6]
 - OS: [e.g. iOS8.1]
 - Browser [e.g. stock browser, safari]
 - Version [e.g. 22]

**Additional context**
Add any other context about the problem here.

cgilis, 2024-05-16:
Looks like the same error in #49

Pho3niX90, 2024-05-16:
Will investigate the reason behind this. Will have to check how I can reproduce this without losing my data on my installation 😅

cgilis, 2024-05-16:
> Will investigate the reason behind this. Will have to check how I can reproduce this without losing my data on my installation 😅

Super @Pho3niX90 !! Would it help to give you access to my test server?
Can you contact me by mail, so I can share the credentials? --> Christoffel@me.com

Pho3niX90, 2024-05-16:
Also, if you enter the IP into a browser window, can you reach the inverters' web interface?

cgilis, 2024-05-16:
Is something I cannot test right now, because it's on a remote server 🥲

cgilis, 2024-05-16:
Will go to the remote location tomorrow morning and will report back!

cgilis, 2024-05-16:
[image]

@Pho3niX90 with VPN it worked the ip is 192.168.0.101

Pho3niX90, 2024-05-17:
> [image]
> 
> @Pho3niX90 with VPN it worked the ip is 192.168.0.101

To be clear, the HA device trying to access the solis is on the same network right? Will send you an email for access.

cgilis, 2024-05-17:
@Pho3niX90 after updating the firmware of the Solis, the problem is fixed and is now visible in HA.
Do you still want to check how you can fix it on older firmwares?

Pho3niX90, 2024-05-17:
> @Pho3niX90 after updating the firmware of the Solis, the problem is fixed and is now visible in HA.
> Do you still want to check how you can fix it on older firmwares?

Aah perfect. I need to note it down, I remember reading that one of the firmwares did have a modbus comms issue.

cgilis, 2024-07-10:
Hi @Pho3niX90 ,

The same error occurred with an updated Solis.
[image]

The log file is here:
[home-assistant_2024-07-10T11-28-48.353Z.log](https://github.com/user-attachments/files/16160032/home-assistant_2024-07-10T11-28-48.353Z.log)

Any idea?

Pho3niX90, 2024-07-10:
> Hi @Pho3niX90 ,
> 
> The same error occurred with an updated Solis.
> [image]
> 
> The log file is here:
> [home-assistant_2024-07-10T11-28-48.353Z.log](https://github.com/user-attachments/files/16160032/home-assistant_2024-07-10T11-28-48.353Z.log)
> 
> Any idea?
> 
> 
> Fyi, i can ping the solis:
> [image]
> 

Strange. The ping seems fine, but the logs state it couldn't open a connection. I am wondering if the modbus port isn't occupied, the only way to release it is to restart the dongle. 

Perhaps try removing the integration completely, restart and install again?

cgilis, 2024-07-10:
I've tried removing the integration and did not work.
Also tried restarting the solis, but was the same error..

cgilis, 2024-07-10:
The solis is also on the latest update (web ver: 2.4.0)
And I see there is a MQTT integration now, is that something we can try? 
[image]

Pho3niX90, 2024-07-10:
The plugin doesn't have mqtt support, as that is something  completely different. 

Will need to try to see if the port is still 502 on the new version, as it's possible that they have changed it. I will need yo request an update on my system to test with the same version.

Pho3niX90, 2024-07-10:
> I've tried removing the integration and did not work.
> Also tried restarting the solis, but was the same error.. not sure if the wifi stick restarts when the solis restarts.. :-) 

Not entirely sure. If power is reset to the stick, it should. The best is to hold in the button on it to restart.

cgilis, 2024-07-10:
> The plugin doesn't have mqtt support, as that is something completely different.
> 
> Will need to try to see if the port is still 502 on the new version, as it's possible that they have changed it. I will need yo request an update on my system to test with the same version.

If you need I can give you access to the HA to test it.. let me know.

Pho3niX90, 2024-07-10:
> I have another inverter on 2.4 and that still works with your plugin @Pho3niX90 so I think the port hasn't changed.

What's the difference between the 2?

Pho3niX90, 2024-07-10:
> > > I have another inverter on 2.4 and that still works with your plugin @Pho3niX90 so I think the port hasn't changed.
> > 
> > What's the difference between the 2?
> 
> No difference between them, strange :( 
> I've restarted the dongle, but there's no difference.

So can you explain the setup to me so that I can mimick it my side? 

Are both connecting to same HA instace?
Are the connected ad mater / slave, or independent from oneanother? 
Are they on the same network? 
What else can you tell me?

cgilis, 2024-07-11:
> So can you explain the setup to me so that I can mimick it my side?
>
> Are both connecting to same HA instace? Are the connected ad mater / slave, or independent from oneanother? Are they on the same [...]

So can you explain the setup to me so that I can mimick it my side?

Are both connecting to same HA instace?
Nope, those are separate homes with separate HA / Inverters
Are the connected ad mater / slave, or independent from oneanother?
Are they on the same network?
Nope different location.
What else can you tell me?
It's exactly the same setup, but in a different location.

cgilis, 2024-07-15:
@Pho3niX90 I've tried now on another location with another Solis inverter (now not on a non hybrid one) and this one is also not connecting.. strange.

Pho3niX90, 2024-07-15:
@cgilis are you able to give me access to one of these systems then I can do some diagnostics? If you have discord, send me a message @ pho3nix90

cgilis, 2024-07-15:
> @cgilis are you able to give me access to one of these systems then I can do some diagnostics? If you have discord, send me a message @ pho3nix90

Great! My profile is cgilis, just send you a message.

closed 2024-10-26 by Pho3niX90