Pho3niX90/solis_modbus #64: Inverter values are zero

sjoerd314, 2024-06-10:
HAS does create all the entities. But al the values are zero. Probably because the datalogger stick is not communicating well with the inverter, although the orange light on the datalogger is steady.

[image]

sjoerd314, 2024-06-11:
I have two inverters. I upgraded one of them with a s2-wl-st datalogger, the other one still has a old one that does not work with HAS. The new datalogger does connect to home assistant but only shows zero's. The webpage of that one is shown in the page above. The old one's web page is shown below 👇.
[image]

fastsonic, 2024-06-14:
I see that he take both solis to one IP Adress... 
[image]
Maybe an Problem from the plugin, or is here somthing to check bevor adding a second one? I also use Different Ports with Different iP

sjoerd314, 2024-06-14:
I already have the problem using one inverter. The second one is not yet connected

fastsonic, 2024-06-17:
I also try the Solax Plugin for the Communication. From My Side its not a Problem from this Integration.  
The Stick is communicate but i can´t register the Solax S6. must be a Problem from the Solis side.
Will give future Update when i have mor Time to test.

fastsonic, 2024-07-15:
I will give an update.
I also order an new Stick. That one communicate with the Solis Cloud and Values are correct. Next step is to Update from Solis support all Firmware for Inverter and Stick and check again. My Tests with SOLAX plugin also failed. Have no Idea where the Error is sitting

fastsonic, 2024-07-15:
I Think i found the Problem..
i test  communication and have access with direct modbus communication with IOBroker modbus Plugin. So the Modbus is working. But the S6 is not compatible to this Plugin, also to the Solax plugin. i will try to make an Update to latest firmware from the S6 Inverter and check again. Otherwise i have to read some register..

Pho3niX90, 2024-09-11:
> I Think i found the Problem.. i test communication and have access with direct modbus communication with IOBroker modbus Plugin. So the Modbus is working. But the S6 is not compatible to this [...]

The S6 is definitely compatible, I myself use it :) 

The compatibility lies more towards the connection, either ethernet or wifi dongle, and it seems there are multiple wifi dongles being shipped. 

Forcing a new IP for the wifi dongle will break forgotten integrations that might be interfering with the connection, since modbus can only have a single connection at a time, the first one to connect, will have data, the ones that follow, will either not connect or simply return zero data.

fastsonic, 2024-09-11:
I think i use first LanAdapter for access and changed to an Wlanadapter. Also at some point i chaged the IP from Modbusadapter..
How i can reset the Problem ? I think only a new IP-Adress is not resetting the Communikation..
Thanks

my-wheel, 2024-09-16:
I can confirm the problem. I observed it when trying to use the Solis Cloud App in parallel to the Solis Modbus Integration.
Some folks here on github have suggested solutions to the problem but these require additional hardware.
See for example 
[](https://github.com/Jumpy07/Solis---SolisCloud-and-Home-Assistant)

I believe Jumpy07  also explains the reason for the issues (see Readme of the link above):
"The RS485 / Modbus comms consist of Slaves and Masters, the Solis Inverter is a Slave, and the Solis Data Stick is a Master, unfortunately the Home Assistant Connection is also a Master and on a RS485 bus, typically only one master is allowed. To therefore allow both the Solis Data Logger and the Home Assistant connections to work without issues or loosing data, we need a way of allowing two masters on the RS485 bus." 

I would like to avoid additional hardware and would prefer (ask asked by fastsonic) a simple way to reset to switch between two masters.

my-wheel, 2024-09-16:
sorry - the link to jumpy07 has not been included in my comment:
https://github.com/Jumpy07/Solis---SolisCloud-and-Home-Assistant
I apologize - still new here ...

fastsonic, 2024-10-01:
Hi, 
I update my Installation with the Manual from Jumpy to have both, Cloud and Homeassistant.
The RHI5k working fine now on both
The S6 only have Data in the Cloud. When i communicate over Modbus all Values are Zero 
In the Communication from the EW11 is see a lot of in and out going communication but values are Zero.
changed IP did not change anything.  any more ideas ?
[image]

fastsonic, 2024-10-17:
I here that the Solis Hybrid had the Same Modbus adresses, but the S6 Inverer Modbusadresses are -1 so 3001 is 3000
How i can change the Adresses temp for an test ?
Thanks

fastsonic, 2024-10-28:
Little Update.
EVCC Also communicate to the Solis over Modbus and EW11. No Problems. So i think adresses are incorrect fot this Solis.
Maybe we have an update for the new Modbusadresses for this Type of Solis 
S6-GR1P4K

matt0020190, 2024-10-29:
> Little Update. EVCC Also communicate to the Solis over Modbus and EW11. No Problems. So i think adresses are incorrect fot this Solis. Maybe we have an update for the new Modbusadresses for this Type of Solis S6-GR1P4K

Hey, sorry I am late to this thread but I also have an Solis S6 Inverter with S2-WL-ST datalogger.

After Solis support upgraded the datalogger firmware, I can connect via Home Asssistant but all values of the sensors remain as 0. I removed the datalogger from Solis Cloud and still cannot seem to connect the inverter locally.

Forgive me for not understanding but is this therefore a known issue or is it my setup?

fastsonic, 2024-10-30:
i use an Elfin and S2 Stick together with an 2Master1Slave modul. Cloud and HA together.
But also become no Values from the Inverter.
EVCC is an HA Addon for Wallbox charge control and this one can access to the Inverter and become Modbus Data. So in the moment its definitv the Plugin here, that, what ever have a Problem with this typ of converter.. I also have an Ticket open for the Solax Plugin that can read my RHI5k but also can´t read my S6

Pho3niX90, 2024-10-30:
There are 2 completely different topics in here. One is about data not being in the Solis cloud app, which is the expected behavior, another is a certain inverter not having values within home assistant. 

Then, when mentioning a solar inverter, please use the full name of the inverter, as S6 is a very broad category, since my **S6**-EH1P6K-L-PRO works perfectly fine. 

To integrate other inverters, I would need the registrars, if you can find documentation for your specific inverter, please post them

Pho3niX90, 2024-10-30:
I am closing this discussion as things are a bit mixed in here. Please use the new one to post the relevant info there, if you are not having an issue with the `Solis S6-GR1P4K`, just create a new ticket with `inverter-model support` in the title.

closed 2024-10-30 by Pho3niX90