Pho3niX90/solis_modbus #69: Not working in 2024.6

cgilis, 2024-06-25:
**Describe the bug**
Hey @Pho3niX90, i'm a bit late to test this, but it was not working on 2024.6?
I've now downgraded back to 2024.5 and its working fine :-)

But 2024.7 is almost coming that's why I'm reporting this :-)

matthewbyrne, 2024-07-02:
I thought the same, but then I updated the plugin in HACS and gave it restart and it's working again.

Running 2024.05.7 with HA version 2024.6.4

cgilis, 2024-07-03:
> I thought the same, but then I updated the plugin in HACS and gave it restart and it's working again.
> 
> Running 2024.05.7 with HA version 2024.6.4

If I check for updates I don't see one in Hacs.. 😢

Pho3niX90, 2024-07-03:
> **Describe the bug** Hey @Pho3niX90, i'm a bit late to test this, but it was not working on 2024.6? I've now downgraded back to 2024.5 and its working fine :-)
> 
> But 2024.7 is almost coming that's why I'm reporting this :-)

this is strange, I am on 2024.6.4 without issues.

cgilis, 2024-07-03:
Ok will try again :)

cgilis, 2024-07-03:
Here it's not working @Pho3niX90 , strange.
The log file is here:
[home-assistant_2024-07-03T11-55-59.294Z.log](https://github.com/user-attachments/files/16083203/home-assistant_2024-07-03T11-55-59.294Z.log)

matthewbyrne, 2024-07-03:
You can hit 192.168.1.247 from your HA instance, right? 

Try `telnet 192.168.1.247 502`. That's the modbus port

cgilis, 2024-07-03:
> You can hit 192.168.1.247 from your HA instance, right?
> 
> Try `telnet 192.168.1.247 502`. That's the modbus port

[image]
This is what i get.. hmmm

Pho3niX90, 2024-07-03:
> running

Those response time might be causing timeouts, default, I think is 4seconds.

Did something change in your networking that might be causing the major delays? You could try and restart your network hardware, and perhaps, choose a less congested wifi channel.

Either way, will see if I can add in some logic to retry at longer timeouts.

cgilis, 2024-07-03:
> Those response time might be causing timeouts, default, I think is 4seconds.
>
> Did something change in your networking that might be causing the major delays? You could try and restart your network [...]

Strange, nothing changed, if i downgrade back to 2024.5 it works perfectly..
Will reboot my router.. and try again

cgilis, 2024-07-03:
[image]

Did reboot of the wifi stick of Solis, and now its working in HA.. crazy :-)
Thx @Pho3niX90 & @matthewbyrne

cgilis, 2024-07-03:
Same issue on 2024.7
Strange 😅

cgilis, 2024-07-03:
Strange now its back, took some time... if you need the logs let me know @Pho3niX90

Pho3niX90, 2024-07-03:
> Strange now its back, took some time... if you need the logs let me know @Pho3niX90 

Can you grab logs and then restart stick again?

cgilis, 2024-07-04:
> > Strange now its back, took some time... if you need the logs let me know @Pho3niX90
> 
> Can you grab logs and then restart stick again?

[home-assistant_2024-07-04T07-50-47.370Z.log](https://github.com/user-attachments/files/16095201/home-assistant_2024-07-04T07-50-47.370Z.log)
Is this ok?

closed 2024-08-06 by Pho3niX90