Pho3niX90/solis_modbus #74: Wrong values

Dominic-070, 2024-09-24:
**The comma is incorrect for these entities. the value indicated is 10 times too high.**
sensor.solis_s6_solis_today_battery_charge_energy
sensor.solis_s6_solis_today_battery_discharge_energy
sensor.solis_s6_solis_yesterday_battery_charge_energy
sensor.solis_s6_solis_yesterday_battery_discharge_energy

**the comma is incorrect for these entities. the value indicated is 100 times too high.**
sensor.solis_s6_solis_total_battery_charge_energy
sensor.solis_s6_solis_total_battery_discharge_energy

Pho3niX90, 2025-02-06:
The sensors are now split out into separate files. If you can update the PR, and perhaps add a third option "hybrid-waveshare", then we can merge it

Dominic-070, 2025-02-06:
Hi that would be graet! 

I dont know how to update the pull request but in my own system i changed solis_modbus/data/hybrid_sensors.py to this to get the right values

`from homeassistant.components.sensor.const import SensorDeviceClass, SensorStateClass
from homeassistant.const import UnitOfElectricPotential, UnitOfElectricCurrent, UnitOfPower, UnitOfTime, UnitOfEnergy, \
    UnitOfReactivePower, UnitOfFrequency, UnitOfTemperature, UnitOfApparentPower, PERCENTAGE

hybrid_sensors = [
    {
        "register_start": 33000,
        "scan_interval": 0,
        "entities": [
            {"type": "SS", "name": "Solis Model No", "unique": "solis_modbus_inverter_model_no",
             "register": ['33000'], "multiplier": 0},
            {"type": "SS", "name": "Solis DSP Version", "unique": "solis_modbus_inverter_dsp_version",
             "register": ['33001'], "multiplier": 0},
            {"type": "SS", "name": "Solis HMI Version", "unique": "solis_modbus_inverter_hmi_version",
             "register": ['33002'], "multiplier": 0},
            {"type": "SS", "name": "Solis Protocol Version", "unique": "solis_modbus_inverter_protocol_version",
             "register": ['33003'], "multiplier": 0},
            {"type": "SS", "name": "Solis Serial Number", "unique": "solis_modbus_inverter_serial_number",
             "register": [
                 '33004', '33005', '33006', '33007', '33008', '33009', '33010', '33011', '33012', '33013', '33014',
                 '33015', '33016', '33017', '33018', '33019',
             ], "multiplier": 0}
        ]
    },
    {
        "register_start": 33025,
        "scan_interval": 60,
        "entities": [

            {"type": "SS", "name": "Solis Clock (Hours)",
             "unique": "solis_modbus_inverter_clock_hours",
             "register": ['33025'], "multiplier": 0,
             "unit_of_measurement": UnitOfTime.HOURS, "state_class": SensorStateClass.MEASUREMENT},
            {"type": "SS", "name": "Solis Clock (Minutes)",
             "unique": "solis_modbus_inverter_clock_minutes",
             "register": ['33026'], "multiplier": 0,
             "unit_of_measurement": UnitOfTime.MINUTES, "state_class": SensorStateClass.MEASUREMENT},
            {"type": "SS", "name": "Solis Clock (Seconds)",
             "unique": "solis_modbus_inverter_clock_seconds",
             "register": ['33027'], "multiplier": 0,
             "unit_of_measurement": UnitOfTime.SECONDS, "state_class": SensorStateClass.MEASUREMENT},

            {"type": "reserve", "register": ['33028']},

            {"type": "SS", "name": "Solis PV Total Energy Generation",
             "unique": "solis_modbus_inverter_pv_total_generation",
             "register": ['33029', '33030'], "device_class": SensorDeviceClass.ENERGY, "multiplier": 0,
             "unit_of_measurement": UnitOfEnergy.KILO_WATT_HOUR, "state_class": SensorStateClass.TOTAL_INCREASING},
            {"type": "SS", "name": "Solis PV Current Month Energy Generation",
             "unique": "solis_modbus_inverter_pv_current_month_generation",
             "register": ['33031', '33032'], "device_class": SensorDeviceClass.ENERGY, "multiplier": 1,
             "unit_of_measurement": UnitOfEnergy.KILO_WATT_HOUR, "state_class": SensorStateClass.TOTAL_INCREASING},
            {"type": "SS", "name": "Solis PV Last Month Energy Generation",
             "unique": "solis_modbus_inverter_pv_last_month_generation",
             "register": ['33033', '33034'], "device_class": SensorDeviceClass.ENERGY, "multiplier": 1,
             "unit_of_measurement": UnitOfEnergy.KILO_WATT_HOUR, "state_class": SensorStateClass.TOTAL_INCREASING},
            {"type": "SS", "name": "Solis PV Today Energy Generation",
             "unique": "solis_modbus_inverter_pv_today_generation",
             "register": ['33035'], "device_class": SensorDeviceClass.ENERGY, "multiplier": 0.1,
             "unit_of_measurement": UnitOfEnergy.KILO_WATT_HOUR, "state_class": SensorStateClass.TOTAL_INCREASING},
            {"type": "SS", "name": "Solis PV Yesterday Energy Generation",
             "unique": "solis_modbus_inverter_pv_yesterday_generation",
             "register": ['33036'], "device_class": SensorDeviceClass.ENERGY, "multiplier": 0.1,
             "unit_of_measurement": UnitOfEnergy.KILO_WATT_HOUR, "state_class": SensorStateClass.TOTAL_INCREASING},
            {"type": "SS", "name": "Solis PV This Year Energy Generation",
             "unique": "solis_modbus_inverter_pv_this_year_generation",
[... 542 lines cut ...]
     "unique": "solis_modbus_inverter_dc_power_2", "device_class": SensorDeviceClass.POWER, "multiplier": 0.1,
     "unit_of_measurement": UnitOfPower.WATT, "state_class": SensorStateClass.MEASUREMENT,
     "register": ['33051', '33052']},

    {"type": "SDS", "name": "Solis PV Power 3",
     "unique": "solis_modbus_inverter_dc_power_3", "device_class": SensorDeviceClass.POWER, "multiplier": 0.1,
     "unit_of_measurement": UnitOfPower.WATT, "state_class": SensorStateClass.MEASUREMENT,
     "register": ['33053', '33054']},

    {"type": "SDS", "name": "Solis PV Power 4",
     "unique": "solis_modbus_inverter_dc_power_4", "device_class": SensorDeviceClass.POWER, "multiplier": 0.1,
     "unit_of_measurement": UnitOfPower.WATT, "state_class": SensorStateClass.MEASUREMENT,
     "register": ['33055', '33056']},

    {"type": "SDS", "name": "Solis Battery Charge Power",
[... cut ...]
     "multiplier": 0.1,
     "unit_of_measurement": UnitOfPower.WATT, "state_class": SensorStateClass.MEASUREMENT,
     "register": ['33149', '33150', '33135', '0']},
    {"type": "SDS", "name": "Solis Battery Discharge Power",
     "unique": "solis_modbus_inverter_battery_discharge_power", "device_class": SensorDeviceClass.POWER,
     "multiplier": 0.1,
     "unit_of_measurement": UnitOfPower.WATT, "state_class": SensorStateClass.MEASUREMENT,
     "register": ['33149', '33150', '33135', '1']}
]`

Pho3niX90, 2025-02-06:
Okay no proble, I will make the additional changes where needed

Pho3niX90, 2025-02-06:
I made it part of the current pre release https://github.com/Pho3niX90/solis_modbus/releases/tag/1.6.0-pre5

closed 2025-02-06 by Pho3niX90

Dominic-070, 2025-02-06:
Works perfectly, had to remove and add again, but now works perfectly. thank you for all the effort!

commit 6890e547 (2025-02-06): fix: add waveshare values, fixes #74
commit 61561ddf (2025-02-06): implements #93 #94 (#102)

feat: #58 & #84 test
fixes: #15 #79
implements #93 #94
addition of time of use
faster startup
add waveshare values, fixes #74