Pho3niX90/solis_modbus #78: After installed the 1.5.2 update my Inverter failed to come back up with the following:-

miketarg, 2024-10-05:
Logger: homeassistant.config_entries
Source: config_entries.py:604
First occurred: 8:09:30 AM (2 occurrences)
Last logged: 8:12:50 AM

Error setting up entry Solis: 192.168.0.83 for solis_modbus
Traceback (most recent call last):
  File "/config/custom_components/solis_modbus/modbus_controller.py", line 25, in connect
    raise _LOGGER.warning(
TypeError: exceptions must derive from BaseException

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/config_entries.py", line 604, in async_setup
    result = await component.async_setup_entry(hass, self)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/solis_modbus/_init_.py", line 65, in async_setup_entry
    await controller.connect()
  File "/config/custom_components/solis_modbus/modbus_controller.py", line 32, in connect
    raise _LOGGER.error(f"Failed to connect to Modbus device. Will retry")
TypeError: exceptions must derive from BaseException

I had to system restore to my backup then it works fine.

Pho3niX90, 2024-10-05:
reverted the problematic code in 1.5.3 until it can be tested.

closed 2024-10-05 by Pho3niX90

miketarg, 2024-10-05:
Wait . . . Wasn't it rested before you pushed it out?