Pho3niX90/solis_modbus #79: Errors in Logas after updating again

miketarg, 2024-10-06:
Logger: homeassistant.config_entries
Source: config_entries.py:594
First occurred: 3:20:22 PM (2 occurrences)
Last logged: 3:21:47 PM

Error setting up entry Solis: 192.168.0.83 for solis_modbus
Traceback (most recent call last):
  File "/config/custom_components/solis_modbus/modbus_controller.py", line 25, in connect
    raise _LOGGER.warning(
TypeError: exceptions must derive from BaseException

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/config_entries.py", line 594, in async_setup
    result = await component.async_setup_entry(hass, self)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/solis_modbus/__init__.py", line 65, in async_setup_entry
    await controller.connect()
  File "/config/custom_components/solis_modbus/modbus_controller.py", line 32, in connect
    raise _LOGGER.error(f"Failed to connect to Modbus device. Will retry")
TypeError: exceptions must derive from BaseException

Updated as the last ticket was closed as fixed but I am getting errors again and my inverter wont connect to HA.

miketarg, 2024-10-09:
Hello, can you please take a look at this as I tried updating as you mentiond this had been fixed but it appears to be causing issues still, please can you let me know if you need any other information?

Pho3niX90, 2024-10-26:
Will delve more deeply into this for the next version. The one released today should be better already

commit ec47a2b3 (2025-01-30): fixes: #79
commit 28ae6d91 (2025-01-31): multi modbus support (#98)

* fix(modbus_controller): log errors and return None in Modbus register methods
Modified `async_read_input_register`, `async_read_holding_register`,
`async_write_holding_register`, and `async_write_holding_registers` to log errors
and return None instead of raising exceptions, ensuring smoother error handling
without interruptions in execution.

* refactor: improve value check

* refactor: remove dupe

* refactor: register fix

* fixes: #15

* fixes: #79
closed 2025-01-31 by Pho3niX90