Pho3niX90/solis_modbus #84: Solis S6-GR1P4K support

Pho3niX90, 2024-10-30:
Little Update.
EVCC Also communicate to the Solis over Modbus and EW11. No Problems. So i think adresses are incorrect fot this Solis.
Maybe we have an update for the new Modbusadresses for this Type of Solis 
S6-GR1P4K

_Originally posted by @fastsonic in https://github.com/Pho3niX90/solis_modbus/issues/64#issuecomment-2441856816_

Pho3niX90, 2024-10-30:
As far as my knowledge goes, the inverter registrars are same across the board for the different inverters.  @fastsonic can you link the project you say is currently successfully reading your values

fastsonic, 2024-10-30:
https://github.com/evcc-io/evcc

It use also the Elfin Stick to read Data. 
Settings:
[image]

Pho3niX90, 2024-10-30:
Perhaps related to https://github.com/Pho3niX90/solis_modbus/issues/58#issuecomment-2205902880

Pho3niX90, 2024-10-30:
Does the plugin not provide any info for any registers? I see `S6-GR1P4K` is in fact a hybrid, which this plugin should support.

fastsonic, 2024-10-30:
only zero Values 
[image]

Pho3niX90, 2024-10-30:
Can you confirm if you inverter is a hybrid or normal inverter?

It seems your inverter might not be hybrid from the looks of things, as even the evcc is stating https://github.com/evcc-io/evcc/blob/4b51a3014bacd9217fe97a5ba9b00f7d59825b90/templates/definition/meter/solis.yaml#L2

instead of 
https://github.com/evcc-io/evcc/blob/4b51a3014bacd9217fe97a5ba9b00f7d59825b90/templates/definition/meter/solis-hybrid.yaml#L4

Pho3niX90, 2024-10-30:
Is this your inverter
https://www.solisinverters.com/uploads/file/Solis_datasheet_S6-GR1P(2,5-6)K-S_IND_V1,2_2023_04.pdf

or this
https://solarclarity.com/en/product/234123/solis-s6-gr1p4k

fastsonic, 2024-10-30:
second one . 1 Phase no battery

Pho3niX90, 2024-10-30:
> second one . 1 Phase no battery

I am going to create a "beta" version that you can test, will let you know when it's available.

fastsonic, 2024-10-30:
Hi,
Solis active Power and Solis Total DC output showing numbers but both to high
Solis Total DC Output Power
11.337.728 W
Solis Active Power
8.519.680 W
I have in the moment ~ 100 W

fastsonic, 2024-10-30:
Sorry.  For Today sun is gone. Wait for the Sun tomorrow to give you an update.
Thanks

Pho3niX90, 2024-10-31:
> Sorry. For Today sun is gone. Wait for the Sun tomorrow to give you an update. Thanks

How was the results today?

fastsonic, 2024-11-01:
Hi Sorry,
all values now 0
the S2 Stick and the Elfinstick,
Both show different Pictures.. 
Elfin Stick over Modbus
[image]
Solis S2 Stick over Modbus
[image]

wgentine, 2024-12-23:
Super interested in this PR since I have the 3K version of the same model.

If you need access to the modbus port to tweak it, I can provide but I can also scan and provide the data.

Pho3niX90, 2025-02-07:
> Super interested in this PR since I have the 3K version of the same model.
> 
> If you need access to the modbus port to tweak it, I can provide but I can also scan and provide the data.

Can you confirm if the newest version is working better? When you setup your solis and choose the type as string or grid

wgentine, 2025-02-17:
Sorry for the late reply. I ended up mapoing every modbus register to a sensor manually

wgentine, 2025-02-17:
Will try again later today and provide feedback.

Pho3niX90, 2025-02-20:
@wgentine @fastsonic any feedback?

This is the newest version. Once again, it contains only a few registers until I am certain they are the correct ones. Will close the issue sunday if I haven't received any confirmation as yet
https://github.com/Pho3niX90/solis_modbus/releases/tag/3.0.0-beta-3

wgentine, 2025-02-23:
Dear @Pho3niX90, apologies for the late reply. Chaotic days.
I just tested and could pull some data.

[image]

[image]

[image]

Pho3niX90, 2025-02-23:
Hiw accurate is the data received? Do you perhaps have some logs for me?

wgentine, 2025-02-23:
DC Current, DC Voltage, Active Power and Total DC Output Power seems to be correct. The rest I would say its just missing.

wgentine, 2025-02-23:
Im running v3.0.1 by the way

Pho3niX90, 2025-02-24:
Perfect, will quickly add them in tonigbt

closed 2025-02-24 by Pho3niX90

Pho3niX90, 2025-02-24:
https://github.com/Pho3niX90/solis_modbus/releases/tag/3.0.2

wgentine, 2025-02-25:
Thats looking very good now! Thank you for the effort. I have just something that I also found out by using direct modbus sensors.... sometimes there registers are allucinating and brings for a very short period of time a completely wrong data. Most of the time its a zero. For example:

[image]

The problem is that homeassistant understands that it went to 0 and back and then you suddently generated 5MWh in a second :D

wgentine, 2025-02-27:
@Pho3niX90 there's also another glitch:

[image]

This sensor is duplicated and one is zeroed with W unit, the one that is actually working has no correct device class thus missing unit.

Pho3niX90, 2025-02-27:
Will quickly have a look today.

In regards to the total sensors, is this still happening? I assume this only happens on restart?

Pho3niX90, 2025-02-27:
https://github.com/Pho3niX90/solis_modbus/releases/tag/3.0.8

wgentine, 2025-02-28:
Thanks. And let me check in a few days if this is still happening. For now its ok.

AaronHoover, 2025-05-24:
Hi, Has anyone ever been able to set Actual value of limited Active power? I can read it but not write it

Snetselaar, 2025-05-28:
hi, i also have problems with the limeted active power function.
I have a Solis-1P3.6K-4G. 
I can read everything using the configuration for the S6 inverter. 
But i cant change limited active power

AaronHoover, 2025-05-28:
Hi, I literally just got mine working now. Write your percentage x 100 (100%=10000) to address 3051 using FC 6. Hope that works for you too.

Snetselaar, 2025-05-28:
Hi thanks for the fast response. 
Im am verry new to all of this so in need a app that works without me learning about al the coding things.

So I installed SolisCloud sensor integration and the have just launched a new beta version thats wokes verry wel for me.
https://github.com/hultenvp/solis-sensor.

I have ful control over the Solis Power limit setting and the Solis System Export Power Limit Value (het watts)

AaronHoover, 2025-05-28:
Hi, Add this to your configuration.yaml , one thing that you will need to change is the ip address, Check if active power looks rigt the try adjusting solis power limitation

automation:
  - alias: Set Solis Power Limitation Manually
    trigger:
      - platform: state
        entity_id: input_number.solis_power_limit
    action:
      - service: modbus.write_register
        data:
          hub: Solis_modbus
          unit: 1
          address: 3051
          value: "{{ states('input_number.solis_power_limit') | int * 100}}"
    mode: single

input_number:
  solis_power_limit:
    name: Solis Power Limitation
    min: 0
    max: 110
    step: 1
    unit_of_measurement: "%"

      
modbus:
  - name: Solis_modbus
    type: tcp
    host: 192.168.0.35
    port: 502
    sensors:
      - name: Solis Active Power
        slave: 1
        address: 3004
        input_type: input
        unit_of_measurement: kW
        device_class: power
        state_class: measurement
        scale: 0.001
        offset: 0
        precision: 1
        data_type: int32
        unique_id: solis_active_power
        
      - name: Solis Total Energy
        slave: 1
        address: 3008
        input_type: input
        unit_of_measurement: kWh
        device_class: energy
        state_class: total_increasing
        scale: 1
        offset: 0
        precision: 1
        data_type: uint32
        unique_id: solis_total_energy

commit bbadad20 (2025-02-21): closes #84
commit 07d651e6 (2025-02-21): closes #84 #58
commit cbc20a80 (2025-02-24): feat: more string sensors, fixes #84