Pho3niX90/solis_modbus #88: Inverter request - S6-GR1P1.5K-M

matt0020190, 2024-10-30:
**Is your feature request related to a problem? Please describe.**
I've added my S6 inverter in the title to Home Assistant using the integration and the S2-WT datalogger. The device is created but all entity values return 0.

**Describe the solution you'd like**
My inverter supported or advice on how to debug this issue.

**Describe alternatives you've considered**
I've tried SolaX integration and cannot get the inverter even registered.
I've deleted my datalogger from Solis Cloud as I understand only one Modbus connection is allowed without extra hardware.
I've checked the datalogger and remote server A shows as unknown status whereas remote server B is connnected. I added the MQTT server information also to try connect via Mosquito broker in HA. No joy as nothing is found by HA.

**Additional context**
Please see image of my inverter data. Let me know what else is needed
[image]

Pho3niX90, 2024-10-31:
Can you try the version `string-04`

matt0020190, 2024-10-31:
Hi, thanks for your advice. I've redownloaded that option and still the same sadly, all 0 values

matt0020190, 2024-11-04:
I tried string-02 also which does populate a few entities, but i have to say they are wild. For example total power was reporting 18,451W

Pho3niX90, 2024-11-06:
Unfortunately the documentation I have found regarding the string inverters are unreliable to blindly code it. I would either need access to a system where I can do testing, or  someone can continue from https://github.com/Pho3niX90/solis_modbus/pull/86 to make a PR

Pho3niX90, 2025-02-07:
related to #58

closed 2025-02-07 by Pho3niX90