Pho3niX90/solis_modbus #89: It seems not to work with a Solis S5-EH1(3-6)K-L

danielTinkiel, 2024-11-28:
**Describe the bug**I did not find any bug, but I have few questions: I have configured the address and left the port in the default port value. I heve a Solis S5-EH1(3-6)K-L, singlr phase, 5KW, with WECO batteries bank of 3x5.3KWh. Is this implementation tested on S5 series? Also, in my case I will also have some of my family members using the Solis App (via cloud) simultaneously.  Can both system coexist? And finally, what is the polling frequency? Can it be adjusted? 
Many thanks.
Daniel

A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: [e.g. iOS]
 - Browser [e.g. chrome, safari]
 - Version [e.g. 22]

**Smartphone (please complete the following information):**
 - Device: [e.g. iPhone6]
 - OS: [e.g. iOS8.1]
 - Browser [e.g. stock browser, safari]
 - Version [e.g. 22]

**Additional context**
Add any other context about the problem here.

wawyed, 2024-11-28:
I have an s5 and it works fine

danielTinkiel, 2024-11-28:
Thanks a lot for your quick response. The only point pending is about the possibility of concurrent use of the Solis App and your great integration. Will the APP still work if I monitor and operate my Solis Hybrid inverter through Home Assistant while other family member monitor the system via the APP?

Thanks again.
Kind regards,
Daniel

Daniel Tinkiel
10 King Edward Park,
Bray, Co. Wicklow, A98EC52
Ireland
+353 86 382 3929

0rangutan, 2024-12-01:
custom_components/solis_modbus/const.py

Change "POLL_INTERVAL_SECONDS" from 15 to your preference.

Ander3488, 2024-12-05:
Hi, I have a Solis Hybrid S5-EH1P6K-L and just had the latest firmware installed last night ( Inverter Version
4C004F-650000 ), which now has six charge slot times with SOC and Charge current in each slot along with six discharge slots also.

Pho3niX90, 2025-01-31:
@Ander3488 it will form part of #94, just need to get the registrar information.

closed 2025-01-31 by Pho3niX90