Pho3niX90/solis_modbus #90: Can't use slots in automation

wawyed, 2024-11-28:
I'm trying to initiate a trigger for my automation based on when the current time matches a charging /discharging slot. However when I try to find the time entity it doesn't show on the list of available entities to compare the current time to.

Have I done something wrong on my end or should this work?

Pho3niX90, 2024-12-07:
Hi

Is this still an issue? 

Can you confirm what the value is of your entity?

wawyed, 2024-12-07:
Looks like it the new version of HA fixes it! :)

wawyed, 2024-12-07:
Claimed victory too soon

>Message malformed: Expected HH:MM, HH:MM:SS, an Entity ID with domain 'input_datetime' or 'sensor', a combination of a timestamp sensor entity and an offset, or Limited Template @ data['at'][0] 

[image]

Pho3niX90, 2025-02-20:
@wawyed https://github.com/Pho3niX90/solis_modbus/releases/tag/3.0.0-beta-1

might be a solution for you, see #137

Pho3niX90, 2025-02-23:
this is resolved in v3

closed 2025-02-23 by Pho3niX90