Pho3niX90/solis_modbus #93: Charge/discharge battery in time slots does not work

debietim, 2024-12-22:
Hi All

I've replaced my Solis S5-RHI3P10K with a brand new S6-EH3P20K-H Hybrid inverter. I've added it to the modbus integration and all parameters coming in correctly. But when I want to charge the battery from the grid using "Time OF Use" is does not work and after about 30/40 seconds the "Time Of Use"-switch turns of again.

josu98, 2024-12-30:
I also have a S6-EH3P Hybrid inverter and noticed the same problem.

debietim, 2024-12-30:
In the meantime I figured out what the possible solution could be. Since this type of inverter, you can set an charge or discharge current for every time block specific and also a targeted SOC (It stops charging/discharging when that state of charge is reached within the set timeframe).
I think because we do not have these parameters in the modbus integration it won't work...

See picture of the settings on my device.
[image]

debietim, 2025-01-10:
I received the complete list of midnight addresses compatible for most of the new S6-EH… series inverters… looking for help to implement…

Pho3niX90, 2025-01-30:
> I received the complete list of midnight addresses compatible for most of the new S6-EH… series inverters… looking for help to implement…

You can post them here, and I can have a look

debietim, 2025-01-31:
> > I received the complete list of midnight addresses compatible for most of the new S6-EH… series inverters… looking for help to implement…
> 
> You can post them here, and I can have a look

See below, one for the S6 hybrid inverters who has now the option to set a charge/discharge AMP value / time slot and also a targeted SOC per time slot.

Next to that also the info about the regular string inverters...

[RS485_MODBUS Solis PV Grid-Connected Inverter.pdf](https://github.com/user-attachments/files/18619173/RS485_MODBUS.Solis.PV.Grid-Connected.Inverter.pdf)
[RS485_MODBUS(ESINV-33000ID) Hybrid Inverter.pdf](https://github.com/user-attachments/files/18619174/RS485_MODBUS.ESINV-33000ID.Hybrid.Inverter.pdf)

Pho3niX90, 2025-02-04:
> See below, one for the S6 hybrid inverters who has now the option to set a charge/discharge AMP value / time slot and also a targeted SOC per time slot.
>
> Next to that also the info about the regular [...]

Thank you for this, solis did send me the docs within the week, however, the ones you provided are newer for some reason. So will compare

Pho3niX90, 2025-02-04:
Please try https://github.com/Pho3niX90/solis_modbus/releases/tag/1.5.7-pre

debietim, 2025-02-05:
Hi! I am testing with 1.6.0 since I need the currents per time slot and the desired SOC per time slot.
I now see that you implemented the current per time slot as a value in %, this while it should be a value in Amps between 1 and 50.

Example: sensor.solis_s6_solis_grid_time_of_use_charge_battery_current_slot_1 is 0,8% while in the inverter it is 8 amps ...

[image]

Pho3niX90, 2025-02-06:
> Hi! I am testing with 1.6.0 since I need the currents per time slot and the desired SOC per time slot. I now see that you implemented the current per time slot as a value in %, this while it should be a value in Amps between 1 and 50.
> 
> Example: sensor.solis_s6_solis_grid_time_of_use_charge_battery_current_slot_1 is 0,8% while in the inverter it is 8 amps ...
> 
> <img alt="Image" width="291" src="https://private-user-images.githubusercontent.com/106590109/410157691-32cb6a72-72d9-4886-989b-2bcbd2a6e479.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3Mzg4NDMzNTksIm5iZiI6MTczODg0MzA1OSwicGF0aCI6Ii8xMDY1OTAxMDkvNDEwMTU3NjkxLTMyY2I2YTcyLTcyZDktNDg4Ni05ODliLTJiY2JkMmE2ZTQ3OS5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjUwMjA2JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI1MDIwNlQxMTU3MzlaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT0yMDJkZTk3ZTJkMDNjZDljMzg4ZGJiMzM2NTllMDliZTg0ZWI5YjdiZGQ2OGVlNjhjY2VjMDFiMmM2OTMzMmI2JlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCJ9.Q3H_D38B3aek5NH-uNMIK58YTJ3vtR9liki2SguuE0o">
> These values, together with the target SOC are not in the top list of "changeable" items, where the timeslots are...

Fixed in https://github.com/Pho3niX90/solis_modbus/releases/tag/1.6.0-pre3

Pho3niX90, 2025-02-06:
> I still see these values in the "non editable list" and as a percentage instead of a value between 1 and 50 amps... 8 amps = 0,8%... what is causing a rejection of the value by the inverter...

Should now be resolved in https://github.com/Pho3niX90/solis_modbus/releases/tag/1.6.0-pre4

debietim, 2025-02-06:
The values are now in Amp, button the inverter is the value 8 amp storded, while in the integration I now see 0,8 amps… the values should be multiplied by 10.

debietim, 2025-02-07:
Next to the previous:
Sensors with name: sensor.solis_s6_solis_grid_time_of_use_discharge_battery_current_slot_1 to 6 and sensor.solis_s6_solis_grid_time_of_use_discharge_cut_off_soc_slot_1 to 6 should be in the controls list.
For each time window in the charge/dicharge, it is now possible to set a individual amps and targeted SOC value

Pho3niX90, 2025-02-07:
> The values are now in Amp, button the inverter is the value 8 amp storded, while in the integration I now see 0,8 amps… the values should be multiplied by 10.

Does the cut off voltage have the same issue?

debietim, 2025-02-07:
No idea, since these values are 0 at my place...

But the charge/discharge current per time slot and desired SOC should be in the controls list to be able to modify them??

[image]

Pho3niX90, 2025-02-07:
This should resolve that https://github.com/Pho3niX90/solis_modbus/releases/tag/1.6.1

debietim, 2025-02-10:
When I want to change a charge or discharge current, even via the settings window, I receive a failure message...
I also can't select the individual time slots to be active like I can on the inverter...

[image]

Pho3niX90, 2025-02-11:
> When I want to change a charge or discharge current, even via the settings window, I receive a failure message... I also can't select the individual time slots to be active like I can on the inverter...
> 
> <img alt="Image" width="730" src="https://private-user-images.githubusercontent.com/106590109/411720770-75db8ef7-72a6-4284-ad5b-cbb48abea0fc.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzkyNjEzMzgsIm5iZiI6MTczOTI2MTAzOCwicGF0aCI6Ii8xMDY1OTAxMDkvNDExNzIwNzcwLTc1ZGI4ZWY3LTcyYTYtNDI4NC1hZDViLWNiYjQ4YWJlYTBmYy5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjUwMjExJTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI1MDIxMVQwODAzNThaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT0zZGIyNWYxMGRmNjYzMTNjMmExY2M4ODhmODBlNzEwNjhjYzNiNWYyM2M5ZTAzMGYyN2FlYTdiMzFiYzUwY2ZlJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCJ9.0ngEQ3zXLMlbwQhlez3WuG60omJDd_5llBHYaztZDJs">

This should fix that error https://github.com/Pho3niX90/solis_modbus/releases/tag/1.6.3-pre7

debietim, 2025-02-11:
I still can't see any charge/discharge switches for time based charging and discharing. I only see the buttons for generator charging periods...

Pho3niX90, 2025-02-11:
> I still can't see any charge/discharge switches for time based charging and discharing. I only see the buttons for generator charging periods...

I am not understand what you mean, what exactly is missing?

debietim, 2025-02-13:
The connection is falling away after 30/60 minutes. Values is HA are not updating anymore and the installation is back visible in Soliscloud...
When I reboot HA, everything works again for a little hour...

commit ff0c1dfe (2025-02-14): feat: add missing TOU Grid switches closes #93
PR #129 (2025-02-15): Add Time of Use grid charging/discharging support and improve Modbus performance

### **User description**
## 🔄 Related Issues
Closes #93

___

### **PR Type**
Enhancement, Bug fix

___

### **Description**
- Added support for "Time of Use" grid charging/discharging periods.

- Removed unnecessary locks to improve Modbus TCP performance.

- Enhanced error handling and logging for Modbus operations.

- Simplified connection management in `ModbusController`.

___

### **Changes walkthrough** 📝
<table><thead><tr><th></th><th align="left">Relevant files</th></tr></thead><tbody><tr><td><strong>Enha [... cut]
commit c5eaf819 (2025-02-15): Add Time of Use grid charging/discharging support and improve Modbus performance
 (#129)

* improve: remove locks to improve performance, as this is not needed for TCP
* feat: add missing TOU Grid switches closes #93
closed 2025-02-15 by Pho3niX90