Pho3niX90/solis_modbus #94: Support for new firmware version?

wawyed, 2024-12-24:
Has this integration updated to support the 6 charging/discharge slots with different SOC and Current for each on the latest solis firmware?

wawyed, 2025-01-07:
FYI I just updated to the new firmware and lost the ability to set the slots. So It's look it's not supported yet.

Pho3niX90, 2025-01-30:
Do you have a firmware version? And which inverter exactly has this?

wawyed, 2025-01-30:
Firmware Version: 4C0050-650000
Inverter: S5-EH1P5K-L

Pho3niX90, 2025-01-31:
Okay just trying to figure out if this is only applicable for the *-EH* versions, as all the requests have been for it. Will need to know the registrars and the logic behind it.

wawyed, 2025-01-31:
Maybe you can find some info here? https://github.com/wills106/homeassistant-solax-modbus/blob/main/custom_components/solax_modbus/plugin_solis_fb00.py

Pho3niX90, 2025-02-04:
Please try https://github.com/Pho3niX90/solis_modbus/releases/tag/1.5.7-pre

wawyed, 2025-02-04:
Hi, just tried im getting all 0. 

[image]

Pho3niX90, 2025-02-04:
> Hi, just tried im getting all 0.
> 
> [image]

Are you looking for "Time Charge" or "Time of Use"?, having a look at this again, it might be "Time of Use" that you are expecting? Busy implementing this now and will let you know once ready for testing

Pho3niX90, 2025-02-04:
@wawyed here is the one with time of use as well https://github.com/Pho3niX90/solis_modbus/releases/tag/1.6.0-pre

wawyed, 2025-02-04:
I'm not getting any data coming through

Pho3niX90, 2025-02-04:
> I'm not getting any data coming through

What do you mean by this? Is nothing coming through anymore? Is the `Solis Grid Time *` sensors not giving any data?

If the former, do you perhaps have logs for me?

wawyed, 2025-02-04:
None of the sensors are giving any data

Pho3niX90, 2025-02-04:
> None of the sensors are giving any data

This should still yield the same results as the previous, only exception being additional sensors. 

Can you see anything in the logs?

wawyed, 2025-02-04:
Line   6: 2025-02-04 21:22:17.237 WARNING (SyncWorker_0) [homeassistant.loader] We found a custom integration solis which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you experience issues with Home Assistant
	Line   7: 2025-02-04 21:22:17.239 WARNING (SyncWorker_0) [homeassistant.loader] We found a custom integration solis_modbus which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you experience issues with Home Assistant
	Line  35: 2025-02-04 21:22:31.342 DEBUG (MainThread) [custom_components.solis_modbus.modbus_controller] connecting
	Line  36: 2025-02-04 21:22:31.466 DEBUG (MainThread) [custom_components.solis_modbus] config entry host = 192.168.50.65, post = 502
	Line  39: 2025-02-04 21:22:31.486 INFO (SyncWorker_4) [custom_components.solax_modbus] trying to load plugin - plugin_name: solis_fb00
	Line  68: 2025-02-04 21:22:35.153 ERROR (MainThread) [homeassistant.components.sensor] Platform solis_modbus does not generate unique IDs. ID solis_modbus_192.168.50.65_solis_modbus_inverter_tou_charge_cutoff_discharge_2 already exists - ignoring sensor.solis_s6_solis_grid_time_of_use_discharge_cut_off_soc_slot_2
	Line  69: 2025-02-04 21:22:35.154 ERROR (MainThread) [homeassistant.components.sensor] Platform solis_modbus does not generate unique IDs. ID solis_modbus_192.168.50.65_solis_modbus_inverter_tou_discharge_v2_battery_current_2 already exists - ignoring sensor.solis_s6_solis_grid_time_of_use_discharge_battery_current_slot_2
	Line  70: 2025-02-04 21:22:35.155 ERROR (MainThread) [homeassistant.components.sensor] Platform solis_modbus does not generate unique IDs. ID solis_modbus_192.168.50.65_solis_modbus_inverter_tou_discharge_v2_cutoff_voltage_2 already exists - ignoring sensor.solis_s6_solis_grid_time_of_use_discharge_cut_off_voltage_slot_2
	Line  71: 2025-02-04 21:22:35.205 INFO (MainThread) [custom_components.solis_modbus.number] Options 0
	Line  72: 2025-02-04 21:22:35.205 INFO (MainThread) [custom_components.solis_modbus.number] Solis platform_config: {'host': '192.168.50.65', 'port': 502}
	Line  73: 2025-02-04 21:22:35.206 INFO (MainThread) [custom_components.solis_modbus.time] Options 0
	Line  74: 2025-02-04 21:22:35.207 INFO (MainThread) [custom_components.solis_modbus.time] Solis platform_config: {'host': '192.168.50.65', 'port': 502}
	Line  75: 2025-02-04 21:22:35.207 DEBUG (MainThread) [custom_components.solis_modbus.modbus_controller] Failed to read Modbus holding register: ModbusClientMixin.read_holding_registers() takes 2 positional arguments but 3 positional arguments (and 1 keyword-only argument) were given
	Line  76: 2025-02-04 21:22:35.208 DEBUG (MainThread) [custom_components.solis_modbus.time] async_added_to_hass Solis Time-Charging Charge Start (Slot 1),  time.solis_s6_solis_time_charging_charge_start_slot_1,  solis_modbus_192.168.50.65_43143
	Line  77: 2025-02-04 21:22:35.208 DEBUG (MainThread) [custom_components.solis_modbus.modbus_controller] Failed to read Modbus holding register: ModbusClientMixin.read_holding_registers() takes 2 positional arguments but 3 positional arguments (and 1 keyword-only argument) were given
	Line  78: 2025-02-04 21:22:35.208 DEBUG (MainThread) [custom_components.solis_modbus.time] async_added_to_hass Solis Time-Charging Charge End (Slot 1),  time.solis_s6_solis_time_charging_charge_end_slot_1,  solis_modbus_192.168.50.65_43145
	Line  79: 2025-02-04 21:22:35.209 DEBUG (MainThread) [custom_components.solis_modbus.modbus_controller] Failed to read Modbus holding register: ModbusClientMixin.read_holding_registers() takes 2 positional arguments but 3 positional arguments (and 1 keyword-only argument) were given
	Line  80: 2025-02-04 21:22:35.209 DEBUG (MainThread) [custom_components.solis_modbus.time] async_added_to_hass Solis Time-Charging Discharge Start (Slot 1),  time.solis_s6_solis_time_charging_discharge_start_slot_1,  solis_modbus_192.168.50.65_43147
	Line  81: 2025-02-04 21:22:35.209 DEBUG (MainThread) [custom_components.solis_modbus.modbus_controller] Failed to read Modbus holding register: ModbusClientMixin.read_holding_registers() takes 2 positional arguments but 3 positional arguments (and 1 keyword-only argument) were given
	Line  82: 2025-02-04 21:22:35.209 DEBUG (MainThread) [custom_components.solis_modbus.time] async_added_to_hass Solis Time-Charging Discharge End (Slot 1),  time.solis_s6_solis_time_charging_discharge_end_slot_1,  solis_modbus_192.168.50.65_43149
	Line  83: 2025-02-04 21:22:35.209 DEBUG (MainThread) [custom_components.solis_modbus.modbus_controller] Failed to read Modbus holding register: ModbusClientMixin.read_holding_registers() takes 2 positional arguments but 3 positional arguments (and 1 keyword-only argument) were given
	Line  84: 2025-02-04 21:22:35.210 DEBUG (MainThread) [custom_components.solis_modbus.time] async_added_to_hass Solis Time-Charging Charge Start (Slot 2),  time.solis_s6_solis_time_charging_charge_start_slot_2,  solis_modbus_192.168.50.65_43153
	Line  85: 2025-02-04 21:22:35.210 DEBUG (MainThread) [custom_components.solis_modbus.modbus_controller] Failed to read Modbus holding register: ModbusClientMixin.read_holding_registers() takes 2 positional arguments but 3 positional arguments (and 1 keyword-only argument) were given
	Line  86: 2025-02-04 21:22:35.210 DEBUG (SyncWorker_3) [custom_components.solis_modbus.switch] state change for 33132-0 from None to False
[... cut ...]
	Line 906: 2025-02-04 21:23:04.605 DEBUG (MainThread) [custom_components.solis_modbus.modbus_controller] Failed to read Modbus holding register: ModbusClientMixin.read_holding_registers() takes 2 positional arguments but 3 positional arguments (and 1 keyword-only argument) were given
	Line 907: 2025-02-04 21:23:04.605 DEBUG (MainThread) [custom_components.solis_modbus.modbus_controller] Failed to read Modbus holding register: ModbusClientMixin.read_holding_registers() takes 2 positional arguments but 3 positional arguments (and 1 keyword-only argument) were given

Pho3niX90, 2025-02-05:
@wawyed thank you. It seems you have solax installed as well, once you reload the solis one, solax took over the modbus connection. There can only be a single connection at a time via modbus.

wawyed, 2025-02-05:
Hmm. I used to have both working at the same time before I upgraded my firmware to the latest version

Pho3niX90, 2025-02-05:
> Hmm. I used to have both working at the same time before I upgraded my firmware to the latest version

I have made some additional changes, since there were some errors in your logs https://github.com/Pho3niX90/solis_modbus/releases/tag/1.6.0-pre2

wawyed, 2025-02-05:
> > Hmm. I used to have both working at the same time before I upgraded my firmware to the latest version
> 
> I have made some additional changes, since there were some errors in your logs https://github.com/Pho3niX90/solis_modbus/releases/tag/1.6.0-pre2

Looks like I'm getting data back now. There's voltage and current coming back as % on the slots.

[image]

[image]

Pho3niX90, 2025-02-06:
Sorry, the % coming back is a mistake on my end, I will fix the unit of measurement

Pho3niX90, 2025-02-06:
fixed sensors OUM https://github.com/Pho3niX90/solis_modbus/releases/tag/1.6.0-pre3

wawyed, 2025-02-06:
The time entities are showing as 0:00, the sensors for hours and minutes seem to work though

wawyed, 2025-02-13:
Is this supposed to work or not yet?

Pho3niX90, 2025-02-13:
> The time entities are showing as 0:00, the sensors for hours and minutes seem to work though

It is suppose to work. All things considered, the new registrars are added the same as the old ones. But, why they aren't working is difficult to say. 

I am not making use of them myself so cannot even test. 

If you are able to provide some logs, that will help

wawyed, 2025-02-13:
I'll send you some logs see if it hleps

wawyed, 2025-02-13:
Here's what I've found in the logs, hope it's useful

[home-assistant_solis_modbus_2025-02-13T12-07-19.933Z.log](https://github.com/user-attachments/files/18784008/home-assistant_solis_modbus_2025-02-13T12-07-19.933Z.log)

```Line     7: 2025-02-13 11:50:13.322 WARNING (SyncWorker_0) [homeassistant.loader] We found a custom integration solis_modbus which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you experience issues with Home Assistant
	Line    27: 2025-02-13 11:50:26.661 DEBUG (MainThread) [custom_components.solis_modbus] config entry host = 192.168.50.65, post = 502
	Line    32: 2025-02-13 11:50:31.249 INFO (MainThread) [custom_components.solis_modbus.number] Options 0
	Line    33: 2025-02-13 11:50:31.251 INFO (MainThread) [custom_components.solis_modbus.number] Solis platform_config: {'host': '192.168.50.65', 'port': 502}
	Line    34: 2025-02-13 11:50:31.255 INFO (MainThread) [custom_components.solis_modbus.time] Options 0
	Line    35: 2025-02-13 11:50:31.255 INFO (MainThread) [custom_components.solis_modbus.time] Solis platform_config: {'host': '192.168.50.65', 'port': 502}
	Line    36: 2025-02-13 11:50:31.257 DEBUG (MainThread) [custom_components.solis_modbus.time] Time: 43143, Hour = 0, Minute = 0
	Line    37: 2025-02-13 11:50:31.258 DEBUG (MainThread) [custom_components.solis_modbus.time] async_added_to_hass Solis Time-Charging Charge Start (Slot 1),  time.solis_s6_solis_time_charging_charge_start_slot_1,  solis_modbus_192.168.50.65_43143
	Line    38: 2025-02-13 11:50:31.258 DEBUG (MainThread) [custom_components.solis_modbus.time] Time: 43145, Hour = 0, Minute = 0
	Line    39: 2025-02-13 11:50:31.259 DEBUG (MainThread) [custom_components.solis_modbus.time] async_added_to_hass Solis Time-Charging Charge End (Slot 1),  time.solis_s6_solis_time_charging_charge_end_slot_1,  solis_modbus_192.168.50.65_43145
	Line    40: 2025-02-13 11:50:31.259 DEBUG (MainThread) [custom_components.solis_modbus.time] Time: 43147, Hour = 0, Minute = 0
	Line    41: 2025-02-13 11:50:31.259 DEBUG (MainThread) [custom_components.solis_modbus.time] async_added_to_hass Solis Time-Charging Discharge Start (Slot 1),  time.solis_s6_solis_time_charging_discharge_start_slot_1,  solis_modbus_192.168.50.65_43147
	Line    42: 2025-02-13 11:50:31.259 DEBUG (MainThread) [custom_components.solis_modbus.time] Time: 43149, Hour = 0, Minute = 0
	Line    43: 2025-02-13 11:50:31.259 DEBUG (MainThread) [custom_components.solis_modbus.time] async_added_to_hass Solis Time-Charging Discharge End (Slot 1),  time.solis_s6_solis_time_charging_discharge_end_slot_1,  solis_modbus_192.168.50.65_43149
	Line    44: 2025-02-13 11:50:31.260 DEBUG (MainThread) [custom_components.solis_modbus.time] Time: 43153, Hour = 0, Minute = 0
	Line    45: 2025-02-13 11:50:31.260 DEBUG (MainThread) [custom_components.solis_modbus.time] async_added_to_hass Solis Time-Charging Charge Start (Slot 2),  time.solis_s6_solis_time_charging_charge_start_slot_2,  solis_modbus_192.168.50.65_43153
	Line    46: 2025-02-13 11:50:31.260 DEBUG (MainThread) [custom_components.solis_modbus.time] Time: 43155, Hour = 0, Minute = 0
	Line    47: 2025-02-13 11:50:31.260 DEBUG (MainThread) [custom_components.solis_modbus.time] async_added_to_hass Solis Time-Charging Charge End (Slot 2),  time.solis_s6_solis_time_charging_charge_end_slot_2,  solis_modbus_192.168.50.65_43155
	Line    48: 2025-02-13 11:50:31.260 DEBUG (MainThread) [custom_components.solis_modbus.time] Time: 43157, Hour = 0, Minute = 0
	Line    49: 2025-02-13 11:50:31.261 DEBUG (MainThread) [custom_components.solis_modbus.time] async_added_to_hass Solis Time-Charging Discharge Start (Slot 2),  time.solis_s6_solis_time_charging_discharge_start_slot_2,  solis_modbus_192.168.50.65_43157
	Line    50: 2025-02-13 11:50:31.261 DEBUG (MainThread) [custom_components.solis_modbus.time] Time: 43159, Hour = 0, Minute = 0
	Line    51: 2025-02-13 11:50:31.261 DEBUG (MainThread) [custom_components.solis_modbus.time] async_added_to_hass Solis Time-Charging Discharge End (Slot 2),  time.solis_s6_solis_time_charging_discharge_end_slot_2,  solis_modbus_192.168.50.65_43159
	Line    52: 2025-02-13 11:50:31.261 DEBUG (MainThread) [custom_components.solis_modbus.time] Time: 43163, Hour = 0, Minute = 0
	Line    53: 2025-02-13 11:50:31.261 DEBUG (MainThread) [custom_components.solis_modbus.time] async_added_to_hass Solis Time-Charging Charge Start (Slot 3),  time.solis_s6_solis_time_charging_charge_start_slot_3,  solis_modbus_192.168.50.65_43163
	Line    54: 2025-02-13 11:50:31.262 DEBUG (MainThread) [custom_components.solis_modbus.time] Time: 43165, Hour = 0, Minute = 0
	Line    55: 2025-02-13 11:50:31.262 DEBUG (MainThread) [custom_components.solis_modbus.time] async_added_to_hass Solis Time-Charging Charge End (Slot 3),  time.solis_s6_solis_time_charging_charge_end_slot_3,  solis_modbus_192.168.50.65_43165
	Line    56: 2025-02-13 11:50:31.262 DEBUG (MainThread) [custom_components.solis_modbus.time] Time: 43167, Hour = 0, Minute = 0
	Line    57: 2025-02-13 11:50:31.262 DEBUG (MainThread) [custom_components.solis_modbus.time] async_added_to_hass Solis Time-Charging Discharge Start (Slot 3),  time.solis_s6_solis_time_charging_discharge_start_slot_3,  solis_modbus_192.168.50.65_43167
	Line    58: 2025-02-13 11:50:31.263 DEBUG (MainThread) [custom_components.solis_modbus.time] Time: 43169, Hour = 0, Minute = 0
[... cut ...]
	Line  1588: 2025-02-13 11:50:47.563 DEBUG (MainThread) [custom_components.solis_modbus.sensor] register_key = 43707, value = 7
	Line  1589: 2025-02-13 11:50:47.563 DEBUG (MainThread) [custom_components.solis_modbus.sensor] register_key = 43708, value = 90
	Line  1590: 2025-02-13 11:50:47.563 DEBUG (MainThread) [custom_components.solis_modbus.sensor] register_key = 43709, value = 600
	Line  1591: 2025-02-13 11:50:47.563 DEBUG (MainThread) [custom_components.solis_modbus.sensor] register_key = 43710, value = 495

0rangutan, 2025-02-13:
I am having similar issues with time slots for my 5G inverter after upgrading firmware to 6 slots.  Model S5-EH1P6K-L, version 4C0050-650000.

I note that in your logs, for example, slot 4 has the entity solis_s6_solis_time_charging_charge_start_slot_5 whereas Slot 5 is solis_s6_solis_time_charging_charge_start_slot_5_2

0rangutan, 2025-02-13:
In my case I can't tell whether something is incorrect in the integration or whether my entities are a mess from manual changes, the firmware update and/or and multiple updates to the integration.
I'm up to 405 entities now, which seems too high?

Pho3niX90, 2025-02-13:
related to #127

Pho3niX90, 2025-02-13:
> In my case I can't tell whether something is incorrect in the integration or whether my entities are a mess from manual changes, the firmware update and/or and multiple updates to the integration. I'm up to 405 entities now, which seems too high?

The entities have been increasing over the last couple of days. Some registrars split into 12 switches etc, so might seem high, but it's all OK.

Pho3niX90, 2025-02-13:
We can continue the time entities discussion over at #127 
From the logs that you provided, it states all time entities are returning 0? Might have been a very fresh log

closed 2025-02-13 by Pho3niX90