Pho3niX90/solis_modbus #95: Two inverter error

matteocaretta72, 2024-12-27:
When I configure two inverters on two different IP addresses in the integration one stops working, and the commands control the other inverter

Pho3niX90, 2025-01-30:
Is this for all commands, or specific ones? How are you triggering the commands? Via the provided buttons?

Pho3niX90, 2025-01-30:
@matteocaretta72 can you try https://github.com/Pho3niX90/solis_modbus/releases/tag/1.5.5

Pho3niX90, 2025-01-31:
Should be fixed in `1.5.6`

matteocaretta72, 2025-01-31:
Updated but it doesn't work.
When I add the first inverter it works well, by adding the second the first one no longer works, while the second one has data that always changes in a confusing way.
The second inverter is a zeroCO2 but still the same model.
Do you have any suggestions please

Pho3niX90, 2025-02-12:
> Updated but it doesn't work. When I add the first inverter it works well, by adding the second the first one no longer works, while the second one has data that always changes in a confusing way. The second inverter is a zeroCO2 but still the same model. Do you have any suggestions please

Could you perhaps provide some logs. 
Does each inverter at least have it's own sensors/entities?

closed 2025-02-27 by github-actions[bot]