Pho3niX90/solis_modbus #96: Cannot connect to Datalogger

morts1, 2025-01-08:
I have swapped my datalogger for a S2-WL-ST as I had one of the S3 loggers when the inverter was installed.
I have configured the datalogger and it is connected to my WiFi and I can connect to it via a browser.

I am running HA Core 2025.1.0

When I attempt to add the integration and fill in the IP address and port (502) I get a generic "Unknown error occurred" in the GUI

I can see the following in the HA logs:

Logger: aiohttp.server
Source: /srv/homeassistant/lib/python3.12/site-packages/aiohttp/web_protocol.py:451
First occurred: 13:49:30 (2 occurrences)
Last logged: 13:52:44

Error handling request
Traceback (most recent call last):
  File "/home/homeassistant/.homeassistant/custom_components/solis_modbus/modbus_controller.py", line 41, in async_read_input_register
    result = await self.client.read_input_registers(register, count, slave=1)
                   ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
TypeError: ModbusClientMixin.read_input_registers() takes 2 positional arguments but 3 positional arguments (and 1 keyword-only argument) were given

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/srv/homeassistant/lib/python3.12/site-packages/aiohttp/web_protocol.py", line 480, in _handle_request
    resp = await request_handler(request)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/srv/homeassistant/lib/python3.12/site-packages/aiohttp/web_app.py", line 569, in _handle
    return await handler(request)
           ^^^^^^^^^^^^^^^^^^^^^^
  File "/srv/homeassistant/lib/python3.12/site-packages/aiohttp/web_middlewares.py", line 117, in impl
    return await handler(request)
           ^^^^^^^^^^^^^^^^^^^^^^
  File "/srv/homeassistant/lib/python3.12/site-packages/homeassistant/components/http/security_filter.py", line 92, in security_filter_middleware
    return await handler(request)
           ^^^^^^^^^^^^^^^^^^^^^^
  File "/srv/homeassistant/lib/python3.12/site-packages/homeassistant/components/http/forwarded.py", line 83, in forwarded_middleware
    return await handler(request)
           ^^^^^^^^^^^^^^^^^^^^^^
  File "/srv/homeassistant/lib/python3.12/site-packages/homeassistant/components/http/request_context.py", line 26, in request_context_middleware
    return await handler(request)
           ^^^^^^^^^^^^^^^^^^^^^^
  File "/srv/homeassistant/lib/python3.12/site-packages/homeassistant/components/http/ban.py", line 86, in ban_middleware
    return await handler(request)
           ^^^^^^^^^^^^^^^^^^^^^^
  File "/srv/homeassistant/lib/python3.12/site-packages/homeassistant/components/http/auth.py", line 242, in auth_middleware
    return await handler(request)
           ^^^^^^^^^^^^^^^^^^^^^^
  File "/srv/homeassistant/lib/python3.12/site-packages/homeassistant/components/http/headers.py", line 32, in headers_middleware
    response = await handler(request)
               ^^^^^^^^^^^^^^^^^^^^^^
  File "/srv/homeassistant/lib/python3.12/site-packages/homeassistant/helpers/http.py", line 73, in handle
    result = await handler(request, **request.match_info)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/srv/homeassistant/lib/python3.12/site-packages/homeassistant/components/http/decorators.py", line 81, in with_admin
    return await func(self, request, *args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/srv/homeassistant/lib/python3.12/site-packages/homeassistant/components/config/config_entries.py", line 222, in post
    return await super().post(request, flow_id)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/srv/homeassistant/lib/python3.12/site-packages/homeassistant/components/http/data_validator.py", line 74, in wrapper
    return await method(view, request, data, *args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/srv/homeassistant/lib/python3.12/site-packages/homeassistant/helpers/data_entry_flow.py", line 122, in post
    result = await self._flow_mgr.async_configure(flow_id, data)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/srv/homeassistant/lib/python3.12/site-packages/homeassistant/data_entry_flow.py", line 341, in async_configure
    result = await self._async_configure(flow_id, user_input)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/srv/homeassistant/lib/python3.12/site-packages/homeassistant/data_entry_flow.py", line 388, in _async_configure
    result = await self._async_handle_step(
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/srv/homeassistant/lib/python3.12/site-packages/homeassistant/data_entry_flow.py", line 491, in _async_handle_step
    result: _FlowResultT = await getattr(flow, method)(user_input)
                           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/homeassistant/.homeassistant/custom_components/solis_modbus/config_flow.py", line 17, in async_step_user
    if await self._validate_config(user_input):
       ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/homeassistant/.homeassistant/custom_components/solis_modbus/config_flow.py", line 34, in _validate_config
    await modbus_controller.async_read_input_register(33093)
  File "/home/homeassistant/.homeassistant/custom_components/solis_modbus/modbus_controller.py", line 45, in async_read_input_register
    raise _LOGGER.error(f"Failed to read Modbus register: {str(e)}")
TypeError: exceptions must derive from BaseException

This error originated from a custom integration.

Logger: custom_components.solis_modbus.modbus_controller
Source: custom_components/solis_modbus/modbus_controller.py:45
integration: Solis Modbus (documentation, issues)
First occurred: 13:49:30 (2 occurrences)
Last logged: 13:52:44

Failed to read Modbus register: ModbusClientMixin.read_input_registers() takes 2 positional arguments but 3 positional arguments (and 1 keyword-only argument) were given

cddevlin6, 2025-01-12:
I am also experiencing the same problem when adding a new data logger stick.

DenisTimofijuk, 2025-01-29:
I also get the same error. My datalogger 7A123A1609300A62 does not seem to have the required 502 port for Modbus, or it is not enabled by default.

Pho3niX90, 2025-01-30:
What inverter is it coupled with?

DenisTimofijuk, 2025-01-30:
Mine datalogger is coupled with S6-EH3P10K-H inverter.

Pho3niX90, 2025-01-30:
Might be related to #58

morts1, 2025-02-01:
I've just tried version 1.5.6 and it all seems to be working, thank you so much!

DenisTimofijuk, 2025-02-01:
I also tried version 1.5.6. 
Now it was able to connect to my inverter, but all entities have '0' or 'unavailable' values... I have restarted HA several times but nothing has changed. If this integration was able to connect to my inverter, how do I know it is compatible? 0 values now I see, is this a bug of this integration or simply incompatibility with my hardware?

From checking debug logs it looks like it is unable to connect, although integration was successfully installed to my HA.

Attaching log file:

[home-assistant_2025-02-01T14-54-50.232Z.log](https://github.com/user-attachments/files/18629174/home-assistant_2025-02-01T14-54-50.232Z.log)

Pho3niX90, 2025-02-04:
> I also tried version 1.5.6. Now I was able to install this integration, it did not showed any errors during installation, but all entities have '0' or 'unavailable' values... I have restarted HA several times but nothing has changed.
> 
> From checking debug logs it looks like it is unable to connect, although integration was successfully installed to my HA.
> 
> 192.168.1.101 is IP of my inverter.
> 
> Attaching log file:
> 
> [home-assistant_2025-02-01T14-54-50.232Z.log](https://github.com/user-attachments/files/18629174/home-assistant_2025-02-01T14-54-50.232Z.log)

Can you try 1.5.4 to see if it works? It's an older version

PR #102 (2025-02-06): implements #93 #94
### **User description**
implements #93 #94
fixes #96 #103  #104 

___

### **PR Type**
Enhancement

___

### **Description**
- Added support for up to 6 charging/discharge slots in the Solis Modbus integration.

- Introduced new sensor definitions for additional time slots in `hybrid_sensors.py`.

- Updated `time.py` to include new time slot configurations.

- Incremented integration version to `1.5.7` in `manifest.json`.

___

### **Changes walkthrough** 📝
<table><thead><tr><th></th><th align="left">Relevant files</th></tr></thead><tbody><tr><td><strong>Enhancement</strong [... cut]
closed 2025-02-06 by Pho3niX90