Pho3niX90/solis_modbus #97: Solis S6 Solis Overdischarge SOC won't update

miketarg, 2025-01-30:
**Describe the bug**
I cannot get the Solis S6 Solis Overdischarge SOC to change
**To Reproduce**
Steps to reproduce the behavior:
1. Go to inverter controls
2. Click on Solis S6 Solis Overdischarge SOC
3. Scroll change the value (15) is what want default is 20 I input 15 it allows me to change it I click away.
4. After seval moments it reverts back to 20 

**Expected behavior**
Allow user input percentage

miketarg, 2025-02-04:
@Pho3niX90 Was there anything else you need?

miketarg, 2025-02-06:
The same thing happens with the switch.solis_s6_solis_self_use_mode as well you toggle it and it goes off after about 15 secs it reverts to previous state.

fraserallanuk, 2025-02-06:
I also noticed this with number.solis_backup_soc and number.solis_overcharge_soc

Pho3niX90, 2025-02-06:
I will revisit them to make sure. 

Which inverter models are you using?

fraserallanuk, 2025-02-06:
RHI-3K-48ES-5G for me

fraserallanuk, 2025-02-06:
I think it might be the inverter doing it as when I disable the HA Solis_Modbus integration, and then update the inverter values from the Solis Cloud app, they don't update!  I also tested using the Solax Modbus integration and got similar results where the values revert back. 

I will check with another inverter and report back but I'm not convinced the issue is with Solis_Modbus. I will also physically check/adjust the valves on the inverter itself too.

Maybe @miketarg could repeat my tests to see if he has a similar outcome

fraserallanuk, 2025-02-06:
Just to add, I don't seem to have this issue with 'Solis S6 Solis Overdischarge SOC' or 'Solis S6 Solis Force Charge SOC'

My issue is with: 'Solis S6 Solis Overcharge SOC' and 'Solis S6 Solis Backup SOC'

miketarg, 2025-02-06:
I dont know if it makes any difference or helps, but my inverter only talks to Home Assistant via modbus, whne its doing this solis cloud doesnt work.

Pho3niX90, 2025-02-06:
> I dont know if it makes any difference or helps, but my inverter only talks to Home Assistant via modbus, whne its doing this solis cloud doesnt work.

yeah this is normal, you would have to disconnect the modbus for the cloud to work

closed 2025-02-12 by Pho3niX90

miketarg, 2025-02-16:
What was the outcome of this issue, is it a problem with the inverter?