Pho3niX90/solis_modbus #99: 1.5.5 broken for my setup

0rangutan, 2025-01-30:
1.5.4 works fine, the 1.5.5 release won't start up.
Error log:

2025-01-30 15:58:55.698 ERROR (MainThread) [homeassistant.config_entries] Error setting up entry Solis: 192.168.0.58 for solis_modbus
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/config_entries.py", line 640, in __async_setup_with_context
    result = await component.async_setup_entry(hass, self)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/solis_modbus/__init__.py", line 56, in async_setup_entry
    hass.data[DOMAIN][CONTROLLER][host] = controller
    ~~~~~~~~~~~~~~~~~^^^^^^^^^^^^
KeyError: 'modbus_controller'

Pho3niX90, 2025-01-31:
1.5.5 was meant as a prerelease. Will update a fix asap.

Pho3niX90, 2025-01-31:
Install1.5.6, which should fix the issue

0rangutan, 2025-01-31:
Yes, fixed in 1.5.6 - many thanks

closed 2025-01-31 by 0rangutan