Pho3niX90/solis_modbus #106: Compatibility with new Solis S6-EH3P(12-20)K-H / Axitec AXIhycon 15H

hbpv, 2025-02-06:
Hey guys,

we just like to share, that we successful testet the integration with the new Axitec Energy AXIhycon 15H wich is OEM based on Solis S6-EH3P(12-20)K-H. Maybe this helps some of you and could be added to the docs as tested. Inverter is connected via LAN using the LAN/Wifi-Datalogger Stick.

Only issue i see is, that using modbus, this breaks Soliscloud/Axicloud App :-/. Seems to be normal as i already read this on a Solis document.

Axitec: https://axitecsolar.com/axihycon-12h-20h
Solis: https://www.solisinverters.com/de/energy_storage_inverters18/S6-EH3P(12-20)K-H_de.html

Best,
Max

debietim, 2025-02-10:
Were you able to charge and discharge the connected battery via HA commands? 
If yes, can you provide some more info because I got a lot of issues at this time...

hbpv, 2025-02-11:
> Were you able to charge and discharge the connected battery via HA commands? If yes, can you provide some more info because I got a lot of issues at this time...

Hi

there is no force-charge button but a force charge power limit button - i don't know if this is a feature or a bug (missing regs), maybe there is no modbus reg or feature for start/stop "force charge" from Solis... @Pho3niX90 ? Could you may check this?

The way i did it, for example to fully charge to 100% once:

1. Set solis_backup_soc to 100
2. Allow charging from Grid
3. Activate "Reserve Battery Mode"
4. It will then charge with max power from grid until 100%. 
5. Set solis_backup_soc back to what it was before
6. Stop Reserve Battery Mode
7. (Self-Used mode in that time can stay activated as it seems to be overwritten by the Reserve Battery Mode)

@Pho3niX90 There is another problem of understanding: When using other Inverters i.e. from Goodwe, you have a Grid-DOD (Depth of Discharge Max) and Backup-DOD - so you can set:

In Grid Mode Grid overdischarge/DOD = 80% - means: Do not not discharge under 20% in grid-connected scenario.
In offgrid-scenario (backup when a grid outage happens): Backup overdischarge/DOD = 10% > Inverter uses 10% more from battery in case 20% DOD is reached in the moment the grid is lost

But for Solis, there is only a "overdischarge SOC" - is that used to be for Grid AND Backup ?!
I was wondering because i thought "solis_backup_soc" could be the off-grid(outage) SOC, but you cannot set it i.e. to 10%, it always jumps back because allowed values (see on soliscloud) are 15 to 100. So i think "reserve battery" is not used to be the offgrid backup soc but something else.. dont know for what reason. What is a scenraio for the "reserve battery"?

Best
Max

Pho3niX90, 2025-02-11:
Afaik, `solis_backup_soc` is to make sure that your batteries are always charged to this %, regardless of your settings. And is used when there is no grid or pv power being supplied, then only will the batteries be used.

If you are in a grid scenario, and and charging the battery from the grid, I would not suggest to dump the battery back into the grid. This will deplete your battery cycles, and you are at a loss with the conversion losses from AC->DC and DC->AC.

In my scenario, I do not allow charging the batteries from the grid, and do not allow the batteries to dump back into the grid. They are there purely for backup when the grid goes down. PV pushes back to the grid, when the batteries are full. 

Now `Solis Backup SOC ` needs to be set higher than `Solis Force Charge SOC`. From the documentation, there doesn't seem to be a minimum for the `Solis Backup SOC `, but `Solis Force Charge SOC` is set default to 10%

Pho3niX90, 2025-02-11:
for the following:
`In Grid Mode Grid overdischarge/DOD = 80% - means: Do not not discharge under 20% in grid-connected scenario.`
I would think, you need to have `Reserve Battery Mode` enabled, and set `Solis Backup SOC` to 20%

`In offgrid-scenario (backup when a grid outage happens): Backup overdischarge/DOD = 10% > Inverter uses 10% more from battery in case 20% DOD is reached in the moment the grid is lost`
same as above, but `Solis Force Charge SOC` set to 10%.

I have not tested the above myself, but to my understanding, this would work

Pho3niX90, 2025-02-11:
This includes the off-grid overdischarge soc:
https://github.com/Pho3niX90/solis_modbus/releases/tag/1.6.3-pre7

hbpv, 2025-02-11:
Hey thanks for the quick reply.

Okay, good idea: When **un**set "allow charging from grid" and battery SOC goes under Force-Charge-SOC (i.e. loss of voltage/draining because not loaded for 1-2 weeks), it should save the battery anyways with power from grid regardles of if allow charging from grid is on/off - when not, this is not an option regarding battery-safety.

As you can see in the screenshot from soliscloud, you cannot set the reserve battery under 20% > In this Screenshot it's @80% the same value in solid_modbus is the entity "number.solis_s6_solis_backup_soc". If you think, this is the value for the SOC in grid-outage to use by the backup, it would sugguest, that the "normal", gridscenario SOC must always be above that this all makes sense.

[image]

**Maybe i could explain the Scenario we like to have again in detail**, so find a solution with the avaibale regs:

- Inventer is a hybrid scenario with grid-connected use and serving some houseloads via backup port
- Normal Grid Mode SOC should be: 20% (80% max discharge)
- Grid-Outage-Backup SOC should be 11% (89% max discharge)
- Forcecharge @10%

As you write, in 1.6.3-pre7 i can now set an overdischarge backup soc, wich could make this possible?:

overdischarge-SOC = 20%
overdischarge-Backup-SOC = 11%
forcecharge-SOC = 10%

So when this is the solution, in my scenario i dont need set anything special for the options solis_backup_soc neither need the reserve-battery-mode?

hbpv, 2025-02-11:
Second, was asked by @debietim : Could you add a force-charge button (on/off), to start charging the battery manually? 

Should be register 43135.

Thanks!

Pho3niX90, 2025-02-11:
> Second, was asked by [@debietim](https://github.com/debietim) : Could you add a force-charge button (on/off), to start charging the battery manually?
> 
> Should be register 43135.
> 
> Thanks!

This is only to charge from PV:

[image]

Is this what you need?

Pho3niX90, 2025-02-11:
Just to be sure we aren't mixing SOC (state of charge) with max depletion

`Normal Grid Mode SOC should be: 20% (80% max discharge)`
Do you want your battery to only be charged to 20%?
`Grid-Outage-Backup SOC should be 11% (89% max discharge)`
Do you want your battery to only be charged to 11%

I believe you are perhaps looking to change:
`Off-Grid Overdischarge SOC`: 11%
`Overdischarge SOC`: 20%
`Backup SOC`: 80%

The above, should keep your battery charged to 80%, but will dump the charge into the grid when available, to 20%. If you do not want the battery to dump into the grid, then you need to enable reserve mode for the battery. 

For my system, it was rather set and forget. Since my SOC is always set to 100%, and the grid is not allowed to charge it, only PV. In my scenario, I only need the battery for when there is no grid power, to sustain blackouts etc. 

Perhaps someone with the same need can chime in to assist, as I am only going from the docs. Another option is to perhaps contact your local Solis, as they should be able to provide you with the correct settings. I do know Solis does have some weird setting combinations compared to other brands.

hbpv, 2025-02-11:
Sorry, not your or my fault in understanding, but some manufacturers have different names for the same thing.

My aim:

When the systems works in normal mode, i want a "DOD (Depth of Discharge)" of 20% / "maximum discharge to" 20% / or "Discharge max of" 80% - choose how you name it ;-)  >> Battery should be used betwenn 100 and 20% SOC.

When offgrid/outage, i want the battery to have a (Backup-)DOD of 11% or "maximum discharge to" 11% or "Discharge max" 81% (to have a reserve of 9% if Standard DOD ist reached during blackout) >> Battery will drain up to 11% in rest charging.

Scenario (wich is not out of space): My Battery stands at 20% SOC because this is set for overdischarge SOC. Weather ist bad, no PV. Then a grid outage happens > At the Moment, the backup would not be able to source the backup port. Therefore we need the backup-overdischarge, wich will be 11%, so das backup port can gibe out another 9%. This is how Goodwe uses Grid-DOD und Backup-DOD to have a reserve to the Standard Grid DOD.

Pho3niX90, 2025-02-11:
From that, then my guess would be the previous mentioned settings:
Off-Grid Overdischarge SOC: 11%
Overdischarge SOC: 20%

Perhaps, post a thread here: https://powerforum.co.za/forum/147-solis/
There are way more knowledgeable people than myself, that have used Solis and other brands, that might be able to give you solid advice.

hbpv, 2025-02-11:
> From that, then my guess would be the previous mentioned settings: Off-Grid Overdischarge SOC: 11% Overdischarge SOC: 20%
>
> Perhaps, post a thread here: https://powerforum.co.za/forum/147-solis/ [...]

Set this with new 1.6.3 and will try it the next days.
Please Tell how to tip you, for your fast responses and Great Work..!

closed 2025-02-19 by hbpv