Pho3niX90/solis_modbus #107: Added Sensors S6-EH1P8K-L-PRO

leonw1090, 2025-02-07:
Hi,

Thank you for the great integration! I want to contribute meaningfully rather than just request a feature.

I need to access the generator settings and statistics. I run completely off-grid and would like to make changes to the settings I normally see in the Solis local app.

My requirements:

Sensor: Wattage generated by the generator
Sensor: Voltage from the generator
Switch: Toggle to turn on the input on the inverter (marked as "Manual")

I’d love to contribute by working on this myself. How can I access the entire Modbus string returned by the Solis? If I can see the data, I can identify the values and reverse-engineer the necessary parameters.

Looking forward to your guidance!

Best regards,
Leon

Pho3niX90, 2025-02-07:
Hey Leon

So it's a bit more trickier than that. 

The Solis doesn't return a single string, but you have to request the specific data from it, then convert it. 

In the code, the sensors.py is what gets the data from the Solis, and the data directory contains the mappings. 

The attached document have the "registrars" that contain specific data. 

Now, for the 2 sensors you mention.

Are you looking for the voltage/wattage from the grid, pv, or just total passed via the solis? Or, do you have a diesel/petrol generator connected to the solis as a backup?

[RS485_MODBUS.ESINV-33000ID.Hybrid.Inverter.pdf](https://github.com/user-attachments/files/18705308/RS485_MODBUS.ESINV-33000ID.Hybrid.Inverter.pdf)

leonw1090, 2025-02-07:
Hi,

Thank you for the PDF.

I have a petrol generator connected to the Smart Load port, and I’d really like to access the following values:

33530 – Generator Phase A Active Power
33531 – Generator’s Daily Power Generation
33532 – Total Generator Output
I don’t see any register that directly shows the Generator/Smart Load Port Voltage, but I can experiment with that.

Additionally, I noticed a toggle in the app that allows the inverter to take power from the generator. I will take a screenshot this afternoon and share it. This feature would be useful to allow the generator some warm-up time before supplying power.

My goal is to monitor the generator’s output and create automations to prevent overcharging the batteries via the generator.

Regards

leonw1090, 2025-02-07:
Hi,

The data seems to be working as expected, but I’m struggling to get the switch to work—I don’t know how to code it.

Here are the relevant Modbus registers:

43340 – Toggles the generator from Auto to Manual mode.
In Auto mode, you set a start SOC and end SOC using 43345.
In Manual mode, you simply have an On/Off switch (Bit 01 of 43340).
43369 – Sets how much of the generator’s capacity can be used for charging (which could also be very useful).
I personally use Manual mode—Home Assistant starts the EcoFlow generator, and I then manually set the switch to "On."

If you have the capacity to code this into the release so that others can benefit from it, that would be amazing! I’m not even close to being a coder, but I’m happy to test everything and report back with results.

Regardless, thank you for the great integration—it’s incredibly useful to see updates every 5 seconds instead of every 5 minutes!

        "register_start": 33530,
        "scan_interval": 5,
        "entities": [
            {"type": "SS", "name": "Solis Generator Active Power",
             "unique": "solis_modbus_inverter_generator_phase_a_active power",
             "register": ['33530'], "device_class": SensorDeviceClass.POWER, "multiplier": 10,
             "unit_of_measurement": UnitOfPower.WATT, "state_class": SensorStateClass.MEASUREMENT},
            {"type": "SS", "name": "Solis Generator Power Today",
             "unique": "solis_modbus_inverter_generator_power_today",
             "register": ['33531'], "device_class": SensorDeviceClass.ENERGY, "multiplier": 0.1,
             "unit_of_measurement": UnitOfEnergy.KILO_WATT_HOUR, "state_class": SensorStateClass.TOTAL_INCREASING},

Regards
Leon

Pho3niX90, 2025-02-07:
Hey leon.

You can try out https://github.com/Pho3niX90/solis_modbus/releases/tag/1.6.3-pre1

leonw1090, 2025-02-08:
Hi,

All the sensors are working, but I’ve noticed that the entire Modbus system seems to be running much slower—polling every 2 minutes instead of 5 seconds. This could be due to the 380 sensors being queried.

The following controls do not work or may not be necessary in my opinion:

Solis S6 Generator Charging Power – Not supported (float error) → Useful to have
Solis S6 Generator Access Location – Not needed
Solis S6 Generator Connection Position – Not needed
Solis S6 Generator Enable Signal – Turns off and then on again → Useful to have
Solis S6 Generator Target Power – Not supported (float error) → Useful to have
Solis S6 Maximum Allowable Percentage of Generator – No idea what this changes, might not be needed
Solis S6 With Generator – Turns off and then on again → Useful to have

For generator control, most settings are "set and forget", but my main goal is to allow the generator to take power. This might correspond to either:

Solis S6 With Generator
Solis S6 Generator Enable Signal

In the Solis app, this is represented as the "Manual" toggle (On/Off).

Let me know if you need any testing on my side.

Have a great Saturday!

Pho3niX90, 2025-02-09:
This should fix the float issues. 
https://github.com/Pho3niX90/solis_modbus/releases/tag/1.6.3-pre4

Regarding the slow polling, where are you having a look how often values are polled? HA's UI can be a bit confusing on the `Last Updated` / `Last Changed`. Those values only change when the previous value doesn't match the current value.

Pho3niX90, 2025-02-09:
So, here is how I understand it:
Solis S6 Generator Charging Power – I assume this is the max power your generator can provide, in kwh
Solis S6 Generator Enable Signal – Turns off and then on again → Useful to have
Solis S6 Generator Target Power – Not supported (float error) → Useful to have
Solis S6 Maximum Allowable Percentage of Generator – I assume this limits overusage of the generator, to perhaps, have the sun provide more than the generator, in avoiding overusage/over reliance? probably needs `Charging Power` set to work correctly.
Solis S6 With Generator – Turns off and then on again → Useful to have

Here are some caveats with `Solis S6 With Generator`

[image]

leonw1090, 2025-02-09:
Hi!

Hope you’re having a great Sunday.

The float switches are working now!

Do you perhaps have a Discord channel? I’d like to send you a video.

Pho3niX90, 2025-02-09:
No discord channel perse, but you can PM me on discord `pho3nix90`

closed 2025-02-11 by Pho3niX90