Pho3niX90/solis_modbus #112: Updating to 2025.2.1 break modbus I have installed the pre release

miketarg, 2025-02-08:
1.6.3-pre3 is installed and I have e the same issue 

**Describe the bug**
This error originated from a custom integration.

Logger: custom_components.solis_modbus.modbus_controller
Source: custom_components/solis_modbus/modbus_controller.py:33
integration: Solis Modbus (documentation, issues)
First occurred: 4:30:40 PM (31 occurrences)
Last logged: 4:30:56 PM

Failed to connect to Modbus device. Will retry, failures = 77
Failed to connect to Modbus device. Will retry, failures = 78
Failed to connect to Modbus device. Will retry, failures = 79
Failed to connect to Modbus device. Will retry, failures = 80
Failed to connect to Modbus device. Will retry, failures = 81

This error originated from a custom integration.

Logger: pymodbus.logging
Source: custom_components/solis_modbus/modbus_controller.py:29
integration: Solis Modbus (documentation, issues)
First occurred: 4:30:09 PM (81 occurrences)
Last logged: 4:30:56 PM

Failed to connect [Errno 111] Connect call failed ('192.168.0.83', 502)

**Update to 2025.2.1**

**Expected behavior**
To connect 

**Screenshots**
See the attached photos to show the unavailable entities affected.

[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

Pho3niX90, 2025-02-09:
Can you attach the logs please.  From the screenshots, it seems it only took a while to connect, but it doesn't indicate any errors.

miketarg, 2025-02-09:
> Can you attach the logs please. From the screenshots, it seems it only took a while to connect, but it doesn't indicate any errors.

Unfortunately I rolled back as it's not an option not to have control over the Inverter. 

I can assure you that it didn't connect (I waited for approximately 40 mins and it just kept sending the above)

Pho3niX90, 2025-02-09:
What did you roll back to? There is currently a `pre6`, so the issue might have already been fixed.

closed 2025-02-09 by Pho3niX90

miketarg, 2025-02-09:
I'm on core  2025.2.0

And V 1.6.0 solis

miketarg, 2025-02-10:
@Pho3niX90 sorry to bother you but am I doing something wrong?  I'm not sure if I should update again as I really want to avoid connection issues.

[image]
[image]

These are the updates that caused me an issue.

Pho3niX90, 2025-02-10:
> .

Yes, you can, I am not having issues with either updates. It's always a possibility for the modbus to not connect directly after the first 5-10mins of HA restart.

But the latest HA is compatible with the latest SM