Pho3niX90/solis_modbus #115: Feature request: Start/Stop modbus connection/polling to be able to use soliscloud

hbpv, 2025-02-11:
Hi

it would be great if we could start/stop the modbus connection (or polling) by a button in the ha-integration 

Almost all users have the problem, that when modbus is polled, the soliscloud is killed. For manufacturer service / updates etc. via soliscloud, it would be cool to be able to stop polling instead of killing the whole inetgration in ha.

Best,
Max

Pho3niX90, 2025-02-11:
https://github.com/Pho3niX90/solis_modbus/releases/tag/1.6.3-pre8

Pho3niX90, 2025-02-11:
Will investigate to probably break connection every 4 minutes to sync to solis cloud.

hbpv, 2025-02-11:
> Will investigate to probably break connection every 4 minutes to sync to solis cloud.

Just to be sure: This will not automaticly break every 4 minutes because in normal use-case, we need consistant data with no breaks.?! Maybe you could explain the code change a little bit more in detail...

Pho3niX90, 2025-02-11:
> > Will investigate to probably break connection every 4 minutes to sync to solis cloud.
> 
> Just to be sure: This will not automaticly break every 4 minutes because in normal use-case, we need consistant data with no breaks.?! Maybe you could explain the code change a little bit more in detail...

That feature is still a long way, probs around 1.7.0 when I am confident it works, but not yet coded.
1.6.3-pre8 just has a switch now to disable modbus, and enable it again:

[image]

closed 2025-02-11 by Pho3niX90