Pho3niX90/solis_modbus #116: Missing Meter Values

hbpv, 2025-02-12:
Hi
@Pho3niX90 

Did you change anything in between 1.6.2 … 1.6.3 with Meter Active Power? Singe yesterday 21:45 i dont have new values… See Screenshot.

Meter is ok i checked it
Soliscloud Shows also correct values

[image]

Pho3niX90, 2025-02-12:
No still the same. If the cloud is giving you values, then there is a modbus connection issue. 

You can try restarting HA, or the dongle

hbpv, 2025-02-12:
After 2 Restarts now values are back.
It Looks Like yesterday Night, the Modbus Integration „Gave up“ the Connection to the Solos cloud and didn‘t „Get it back“ Till My restarts.. thatswhy i saw values on soliscloud from around 21:45 yesterday Till my restarts.

So it Looks like we may Check why the Integration lost Connection and was Not able to gain it back?

Pho3niX90, 2025-02-12:
Are you using any other modbus integrations? Will need logs to see what happened

hbpv, 2025-02-12:
> Are you using any other modbus integrations? Will need logs to see what happened

No, not yet.. Maybe this helps? It startet with the error on job 21:16 and then was on error over the whole night...

`2025-02-11 21:14:05.023 WARNING (MainThread) [pymodbus.logging] Failed to connect [Errno 111] Connect call failed ('192.168.0.26', 502)
2025-02-11 21:16:42.828 ERROR (MainThread) [homeassistant] Error doing job: Fatal error: protocol.data_received() call failed. (None)
Traceback (most recent call last):
  File "/usr/local/lib/python3.13/asyncio/selector_events.py", line 1019, in _read_ready__data_received
    self._protocol.data_received(data)
    ~~~~~~~~~~~~~~~~~~~~~~~~~~~~^^^^^^
  File "/usr/local/lib/python3.13/site-packages/pymodbus/transport/transport.py", line 304, in data_received
    self.datagram_received(data, None)
    ~~~~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^
  File "/usr/local/lib/python3.13/site-packages/pymodbus/transport/transport.py", line 338, in datagram_received
    cut = self.callback_data(self.recv_buffer, addr=addr)
  File "/usr/local/lib/python3.13/site-packages/pymodbus/transaction/transaction.py", line 183, in callback_data
    self.response_future.set_result(pdu)
    ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^^^^^
asyncio.exceptions.InvalidStateError: invalid state
2025-02-11 21:16:42.933 WARNING (MainThread) [pymodbus.logging] Failed to connect [Errno 111] Connect call failed ('192.168.0.26', 502)
2025-02-11 21:16:43.136 WARNING (MainThread) [pymodbus.logging] Failed to connect [Errno 111] Connect call failed ('192.168.0.26', 502)
2025-02-11 21:16:43.538 WARNING (MainThread) [pymodbus.logging] Failed to connect [Errno 111] Connect call failed ('192.168.0.26', 502)
2025-02-11 21:16:44.342 WARNING (MainThread) [pymodbus.logging] Failed to connect [Errno 111] Connect call failed ('192.168.0.26', 502)
2025-02-11 21:16:45.944 WARNING (MainThread) [pymodbus.logging] Failed to connect [Errno 111] Connect call failed ('192.168.0.26', 502)
2025-02-11 21:16:49.146 WARNING (MainThread) [pymodbus.logging] Failed to connect [Errno 111] Connect call failed ('192.168.0.26', 502)
2025-02-11 21:16:49.262 WARNING (MainThread) [pymodbus.logging] Failed to connect [Errno 111] Connect call failed ('192.168.0.26', 502)
2025-02-11 21:16:49.363 WARNING (MainThread) [custom_components.solis_modbus.modbus_controller] Failed to connect to Modbus device. Will retry, failures = 331
2025-02-11 21:16:49.365 WARNING (MainThread) [pymodbus.logging] Failed to connect [Errno 111] Connect call failed ('192.168.0.26', 502)
2025-02-11 21:16:49.466 WARNING (MainThread) [custom_components.solis_modbus.modbus_controller] Failed to connect to Modbus device. Will retry, failures = 332
2025-02-11 21:16:49.468 WARNING (MainThread) [pymodbus.logging] Failed to connect [Errno 111] Connect call failed ('192.168.0.26', 502)
2025-02-11 21:16:49.569 WARNING (MainThread) [custom_components.solis_modbus.modbus_controller] Failed to connect to Modbus device. Will retry, failures = 333
2025-02-11 21:16:49.571 WARNING (MainThread) [pymodbus.logging] Failed to connect [Errno 111] Connect call failed ('192.168.0.26', 502)
2025-02-11 21:16:49.672 WARNING (MainThread) [custom_components.solis_modbus.modbus_controller] Failed to connect to Modbus device. Will retry, failures = 334
2025-02-11 21:16:49.674 WARNING (MainThread) [pymodbus.logging] Failed to connect [Errno 111] Connect call failed ('192.168.0.26', 502)
2025-02-11 21:16:49.775 WARNING (MainThread) [custom_components.solis_modbus.modbus_controller] Failed to connect to Modbus device. Will retry, failures = 335
2025-02-11 21:16:49.776 WARNING (MainThread) [pymodbus.logging] Failed to connect [Errno 111] Connect call failed ('192.168.0.26', 502)`

Pho3niX90, 2025-02-12:
It might have been a temporarily lockup of pymodbus. I am going to close this, if it happens again, just reopen

closed 2025-02-12 by Pho3niX90

hbpv, 2025-02-12:
> It might have been a temporarily lockup of pymodbus. I am going to close this, if it happens again, just reopen

After restarting for testing #123 , problem ist that it needs about >50 trys of modbus connection, then it connects, gets some value and then "hangs" on this values, dont receiving/updating ... SOMEtimes, after SOME time, it starts.. here with 2 ha restarts >

Debug is attached... 

[home-assistant_solis_modbus_2025-02-12T12-40-14.013Z.log](https://github.com/user-attachments/files/18767923/home-assistant_solis_modbus_2025-02-12T12-40-14.013Z.log)

Wilfried-Jordan, 2025-11-25:
Hello,
I screend this site 
https://solis-modbus.readthedocs.io/en/stable/ 
to find a holding register, where I can change the type of meter.
Eastron 1Ph / Eastron 3Ph / Acrel ...
My thanks in advance for a solution.