Pho3niX90/solis_modbus #123: [Feature]: Remote Control Force Charge Battery On/off

hbpv, 2025-02-12:
[image]

**Force-Charge manually via On/off switch**

The comment "Allow or not allow using grid.." in 43115 could mean, when not enough PV is available to force-charge, it will use grid.
43136 is Power setting for this Option

_If you like, i can try this out._

Premium: Add an option for timeout (time the switch automatacly switches back to off), in minutes.

Pho3niX90, 2025-02-12:
This should do it, if you can test and let me know:
https://github.com/Pho3niX90/solis_modbus/releases/tag/1.6.4-pre2

hbpv, 2025-02-12:
Entity is "not available"

[image]

Pho3niX90, 2025-02-12:
okay let me do some tests, not sure if that registrar is readable since it's prefixed with "Remote Control"

Pho3niX90, 2025-02-12:
https://github.com/Pho3niX90/solis_modbus/releases/tag/1.6.4-pre3 should resolve that @hbpv

hbpv, 2025-02-13:
> https://github.com/Pho3niX90/solis_modbus/releases/tag/1.6.4-pre3 should resolve that [@hbpv](https://github.com/hbpv)

Switch works, but Inverter does not start to load from grid (as thought). I will have a try at sunnier weather with sun-power-overflow if it loads the batt from PV. Please keep open.

wawyed, 2025-02-13:
Is this related to this new option in solis cloud? 

[image]

I've been very interested in being able to replicate it through home asisstant

wawyed, 2025-02-28:
@Pho3niX90 could we try to make this feature work? I'm very interested in it for purposes of being able to charge or discharge the battery programmatically when my energy provider sets sporadic free energy sessions

Pho3niX90, 2025-02-28:
It was implemented but had no response to confirm if it worked. I do not use the feature myself, so I can not confirm

wawyed, 2025-02-28:
Okay @Pho3niX90 , I've set

`solis_s6_eh3p_rc_force_battery_charge_power: 5000`

If I set: `solis_s6_solis_rc_force_battery_charge: on`

It nicely starts charging battery from the grid amazing!

Unfortunately if I set `solis_s6_solis_rc_force_battery_discharge: on`

It starts charging the battery again as the other switch, so I'm assuming it needs to point to another registry or another bit?

Another thing to note solis_s6_eh3p_rc_force_battery_charge_power unit is set to kW should be set to W

wawyed, 2025-02-28:
@hbpv do you which registry to remote force battery discharge in the documentation?

wawyed, 2025-02-28:
As I said above I managed to get it working for charging. Did you set solis_s6_eh3p_rc_force_battery_charge_power ?

hbpv, 2025-02-28:
I now had about 2,5kw from PV
Put Force Charge on
Funny: seconds After, the force Charge Button jumps back to off, and the Force DIScharge Button automatische jumps to on :-D

Regarding battery power: stays 0

[image]

hbpv, 2025-02-28:
> As I said above I managed to get it working for charging. Did you set solis_s6_eh3p_rc_force_battery_charge_power ?

Yes, 1000W

hbpv, 2025-02-28:
> > As I said above I managed to get it working for charging. Did you set solis_s6_eh3p_rc_force_battery_charge_power ?
> 
> Yes, 1000W

Oh. The wrong one, without _rc_… let Me try…

wawyed, 2025-02-28:
[image]

hbpv, 2025-02-28:
Cannot set it, it always jumps back to 0:

https://github.com/user-attachments/assets/493415f2-b601-4bef-9d3e-672cacbc3eed

hbpv, 2025-02-28:
When i set 3000kW it persists, @phoenix it should be Charger to W

Putting on Force Charge, it works now.

[image]

hbpv, 2025-02-28:
But forcecharge automaticly turned of and now forcedischarge is on.. maybe entities are matched to wrong/opposite registers @Pho3niX90 

So two Problems:

kW Must be W
Switch is opposite

[image]

wawyed, 2025-02-28:
@hbpv did you manage to get it to force discharge?

Pho3niX90, 2025-02-28:
This one uses only a single register, but the on value is 1, and 2. 
To diagnose, can you enable debug mode, try both, and post the logs please.

wawyed, 2025-02-28:
There has to be an option because solis cloud has that setting in the screenshot above

wawyed, 2025-02-28:
@Pho3niX90 what would you like me to try exactly ?

Pho3niX90, 2025-02-28:
> There has to be an option because solis cloud has that setting in the screenshot above

You are probably misunderstanding.
They both use a single register, if the value is 1, it's discharging, if 2, it's charging, 0, both disabled. 

I have made a change that I believe is the cause to the issue:
https://github.com/Pho3niX90/solis_modbus/releases/tag/3.0.9

Pho3niX90, 2025-02-28:
> [@Pho3niX90](https://github.com/Pho3niX90) what would you like me to try exactly ?

Just post the debug logs, when you try the charge and discharge, so that I can see how it's acting

wawyed, 2025-02-28:
Just checked the docs @Pho3niX90  you are right. 2 means discharge

[image]

wawyed, 2025-02-28:
Are the current toggles set in such way?

hbpv, 2025-02-28:
> Are the current toggles set in such way?

Toggle/Switch ForceCharge should set 1 if on and 0 if off

Toggle/Switch ForceDischarge should set 2 if on and 0 of off

wawyed, 2025-02-28:
I also found this:

[image]

It might be related?

Pho3niX90, 2025-02-28:
it is separated in 2 entities yes, the logic should dictate if it's showing as on or not, depending on their on value. Have you tried 3.0.9 to see if it resolves the issue?

hbpv, 2025-02-28:
> https://github.com/Pho3niX90/solis_modbus/releases/tag/3.1.0 with fixes to unit of measurement

💪 is Max 6000 given by Solis? My battery can c/ds with 7200 max, in other configurations its up to 12300w

Pho3niX90, 2025-02-28:
> > https://github.com/Pho3niX90/solis_modbus/releases/tag/3.1.0 with fixes to unit of measurement
> 
> 💪 is Max 6000 given by Solis? My battery can c/ds with 7200 max, in other configurations its up to 12300w

It's a bit complicated, this is vastly different between inverters, it can range from as low as 800w, up to 300000kw.

The main reason between the split of inverter models, is to control this better, and dynamically set the max and min values based on the selected inverter.

Pho3niX90, 2025-02-28:
> [@Pho3niX90](https://github.com/Pho3niX90) I'm getting charge switch discharging and discharching switch charging :D But discharge seems to be working now that I was able to set the discharge charge
> 
> I think those need to be inverted:
> 
> https://github.com/Pho3niX90/solis_modbus/blob/master/custom_components/solis_modbus/switch.py#L96

should solve that https://github.com/Pho3niX90/solis_modbus/compare/3.1.0...3.1.1

hbpv, 2025-02-28:
> > [@Pho3niX90](https://github.com/Pho3niX90) I'm getting charge switch discharging and discharching switch charging :D But discharge seems to be working now that I was able to set the discharge charge
> > I think those need to be inverted:
> > https://github.com/Pho3niX90/solis_modbus/blob/master/custom_components/solis_modbus/switch.py#L96
> 
> should solve that [3.1.0...3.1.1](https://github.com/Pho3niX90/solis_modbus/compare/3.1.0...3.1.1)

So no 3-was dropdown? Would be much more suiteable ❤️

Pho3niX90, 2025-02-28:
> So no 3-was dropdown? Would be much more suiteable ❤️

I will investigate it, but for now the integration doesn't have dropdowns used anywhere. Need to implement that first. Will do it during the day

wawyed, 2025-02-28:
Working as expected at least for now. I do agree dropdown would be fab. But thank you so much for this. It'll make my life so much easier :D

closed 2025-03-01 by Pho3niX90