Pho3niX90/solis_modbus #127: [Bug]: Hour and minute registers have values outside expected range causing values to not update

morts1, 2025-02-13:
## 🐛 Bug Description
Hour and minute registers have values outside expected range causing values to not update

## 🔄 To Reproduce
Steps to reproduce the behavior:
I've noticed that the charge/discharge times are not updating and in the logs with debug enabled it seems register values being polled are outside the expected range

## ✅ Expected Behavior
A clear and concise description of what you expected to happen.

## 📸 Screenshots / Logs
2025-02-13 12:17:08.940 DEBUG (MainThread) [custom_components.solis_modbus.time] Time: 43143, Hour = 23, Minute = 30
2025-02-13 12:17:08.940 DEBUG (MainThread) [custom_components.solis_modbus.time] Time: 43145, Hour = 5, Minute = 30
2025-02-13 12:17:08.940 DEBUG (MainThread) [custom_components.solis_modbus.time] Time: 43147, Hour = 16, Minute = 30
2025-02-13 12:17:08.940 DEBUG (MainThread) [custom_components.solis_modbus.time] Time: 43149, Hour = 16, Minute = 30
2025-02-13 12:17:08.940 DEBUG (MainThread) [custom_components.solis_modbus.time] Time: 43153, Hour = 9, Minute = 0
2025-02-13 12:17:08.940 DEBUG (MainThread) [custom_components.solis_modbus.time] Time: 43155, Hour = 9, Minute = 0
2025-02-13 12:17:08.940 DEBUG (MainThread) [custom_components.solis_modbus.time] Time: 43157, Hour = 0, Minute = 0
2025-02-13 12:17:08.941 DEBUG (MainThread) [custom_components.solis_modbus.time] Time: 43159, Hour = 0, Minute = 0
2025-02-13 12:17:08.941 DEBUG (MainThread) [custom_components.solis_modbus.time] Time: 43163, Hour = 6, Minute = 0
2025-02-13 12:17:08.941 DEBUG (MainThread) [custom_components.solis_modbus.time] Time: 43165, Hour = 6, Minute = 0
2025-02-13 12:17:08.941 DEBUG (MainThread) [custom_components.solis_modbus.time] Time: 43167, Hour = 0, Minute = 0
2025-02-13 12:17:08.941 DEBUG (MainThread) [custom_components.solis_modbus.time] Time: 43169, Hour = 0, Minute = 0
2025-02-13 12:17:08.941 DEBUG (MainThread) [custom_components.solis_modbus.time] Time: 43173, Hour = 0, Minute = 0
2025-02-13 12:17:08.941 DEBUG (MainThread) [custom_components.solis_modbus.time] Time: 43175, Hour = 0, Minute = 0
2025-02-13 12:17:08.941 DEBUG (MainThread) [custom_components.solis_modbus.time] Time: 43177, Hour = 0, Minute = 0
2025-02-13 12:17:08.941 DEBUG (MainThread) [custom_components.solis_modbus.time] Time: 43179, Hour = 0, Minute = 0
2025-02-13 12:17:08.942 DEBUG (MainThread) [custom_components.solis_modbus.time] Time: 43183, Hour = 0, Minute = 0
2025-02-13 12:17:08.942 DEBUG (MainThread) [custom_components.solis_modbus.time] Time: 43185, Hour = 0, Minute = 0
2025-02-13 12:17:08.942 DEBUG (MainThread) [custom_components.solis_modbus.time] Time: 43187, Hour = 0, Minute = 0
2025-02-13 12:17:08.942 DEBUG (MainThread) [custom_components.solis_modbus.time] Time: 43189, Hour = 0, Minute = 0
2025-02-13 12:17:08.942 DEBUG (MainThread) [custom_components.solis_modbus.time] Time: 43711, Hour = 0, Minute = 1
2025-02-13 12:17:08.942 DEBUG (MainThread) [custom_components.solis_modbus.time] Time: 43713, Hour = 2300, Minute = 5000
2025-02-13 12:17:08.942 DEBUG (MainThread) [custom_components.solis_modbus.time] Time: 43753, Hour = 0, Minute = 0
2025-02-13 12:17:08.942 DEBUG (MainThread) [custom_components.solis_modbus.time] Time: 43755, Hour = 0, Minute = 0
2025-02-13 12:17:08.942 DEBUG (MainThread) [custom_components.solis_modbus.time] Time: 43718, Hour = 650, Minute = 650
2025-02-13 12:17:08.943 DEBUG (MainThread) [custom_components.solis_modbus.time] Time: 43720, Hour = 420, Minute = 535
2025-02-13 12:17:08.943 DEBUG (MainThread) [custom_components.solis_modbus.time] Time: 43760, Hour = 3, Minute = 0
2025-02-13 12:17:08.943 DEBUG (MainThread) [custom_components.solis_modbus.time] Time: 43762, Hour = 50, Minute = 2498
2025-02-13 12:17:08.943 DEBUG (MainThread) [custom_components.solis_modbus.time] Time: 43725, Hour = 20, Minute = 10
2025-02-13 12:17:08.943 DEBUG (MainThread) [custom_components.solis_modbus.time] Time: 43727, Hour = 0, Minute = 0
2025-02-13 12:17:08.943 DEBUG (MainThread) [custom_components.solis_modbus.time] Time: 43767, Hour = 0, Minute = 0
2025-02-13 12:17:08.943 DEBUG (MainThread) [custom_components.solis_modbus.time] Time: 43769, Hour = 0, Minute = 1941
2025-02-13 12:17:08.943 DEBUG (MainThread) [custom_components.solis_modbus.time] Time: 43732, Hour = 0, Minute = 0
2025-02-13 12:17:08.943 DEBUG (MainThread) [custom_components.solis_modbus.time] Time: 43734, Hour = 0, Minute = 0
2025-02-13 12:17:08.944 DEBUG (MainThread) [custom_components.solis_modbus.time] Time: 43774, Hour = 0, Minute = 0
2025-02-13 12:17:08.944 DEBUG (MainThread) [custom_components.solis_modbus.time] Time: 43776, Hour = 1941, Minute = 65535
2025-02-13 12:17:08.944 DEBUG (MainThread) [custom_components.solis_modbus.time] Time: 43739, Hour = 0, Minute = 0
2025-02-13 12:17:08.944 DEBUG (MainThread) [custom_components.solis_modbus.time] Time: 43741, Hour = 0, Minute = 0
2025-02-13 12:17:08.944 DEBUG (MainThread) [custom_components.solis_modbus.time] Time: 43781, Hour = 0, Minute = 0
2025-02-13 12:17:08.944 DEBUG (MainThread) [custom_components.solis_modbus.time] Time: 43783, Hour = 65535, Minute = 65478
2025-02-13 12:17:08.944 DEBUG (MainThread) [custom_components.solis_modbus.time] Time: 43746, Hour = 0, Minute = 0
2025-02-13 12:17:08.944 DEBUG (MainThread) [custom_components.solis_modbus.time] Time: 43748, Hour = 0, Minute = 0
2025-02-13 12:17:08.944 DEBUG (MainThread) [custom_components.solis_modbus.time] Time: 43788, Hour = 0, Minute = 0
2025-02-13 12:17:08.945 DEBUG (MainThread) [custom_components.solis_modbus.time] Time: 43790, Hour = 0, Minute = 0
2025-02-13 12:17:08.945 ERROR (MainThread) [homeassistant] Error doing job: Task exception was never retrieved (None)
Traceback (most recent call last):
  File "/home/homeassistant/.homeassistant/custom_components/solis_modbus/time.py", line 117, in async_update
    await asyncio.gather(*[entity.async_update() for entity in hass.data[DOMAIN][TIME_ENTITIES]])
  File "/home/homeassistant/.homeassistant/custom_components/solis_modbus/time.py", line 170, in async_update
    self._attr_native_value = time(hour=hour, minute=minute)
                              ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
ValueError: hour must be in 0..23

## 🔧 System Information
### 🔹 Home Assistant:
- **Version**: Core-2025.1.0
- **Installation Method**: [e.g., Home Assistant OS, Docker, Supervised]

### 🔹 Integration:
- **Plugin Version**:1.6.4
- **Device Model**: Solis 5G 3.6kW hybrid
- **Modbus Connection Type?**: WiFi
- **Custom Config Entries?**: [Yes/No] No

## ➕ Additional Context
Any other details that might be helpful for debugging.

Pho3niX90, 2025-02-13:
For the following values:
```2025-02-13 12:17:08.942 DEBUG (MainThread) [custom_components.solis_modbus.time] Time: 43713, Hour = 2300, Minute = 5000
2025-02-13 12:17:08.942 DEBUG (MainThread) [custom_components.solis_modbus.time] Time: 43718, Hour = 650, Minute = 650
2025-02-13 12:17:08.943 DEBUG (MainThread) [custom_components.solis_modbus.time] Time: 43720, Hour = 420, Minute = 535
2025-02-13 12:17:08.943 DEBUG (MainThread) [custom_components.solis_modbus.time] Time: 43762, Hour = 50, Minute = 2498
2025-02-13 12:17:08.943 DEBUG (MainThread) [custom_components.solis_modbus.time] Time: 43725, Hour = 20, Minute = 10
2025-02-13 12:17:08.943 DEBUG (MainThread) [custom_components.solis_modbus.time] Time: 43769, Hour = 0, Minute = 1941
2025-02-13 12:17:08.944 DEBUG (MainThread) [custom_components.solis_modbus.time] Time: 43776, Hour = 1941, Minute = 65535
2025-02-13 12:17:08.944 DEBUG (MainThread) [custom_components.solis_modbus.time] Time: 43783, Hour = 65535, Minute = 65478```

Can you give guidance what they are on your inverter? 
43713 = Grid connection Time of Use charge period 1
43718 = Grid connection Time of Use charge period 2
43720 = Grid connection Time of Use charge period 3
43762 = Grid connection Time of Use discharge period 2
43725 = Grid connection Time of Use charge period 3
43769 = Grid connection Time of Use discharge period 3
43776 = Grid connection Time of Use discharge period 4
43783 = Grid connection Time of Use discharge period 5

These are all new since the firmware update, and documentation around them are non existent. 
43713 can be visually interpreted as 23:50

But the rest make no sense

Pho3niX90, 2025-02-13:
There is possibly a frame limitation, these groups of sensors were on the max (50 registrars), so they are separated into 2 frames now: https://github.com/Pho3niX90/solis_modbus/releases/tag/1.6.5-pre8

If you can see if that resolves it
@wawyed

wawyed, 2025-02-13:
No :( It's all 0 for the time entities.

The readonly sensors show the correct value though

[image]

wawyed, 2025-02-13:
[home-assistant_solis_modbus_2025-02-13T12-07-19.933Z.log](https://github.com/user-attachments/files/18785459/home-assistant_solis_modbus_2025-02-13T12-07-19.933Z.log)

Nothing here?

Pho3niX90, 2025-02-13:
> [home-assistant_solis_modbus_2025-02-13T12-07-19.933Z.log](https://github.com/user-attachments/files/18785459/home-assistant_solis_modbus_2025-02-13T12-07-19.933Z.log)
> 
> Nothing here?

Can you pinpoint the exact title of the sensors having values, but not in the time entities. Just need 1 or 2 examples. All I see from the screenshot is `Solis Grid Time of Use Charge`, which has 42 entities

wawyed, 2025-02-13:
```	

        Line  1635: 2025-02-13 11:50:47.663 DEBUG (MainThread) [custom_components.solis_modbus.sensor] register_key = 43753, value = 17
	Line  1636: 2025-02-13 11:50:47.663 DEBUG (MainThread) [custom_components.solis_modbus.sensor] register_key = 43754, value = 30
	Line  1637: 2025-02-13 11:50:47.664 DEBUG (MainThread) [custom_components.solis_modbus.sensor] register_key = 43755, value = 18
	Line  1638: 2025-02-13 11:50:47.664 DEBUG (MainThread) [custom_components.solis_modbus.sensor] register_key = 43756, value = 30

```

This would be Discharge slot 1 from 17:30 to 18:30

Pho3niX90, 2025-02-13:
> ```
> 
>         Line  1635: 2025-02-13 11:50:47.663 DEBUG (MainThread) [custom_components.solis_modbus.sensor] register_key = 43753, value = 17
> 	Line  1636: 2025-02-13 11:50:47.663 DEBUG (MainThread) [custom_components.solis_modbus.sensor] register_key = 43754, value = 30
> 	Line  1637: 2025-02-13 11:50:47.664 DEBUG (MainThread) [custom_components.solis_modbus.sensor] register_key = 43755, value = 18
> 	Line  1638: 2025-02-13 11:50:47.664 DEBUG (MainThread) [custom_components.solis_modbus.sensor] register_key = 43756, value = 30
> ```
> 
> This would be Discharge slot 1 from 17:30 to 18:30
> 
> 43711 that belongs to solis_s6_solis_grid_time_of_use_charge_start_hour_slot_1 it seems

Extremely strange as the time entity does pick it up correctly as well:

[image]

Are the old time entities still working as expected?

morts1, 2025-02-13:
> Extremely strange as the time entity does pick it up correctly as well:
>
> [image]
>
> Are the old time entities still working as expected?

It doesn't update the standard charge discharge timeslots 1 through to 3. I assume this is because async_update in time.py fails because of the out of range values in the new registers?

Pho3niX90, 2025-02-13:
> It doesn't update the standard charge discharge timeslots 1 through to 3. I assume this is because async_update in time.py fails because of the out of range values in the new registers?

What is your `Time of Use V2 Switch` providing?

Pho3niX90, 2025-02-13:
@wawyed this one should fixed the time entity issues you are having:
https://github.com/Pho3niX90/solis_modbus/releases/tag/1.6.5-pre9

wawyed, 2025-02-13:
Yes, that seems to work now :). Thank you

0rangutan, 2025-02-14:
Thanks for this update (and the v2.0!)

I am finding that a number of changes that I make aren't being applied and revert on the next poll.  I can't tell yet whether certain registers are more prone than others or whether it is just luck.  Sometimes it works, many times not.
Is this what the 2.0 Performance Testing build is looking at?

morts1, 2025-02-14:
> [@wawyed](https://github.com/wawyed) this one should fixed the time entity issues you are having, same for yours [@morts1](https://github.com/morts1), just not sure on the erroneous values coming from the inverter https://github.com/Pho3niX90/solis_modbus/releases/tag/1.6.5-pre9

I've upgraded to 2.0.0 and all looks good, thank you!

I am still seeing an exception in the logs related to the same area of code it seems

2025-02-14 11:45:00.975 ERROR (MainThread) [homeassistant] Error doing job: Task exception was never retrieved (None)
Traceback (most recent call last):
  File "/srv/homeassistant/lib/python3.12/site-packages/homeassistant/helpers/entity_platform.py", line 1065, in _async_update_entity_states
    await asyncio.gather(*tasks)
  File "/srv/homeassistant/lib/python3.12/site-packages/homeassistant/helpers/entity.py", line 978, in async_update_ha_state
    self._async_write_ha_state()
  File "/srv/homeassistant/lib/python3.12/site-packages/homeassistant/helpers/entity.py", line 1148, in _async_write_ha_state
    self.__async_calculate_state()
  File "/srv/homeassistant/lib/python3.12/site-packages/homeassistant/helpers/entity.py", line 1085, in __async_calculate_state
    state = self._stringify_state(available)
            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/srv/homeassistant/lib/python3.12/site-packages/homeassistant/helpers/entity.py", line 1029, in _stringify_state
    if (state := self.state) is None:
                 ^^^^^^^^^^
  File "/srv/homeassistant/lib/python3.12/site-packages/homeassistant/components/time/__init__.py", line 95, in state
    if self.native_value is None:
       ^^^^^^^^^^^^^^^^^
  File "/home/homeassistant/.homeassistant/custom_components/solis_modbus/time.py", line 139, in native_value
    return datetime.time(hour=vals[str(self._register)], minute=vals[str(self._register + 1)])
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
ValueError: hour must be in 0..23

See the following log for output from the integration around that time

[solis_modbus_home-assistant.log](https://github.com/user-attachments/files/18798773/solis_modbus_home-assistant.log)

Pho3niX90, 2025-02-14:
It seems you inverter might not support these time values, since 43707 is returning 0. It might need a firmware upgrade to support it

morts1, 2025-02-14:
I've never upgraded it as it was installed by a supplier. I will look into checking if its due an upgrade, thanks.

0rangutan, 2025-02-14:
You can raise a support ticket with Solis and ask for the latest firmware to be installed remotely.
They turned mine around in under 24 hours.

morts1, 2025-02-15:
My Inverter firmware has been upgraded and all looks good, thank you!

Pho3niX90, 2025-02-16:
Awesome, thanks good to know

closed 2025-02-16 by Pho3niX90