Pho3niX90/solis_modbus #130: [Bug]: Discharge entities naming conventions for 6 slot firmware

0rangutan, 2025-02-14:
I am have noticed problems with my discharge slots since moving to the Solis 6 slot firmware and these aren't fixed in 2.0.

1. Some of my slot entities exist twice, eg.
time.solis_solis_time_charging_discharge_start_slot_1   <- Not working
time.solis_solis_time_charging_discharge_start_slot_1_2.  <- Working

2. Entity naming conventions vary, eg.
Discharge slots 1-3 use "solis_time_charging_discharge", while 4-6 use "solis_grid_time_of_use_discharge"

This is what my Discharge entities looks like once I have identified which ones are correct and working, demonstrating both problems:

  - entity: time.solis_solis_time_charging_discharge_start_slot_1_2
    name: Discharge 1 Start
  - entity: time.solis_solis_time_charging_discharge_end_slot_1
    name: Discharge 1 End
  - entity: time.solis_solis_time_charging_discharge_start_slot_2_2
    name: Discharge 2 Start
  - entity: time.solis_solis_time_charging_discharge_end_slot_2
    name: Discharge 2 End
  - entity: time.solis_solis_time_charging_discharge_start_slot_3_2
    name: Discharge 3 Start
  - entity: time.solis_solis_time_charging_discharge_end_slot_3
    name: Discharge 3 End
  - entity: time.solis_solis_grid_time_of_use_discharge_start_slot_4
    name: Discharge 4 Start
  - entity: time.solis_solis_grid_time_of_use_discharge_end_slot_4
    name: Discharge 4 End
  - entity: time.solis_solis_grid_time_of_use_discharge_start_slot_5
    name: Discharge 5 Start
  - entity: time.solis_solis_grid_time_of_use_discharge_end_slot_5
    name: Discharge 5 End
  - entity: time.solis_solis_grid_time_of_use_discharge_start_slot_6
    name: Discharge 6 Start
  - entity: time.solis_solis_grid_time_of_use_discharge_end_slot_6
    name: Discharge 6 End  

Is this an issue with the integration or with my setup?  Thanks

Pho3niX90, 2025-02-14:
Yes this is a know issue, you can delete the ones not working, I did make a note in the patch notes https://github.com/Pho3niX90/solis_modbus/releases/tag/2.0.0

0rangutan, 2025-02-14:
Thanks for confirming.
So if I delete the non-working ones and re-name the others, they will survive future updates?

Pho3niX90, 2025-02-14:
> Thanks for confirming. So if I delete the non-working ones and re-name the others, they will survive future updates?

Yes that is correct.

closed 2025-02-18 by Pho3niX90