Pho3niX90/solis_modbus #131: [Bug]: Modbus connection breaks up

debietim, 2025-02-15:
I am on the latest version of HA software and Solis Modbus. But for some reason the modes connection keeps breaking up.
After reboot of HA, everything works well for about 30 minutes. After that connection breaks up and everything is back in SolisCloud

Pho3niX90, 2025-02-15:
you need to provide logs. otherwise I will have to close the issue

debietim, 2025-02-15:
Sorry for this... hereby the logs...

(could not load the file because of the type, so sorry for this dump... hope you have enough info with this?)

homeassistant.exceptions.NoEntitySpecifiedError: No entity id specified for entity None
2025-02-15 18:32:32.795 ERROR (MainThread) [homeassistant] Error doing job: Exception in callback Entity._async_write_ha_state_from_call_soon_threadsafe() (None)
Traceback (most recent call last):
  File "/usr/local/lib/python3.13/asyncio/events.py", line 89, in _run
    self._context.run(self._callback, *self._args)
    ~~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 1013, in _async_write_ha_state_from_call_soon_threadsafe
    self._async_verify_state_writable()
    ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^^
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 1003, in _async_verify_state_writable
    raise NoEntitySpecifiedError(
        f"No entity id specified for entity {self.name}"
    )
homeassistant.exceptions.NoEntitySpecifiedError: No entity id specified for entity None
2025-02-15 18:32:37.183 DEBUG (MainThread) [custom_components.solis_modbus.modbus_controller] Connecting to Modbus TCP...
2025-02-15 18:32:37.869 INFO (MainThread) [custom_components.solis_modbus.number] calling number update for 51 groups
2025-02-15 18:32:39.910 ERROR (MainThread) [pymodbus.logging] Cancel send, because not connected!
2025-02-15 18:32:47.186 WARNING (MainThread) [pymodbus.logging] Failed to connect 
2025-02-15 18:32:47.287 WARNING (MainThread) [custom_components.solis_modbus.modbus_controller] Failed to connect (Attempt 369)
2025-02-15 18:32:49.912 ERROR (MainThread) [pymodbus.logging] Cancel send, because not connected!
2025-02-15 18:32:50.970 WARNING (MainThread) [pymodbus.logging] Failed to connect 
2025-02-15 18:32:52.184 DEBUG (MainThread) [custom_components.solis_modbus.modbus_controller] Connecting to Modbus TCP...
2025-02-15 18:32:59.914 ERROR (MainThread) [pymodbus.logging] Cancel send, because not connected!
2025-02-15 18:33:02.186 WARNING (MainThread) [pymodbus.logging] Failed to connect 
2025-02-15 18:33:02.287 WARNING (MainThread) [custom_components.solis_modbus.modbus_controller] Failed to connect (Attempt 370)
2025-02-15 18:33:07.186 DEBUG (MainThread) [custom_components.solis_modbus.modbus_controller] Connecting to Modbus TCP...
2025-02-15 18:33:09.917 ERROR (MainThread) [pymodbus.logging] Cancel send, because not connected!
2025-02-15 18:33:17.188 WARNING (MainThread) [pymodbus.logging] Failed to connect 
2025-02-15 18:33:17.289 WARNING (MainThread) [custom_components.solis_modbus.modbus_controller] Failed to connect (Attempt 371)
2025-02-15 18:33:19.919 ERROR (MainThread) [pymodbus.logging] Cancel send, because not connected!
2025-02-15 18:33:20.973 WARNING (MainThread) [pymodbus.logging] Failed to connect 
2025-02-15 18:33:22.186 DEBUG (MainThread) [custom_components.solis_modbus.modbus_controller] Connecting to Modbus TCP...
2025-02-15 18:33:22.869 INFO (MainThread) [custom_components.solis_modbus.number] calling number update for 51 groups
2025-02-15 18:33:29.806 WARNING (MainThread) [bluetooth_auto_recovery.recover] Could not determine the power state of the Bluetooth adapter hci0 [00:00:00:00:00:00] due to timeout after 5 seconds
2025-02-15 18:33:29.919 ERROR (MainThread) [pymodbus.logging] Cancel send, because not connected!
2025-02-15 18:33:32.187 WARNING (MainThread) [pymodbus.logging] Failed to connect 
2025-02-15 18:33:32.288 WARNING (MainThread) [custom_components.solis_modbus.modbus_controller] Failed to connect (Attempt 372)
2025-02-15 18:33:32.650 ERROR (MainThread) [homeassistant] Error doing job: Exception in callback Entity._async_write_ha_state_from_call_soon_threadsafe() (None)
Traceback (most recent call last):
  File "/usr/local/lib/python3.13/asyncio/events.py", line 89, in _run
    self._context.run(self._callback, *self._args)
    ~~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 1013, in _async_write_ha_state_from_call_soon_threadsafe
    self._async_verify_state_writable()
    ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^^
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 1003, in _async_verify_state_writable
    raise NoEntitySpecifiedError(
        f"No entity id specified for entity {self.name}"
    )
homeassistant.exceptions.NoEntitySpecifiedError: No entity id specified for entity None
2025-02-15 18:33:32.785 ERROR (MainThread) [homeassistant] Error doing job: Exception in callback Entity._async_write_ha_state_from_call_soon_threadsafe() (None)
Traceback (most recent call last):
  File "/usr/local/lib/python3.13/asyncio/events.py", line 89, in _run
    self._context.run(self._callback, *self._args)
    ~~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 1013, in _async_write_ha_state_from_call_soon_threadsafe
    self._async_verify_state_writable()
    ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^^
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 1003, in _async_verify_state_writable
    raise NoEntitySpecifiedError(
        f"No entity id specified for entity {self.name}"
    )
homeassistant.exceptions.NoEntitySpecifiedError: No entity id specified for entity None
2025-02-15 18:33:37.184 DEBUG (MainThread) [custom_components.solis_modbus.modbus_controller] Connecting to Modbus TCP...
[... cut ...]
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 1003, in _async_verify_state_writable
    raise NoEntitySpecifiedError(
        f"No entity id specified for entity {self.name}"
    )
homeassistant.exceptions.NoEntitySpecifiedError: No entity id specified for entity None
2025-02-15 18:39:37.216 DEBUG (MainThread) [custom_components.solis_modbus.modbus_controller] Connecting to Modbus TCP...
2025-02-15 18:39:39.983 ERROR (MainThread) [pymodbus.logging] Cancel send, because not connected!
2025-02-15 18:39:47.218 WARNING (MainThread) [pymodbus.logging] Failed to connect

Pho3niX90, 2025-02-16:
What version?

debietim, 2025-02-16:
It is version 2.1.0...
Core version 2025.2.4

hbpv, 2025-02-18:
For me today is the same problem, loosing connection and not able to gain it back in acceptable time:

`2025-02-18 07:59:36.271 WARNING (MainThread) [homeassistant.components.sensor] Entity sensor.solis_s6_solis_off_grid_overdischarge_soc (<class 'custom_components.solis_modbus.sensor.SolisSensor'>) is using native unit of measurement '%' which is not a valid unit for the device class ('current') it is using; expected one of ['mA', 'A']; Please update your configuration if your entity is manually configured, otherwise create a bug report at https://github.com/Pho3niX90/solis_modbus/issues
`

`2025-02-18 09:44:03.476 ERROR (MainThread) [homeassistant] Error doing job: Fatal error: protocol.data_received() call failed. (None)
Traceback (most recent call last):
  File "/usr/local/lib/python3.13/asyncio/selector_events.py", line 1019, in _read_ready__data_received
    self._protocol.data_received(data)
    ~~~~~~~~~~~~~~~~~~~~~~~~~~~~^^^^^^
  File "/usr/local/lib/python3.13/site-packages/pymodbus/transport/transport.py", line 304, in data_received
    self.datagram_received(data, None)
    ~~~~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^
  File "/usr/local/lib/python3.13/site-packages/pymodbus/transport/transport.py", line 338, in datagram_received
    cut = self.callback_data(self.recv_buffer, addr=addr)
  File "/usr/local/lib/python3.13/site-packages/pymodbus/transaction/transaction.py", line 183, in callback_data
    self.response_future.set_result(pdu)
    ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^^^^^
asyncio.exceptions.InvalidStateError: invalid state
2025-02-18 09:44:06.619 WARNING (MainThread) [pymodbus.logging] Failed to connect [Errno 111] Connect call failed ('192.168.0.26', 502)
2025-02-18 09:44:06.721 WARNING (MainThread) [custom_components.solis_modbus.modbus_controller] Failed to connect (Attempt 1)
2025-02-18 09:44:13.490 WARNING (MainThread) [pymodbus.logging] Failed to connect [Errno 111] Connect call failed ('192.168.0.26', 502)
2025-02-18 09:44:21.661 WARNING (MainThread) [pymodbus.logging] Failed to connect [Errno 111] Connect call failed ('192.168.0.26', 502)
2025-02-18 09:44:21.762 WARNING (MainThread) [custom_components.solis_modbus.modbus_controller] Failed to connect (Attempt 2)
2025-02-18 09:44:33.493 WARNING (MainThread) [pymodbus.logging] Failed to connect [Errno 111] Connect call failed ('192.168.0.26', 502)
2025-02-18 09:44:36.647 WARNING (MainThread) [pymodbus.logging] Failed to connect [Errno 111] Connect call failed ('192.168.0.26', 502)
2025-02-18 09:44:36.748 WARNING (MainThread) [custom_components.solis_modbus.modbus_controller] Failed to connect (Attempt 3)
2025-02-18 09:44:51.663 WARNING (MainThread) [pymodbus.logging] Failed to connect [Errno 111] Connect call failed ('192.168.0.26', 502)
2025-02-18 09:44:51.764 WARNING (MainThread) [custom_components.solis_modbus.modbus_controller] Failed to connect (Attempt 4)
2025-02-18 09:44:53.495 WARNING (MainThread) [pymodbus.logging] Failed to connect [Errno 111] Connect call failed ('192.168.0.26', 502)
2025-02-18 09:45:06.651 WARNING (MainThread) [pymodbus.logging] Failed to connect [Errno 111] Connect call failed ('192.168.0.26', 502)
2025-02-18 09:45:06.752 WARNING (MainThread) [custom_components.solis_modbus.modbus_controller] Failed to connect (Attempt 5)
2025-02-18 09:45:13.497 WARNING (MainThread) [pymodbus.logging] Failed to connect [Errno 111] Connect call failed ('192.168.0.26', 502)
2025-02-18 09:45:21.670 WARNING (MainThread) [pymodbus.logging] Failed to connect [Errno 111] Connect call failed ('192.168.0.26', 502)
2025-02-18 09:45:21.772 WARNING (MainThread) [custom_components.solis_modbus.modbus_controller] Failed to connect (Attempt 6)
2025-02-18 09:45:33.498 WARNING (MainThread) [pymodbus.logging] Failed to connect [Errno 111] Connect call failed ('192.168.0.26', 502)
2025-02-18 09:45:36.663 WARNING (MainThread) [pymodbus.logging] Failed to connect [Errno 111] Connect call failed ('192.168.0.26', 502)
2025-02-18 09:45:36.764 WARNING (MainThread) [custom_components.solis_modbus.modbus_controller] Failed to connect (Attempt 7)
2025-02-18 09:45:51.670 WARNING (MainThread) [pymodbus.logging] Failed to connect [Errno 111] Connect call failed ('192.168.0.26', 502)
2025-02-18 09:45:51.771 WARNING (MainThread) [custom_components.solis_modbus.modbus_controller] Failed to connect (Attempt 8)
2025-02-18 09:45:53.500 WARNING (MainThread) [pymodbus.logging] Failed to connect [Errno 111] Connect call failed ('192.168.0.26', 502)
2025-02-18 09:46:06.862 WARNING (MainThread) [pymodbus.logging] Failed to connect [Errno 111] Connect call failed ('192.168.0.26', 502)
2025-02-18 09:46:06.963 WARNING (MainThread) [custom_components.solis_modbus.modbus_controller] Failed to connect (Attempt 9)
2025-02-18 09:46:13.503 WARNING (MainThread) [pymodbus.logging] Failed to connect [Errno 111] Connect call failed ('192.168.0.26', 502)
2025-02-18 09:46:31.615 WARNING (MainThread) [pymodbus.logging] Failed to connect 
2025-02-18 09:46:31.716 WARNING (MainThread) [custom_components.solis_modbus.modbus_controller] Failed to connect (Attempt 10)
`

Pho3niX90, 2025-02-18:
The above logs suggest one of the following:

1.  Unstable network conditions. Either the inverter is losing connection, or the connection is a unstable. Check the inverters web page in the browser to see the signal strength. 
2. Another modbus integration is trying to connect to the inverter, causing issues. Only a single integration/device can access a modbus at a given time.

hbpv, 2025-02-18:
Hi @Pho3niX90 

1. The WIFI/LAN Stick is directly LAN-connected (4meters) to the home-switch > There's no way to get a more stable connection
2. There is no other modbus device in this household, otherwise and maybe later, i would configure a modbus proxy, but as i said, there is only the inverter with modbus in the household-configuration
3. MAYBE: What if the stick try's to connect to solis cloud or vice versa and "kills" that connection the same way, the modbus kill's it when trying to connect? (a fight between Cloud and Modbus starting from the solis side?) Did you heard anything about that?
4. Could we integrate a more detailed logging instead only "Failed to connect [Errno 111] Connect call failed" ?

hbpv, 2025-02-18:
Maybe in this moment (see also my inital post from today), connection is lost and soliscloud "jumps in", then it needs about 10 attempts to kick soliscloud out:

`2025-02-18 09:44:03.476 ERROR (MainThread) [homeassistant] Error doing job: Fatal error: protocol.data_received() call failed. (None) Traceback (most recent call last): File "/usr/local/lib/python3.13/asyncio/selector_events.py", line 1019, in _read_ready__data_received self._protocol.data_received(data) ~~~~~~~~~~~~~~~~~~~~~~~~~~~~^^^^^^ File "/usr/local/lib/python3.13/site-packages/pymodbus/transport/transport.py", line 304, in data_received self.datagram_received(data, None) ~~~~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^ File "/usr/local/lib/python3.13/site-packages/pymodbus/transport/transport.py", line 338, in datagram_received cut = self.callback_data(self.recv_buffer, addr=addr) File "/usr/local/lib/python3.13/site-packages/pymodbus/transaction/transaction.py", line 183, in callback_data self.response_future.set_result(pdu) ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^^^^^ asyncio.exceptions.InvalidStateError: invalid state`

Pho3niX90, 2025-02-18:
> Hi [@Pho3niX90](https://github.com/Pho3niX90)
>
> 1. The WIFI/LAN Stick is directly LAN-connected (4meters) to the home-switch > There's no way to get a more stable connection
> 2. There is no other [...]

1. Please do some ping tests to the inverter to check stability throughout the day, even though it's cable, there could be an issue possibly. 
2. That rules it out then.
3. Not impossible, but I do not have the same issue, thusfar, it "seems" as it's only related to the users in this thread
4. 
That is as detailed as it gets I am afraid, 
`(MainThread) [pymodbus.logging] Failed to connect [Errno 111] Connect call failed`

Is from the underlying library called `pymodbus` that this integration and home assistant uses, to speak to your inverter, 111 means the inverter is refusing connection outright. This could be due to a connection issue

One other possibility is, that your home assistant is locking up, which causes connection to drop. How is the CPU utilization during these occurrences? Is any other integration causing errors around the same time? With full logs these are easier to diagnose, but with what is provided, I cannot give more guidance.

hbpv, 2025-02-18:
Please see the log after the last restart attached.

The Modbus Error is on ~ 9:44 and there's next Error before is "2025-02-18 09:09:51.865 ERROR (MainThread) [homeassistant.components.overkiz] Unexpected error fetching device events data" ... see in the log. Thats about a half an hour before, so i dont think this is related.

[home-assistant-today.log](https://github.com/user-attachments/files/18843616/home-assistant-today.log)

hbpv, 2025-02-18:
And it happend again... no data since 27 minutes..

`2025-02-18 12:17:02.793 ERROR (MainThread) [homeassistant] Error doing job: Fatal error: protocol.data_received() call failed. (None)
Traceback (most recent call last):
  File "/usr/local/lib/python3.13/asyncio/selector_events.py", line 1019, in _read_ready__data_received
    self._protocol.data_received(data)
    ~~~~~~~~~~~~~~~~~~~~~~~~~~~~^^^^^^
  File "/usr/local/lib/python3.13/site-packages/pymodbus/transport/transport.py", line 304, in data_received
    self.datagram_received(data, None)
    ~~~~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^
  File "/usr/local/lib/python3.13/site-packages/pymodbus/transport/transport.py", line 338, in datagram_received
    cut = self.callback_data(self.recv_buffer, addr=addr)
  File "/usr/local/lib/python3.13/site-packages/pymodbus/transaction/transaction.py", line 183, in callback_data
    self.response_future.set_result(pdu)
    ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^^^^^
asyncio.exceptions.InvalidStateError: invalid state
2025-02-18 12:17:07.242 WARNING (MainThread) [pymodbus.logging] Failed to connect [Errno 111] Connect call failed ('192.168.0.26', 502)
2025-02-18 12:17:07.343 WARNING (MainThread) [custom_components.solis_modbus.modbus_controller] Failed to connect (Attempt 11)
2025-02-18 12:17:12.797 WARNING (MainThread) [pymodbus.logging] Failed to connect [Errno 111] Connect call failed ('192.168.0.26', 502)
2025-02-18 12:17:22.232 WARNING (MainThread) [pymodbus.logging] Failed to connect [Errno 111] Connect call failed ('192.168.0.26', 502)
2025-02-18 12:17:22.333 WARNING (MainThread) [custom_components.solis_modbus.modbus_controller] Failed to connect (Attempt 12)
2025-02-18 12:17:32.800 WARNING (MainThread) [pymodbus.logging] Failed to connect [Errno 111] Connect call failed ('192.168.0.26', 502)
2025-02-18 12:17:37.219 WARNING (MainThread) [pymodbus.logging] Failed to connect [Errno 111] Connect call failed ('192.168.0.26', 502)
2025-02-18 12:17:37.320 WARNING (MainThread) [custom_components.solis_modbus.modbus_controller] Failed to connect (Attempt 13)
2025-02-18 12:17:52.234 WARNING (MainThread) [pymodbus.logging] Failed to connect [Errno 111] Connect call failed ('192.168.0.26', 502)
2025-02-18 12:17:52.336 WARNING (MainThread) [custom_components.solis_modbus.modbus_controller] Failed to connect (Attempt 14)
2025-02-18 12:17:52.802 WARNING (MainThread) [pymodbus.logging] Failed to connect [Errno 111] Connect call failed ('192.168.0.26', 502)
2025-02-18 12:18:07.231 WARNING (MainThread) [pymodbus.logging] Failed to connect [Errno 111] Connect call failed ('192.168.0.26', 502)
2025-02-18 12:18:07.332 WARNING (MainThread) [custom_components.solis_modbus.modbus_controller] Failed to connect (Attempt 15)
2025-02-18 12:18:12.804 WARNING (MainThread) [pymodbus.logging] Failed to connect [Errno 111] Connect call failed ('192.168.0.26', 502)
2025-02-18 12:18:22.224 WARNING (MainThread) [pymodbus.logging] Failed to connect [Errno 111] Connect call failed ('192.168.0.26', 502)
2025-02-18 12:18:22.325 WARNING (MainThread) [custom_components.solis_modbus.modbus_controller] Failed to connect (Attempt 16)
2025-02-18 12:18:32.806 WARNING (MainThread) [pymodbus.logging] Failed to connect [Errno 111] Connect call failed ('192.168.0.26', 502)
2025-02-18 12:18:37.225 WARNING (MainThread) [pymodbus.logging] Failed to connect [Errno 111] Connect call failed ('192.168.0.26', 502)
2025-02-18 12:18:37.326 WARNING (MainThread) [custom_components.solis_modbus.modbus_controller] Failed to connect (Attempt 17)
2025-02-18 12:19:02.176 WARNING (MainThread) [pymodbus.logging] Failed to connect 
2025-02-18 12:19:02.276 WARNING (MainThread) [custom_components.solis_modbus.modbus_controller] Failed to connect (Attempt 18)
2025-02-18 12:19:02.809 WARNING (MainThread) [pymodbus.logging] Failed to connect 
2025-02-18 12:19:22.812 WARNING (MainThread) [pymodbus.logging] Failed to connect [Errno 111] Connect call failed ('192.168.0.26', 502)
2025-02-18 12:19:42.814 WARNING (MainThread) [pymodbus.logging] Failed to connect [Errno 111] Connect call failed ('192.168.0.26', 502)
2025-02-18 12:20:22.817 WARNING (MainThread) [pymodbus.logging] Failed to connect [Errno 111] Connect call failed ('192.168.0.26', 502)
2025-02-18 12:21:42.819 WARNING (MainThread) [pymodbus.logging] Failed to connect [Errno 111] Connect call failed ('192.168.0.26', 502)
2025-02-18 12:24:22.821 WARNING (MainThread) [pymodbus.logging] Failed to connect [Errno 111] Connect call failed ('192.168.0.26', 502)
2025-02-18 12:29:22.825 WARNING (MainThread) [pymodbus.logging] Failed to connect [Errno 111] Connect call failed ('192.168.0.26', 502)
2025-02-18 12:34:22.828 WARNING (MainThread) [pymodbus.logging] Failed to connect [Errno 111] Connect call failed ('192.168.0.26', 502)
2025-02-18 12:39:22.830 WARNING (MainThread) [pymodbus.logging] Failed to connect [Errno 111] Connect call failed ('192.168.0.26', 502)
2025-02-18 12:44:22.832 WARNING (MainThread) [pymodbus.logging] Failed to connect [Errno 111] Connect call failed ('192.168.0.26', 502)
2025-02-18 12:49:22.835 WARNING (MainThread) [pymodbus.logging] Failed to connect [Errno 111] Connect call failed ('192.168.0.26', 502)
2025-02-18 12:54:22.837 WARNING (MainThread) [pymodbus.logging] Failed to connect [Errno 111] Connect call failed ('192.168.0.26', 502)
2025-02-18 12:59:22.838 WARNING (MainThread) [pymodbus.logging] Failed to connect [Errno 111] Connect call failed ('192.168.0.26', 502)
2025-02-18 13:04:22.841 WARNING (MainThread) [pymodbus.logging] Failed to connect [Errno 111] Connect call failed ('192.168.0.26', 502)
2025-02-18 13:09:22.843 WARNING (MainThread) [pymodbus.logging] Failed to connect [Errno 111] Connect call failed ('192.168.0.26', 502)
`

hbpv, 2025-02-18:
After another restart... and another restart.. no connection, see debuglog

[home-assistant_solis_modbus_2025-02-18T12-34-04.810Z.log](https://github.com/user-attachments/files/18845187/home-assistant_solis_modbus_2025-02-18T12-34-04.810Z.log)

hbpv, 2025-02-19:
Today:

[home-assistant.log](https://github.com/user-attachments/files/18867330/home-assistant.log)

computic2023, 2025-02-19:
Hello, I am having similar problems. Everything was working perfectly before the update.
Now, data sometimes arrives in home assistant, and when it arrives, it's only for a few minutes.
The installed HA version is 2025.2.4 and the solis plugin version is 2.1.2.
Thank you in advance for your help.

[home-assistant_solis_modbus_2025-02-19T13-33-16.646Z.log](https://github.com/user-attachments/files/18869262/home-assistant_solis_modbus_2025-02-19T13-33-16.646Z.log)

Pho3niX90, 2025-02-19:
> Today:
> 
> [home-assistant.log](https://github.com/user-attachments/files/18867330/home-assistant.log)

These logs seem to be taken before the restart? If it happens again, please download the logs first before restarting.

Pho3niX90, 2025-02-20:
please try the beta release -> https://github.com/Pho3niX90/solis_modbus/releases/tag/3.0.0-beta-1

There is better updating of sensor values, and some fixes pertaining to the registrar code blocks.

hbpv, 2025-02-20:
> please try the beta release -> https://github.com/Pho3niX90/solis_modbus/releases/tag/3.0.0-beta-1
> 
> There is better updating of sensor values, and some fixes pertaining to the registrar code blocks.

Great, i will try it..but.. You did not change necessary sensor unique_id's right? Just friendly names? My system is semi-live so this would be bad

best,
Max

debietim, 2025-02-20:
Just installed the new version and selected the right inverter. Since then, my HomeAssistant server is rebooting every minute...

Pho3niX90, 2025-02-20:
Try https://github.com/Pho3niX90/solis_modbus/releases/tag/3.0.0-beta-2

Pho3niX90, 2025-02-20:
> Great, i will try it..but.. You did not change necessary sensor unique_id's right? Just friendly names? My system is semi-live so this would be bad
>
> best, Max

Just friendly names are changed. Previously all was prefixed with solis, which would cause duplicated names in some scenarios where it would have shown `Solis S6 Solis`

hbpv, 2025-02-20:
> Just installed the new version and selected the right inverter. Since then, my HomeAssistant server is rebooting every minute...

@Pho3niX90 so good i was having lunch :) i did not update..

Pho3niX90, 2025-02-20:
@hbpv @debietim let me know if all is well, if not, please upload the logs BEFORE restart

debietim, 2025-02-20:
I just was able to quit the rebooting. After installing 3.0.0 beta 2, rebooting stopped :-)
Was not able to fetch logs, since the device rebooted every 30 to 45 seconds...

debietim, 2025-02-20:
But the values are not correct. I see on the inverter that my battery is charging with about 4000W, while the modbus values keep showing data what is old (discharging 489W)...

hbpv, 2025-02-20:
> I just was able to quit the rebooting. After installing 3.0.0 beta 2, rebooting stopped :-) Was not able to fetch logs, since the device rebooted every 30 to 45 seconds...

So it's now safe to try 3 beta 2 ? :)

debietim, 2025-02-20:
First impression: the latest version seems the most stable version in weeks. Modbus connection stays up-and-running for the longest time now in weeks.

hbpv, 2025-02-20:
> First impression: the latest version seems the most stable version in weeks. Modbus connection stays up-and-running for the longest time now in weeks.

Thanks! Waiting for tomorrow, if you dont have any addional trouble tilll then, i will have a try too.. 

I hope 3.0 also fixes the "not permanent" but the recurring loses of connection... this is <one hour> at 2.1.2, it fetches back with no permanent loss, but i think this is not the way it should work and maybe sometimes leads to a "full crash" of the connection...:

`2025-02-20 12:02:20.326 WARNING (MainThread) [pymodbus.logging] Failed to connect [Errno 111] Connect call failed ('192.168.0.26', 502)
2025-02-20 12:04:08.670 WARNING (MainThread) [pymodbus.logging] Failed to connect [Errno 111] Connect call failed ('192.168.0.26', 502)
2025-02-20 12:07:20.328 WARNING (MainThread) [pymodbus.logging] Failed to connect [Errno 111] Connect call failed ('192.168.0.26', 502)
2025-02-20 12:09:08.677 WARNING (MainThread) [pymodbus.logging] Failed to connect [Errno 111] Connect call failed ('192.168.0.26', 502)
2025-02-20 12:12:20.332 WARNING (MainThread) [pymodbus.logging] Failed to connect [Errno 111] Connect call failed ('192.168.0.26', 502)
2025-02-20 12:14:08.684 WARNING (MainThread) [pymodbus.logging] Failed to connect [Errno 111] Connect call failed ('192.168.0.26', 502)
2025-02-20 12:17:20.334 WARNING (MainThread) [pymodbus.logging] Failed to connect [Errno 111] Connect call failed ('192.168.0.26', 502)
2025-02-20 12:19:08.723 WARNING (MainThread) [pymodbus.logging] Failed to connect [Errno 111] Connect call failed ('192.168.0.26', 502)
2025-02-20 12:22:20.337 WARNING (MainThread) [pymodbus.logging] Failed to connect [Errno 111] Connect call failed ('192.168.0.26', 502)
2025-02-20 12:24:08.737 WARNING (MainThread) [pymodbus.logging] Failed to connect [Errno 111] Connect call failed ('192.168.0.26', 502)
2025-02-20 12:27:20.340 WARNING (MainThread) [pymodbus.logging] Failed to connect [Errno 111] Connect call failed ('192.168.0.26', 502)
2025-02-20 12:29:08.774 WARNING (MainThread) [pymodbus.logging] Failed to connect [Errno 111] Connect call failed ('192.168.0.26', 502)
2025-02-20 12:32:20.341 WARNING (MainThread) [pymodbus.logging] Failed to connect [Errno 111] Connect call failed ('192.168.0.26', 502)
2025-02-20 12:34:08.777 WARNING (MainThread) [pymodbus.logging] Failed to connect [Errno 111] Connect call failed ('192.168.0.26', 502)
2025-02-20 12:37:20.345 WARNING (MainThread) [pymodbus.logging] Failed to connect [Errno 111] Connect call failed ('192.168.0.26', 502)
2025-02-20 12:39:08.797 WARNING (MainThread) [pymodbus.logging] Failed to connect [Errno 111] Connect call failed ('192.168.0.26', 502)
2025-02-20 12:42:20.347 WARNING (MainThread) [pymodbus.logging] Failed to connect [Errno 111] Connect call failed ('192.168.0.26', 502)
2025-02-20 12:44:08.819 WARNING (MainThread) [pymodbus.logging] Failed to connect [Errno 111] Connect call failed ('192.168.0.26', 502)
2025-02-20 12:47:20.349 WARNING (MainThread) [pymodbus.logging] Failed to connect [Errno 111] Connect call failed ('192.168.0.26', 502)
2025-02-20 12:49:08.833 WARNING (MainThread) [pymodbus.logging] Failed to connect [Errno 111] Connect call failed ('192.168.0.26', 502)
2025-02-20 12:52:20.352 WARNING (MainThread) [pymodbus.logging] Failed to connect [Errno 111] Connect call failed ('192.168.0.26', 502)
2025-02-20 12:54:08.856 WARNING (MainThread) [pymodbus.logging] Failed to connect [Errno 111] Connect call failed ('192.168.0.26', 502)
2025-02-20 12:57:20.354 WARNING (MainThread) [pymodbus.logging] Failed to connect [Errno 111] Connect call failed ('192.168.0.26', 502)
2025-02-20 12:59:08.858 WARNING (MainThread) [pymodbus.logging] Failed to connect [Errno 111] Connect call failed ('192.168.0.26', 502)
2025-02-20 13:02:20.357 WARNING (MainThread) [pymodbus.logging] Failed to connect [Errno 111] Connect call failed ('192.168.0.26', 502)
2025-02-20 13:04:08.860 WARNING (MainThread) [pymodbus.logging] Failed to connect [Errno 111] Connect call failed ('192.168.0.26', 502)`

Pho3niX90, 2025-02-20:
> > First impression: the latest version seems the most stable version in weeks. Modbus connection stays up-and-running for the longest time now in weeks.
> 
> Thanks! Waiting for tomorrow, if you dont have any addional trouble tilll then, i will have a try too..
> 
> I hope 3.0 also fixes the "not permanent" but the recurring loses of connection... this is at 2.1.2, it fetches back with no permanent loss, but i think this is not the way it should work and maybe sometimes leads to a "full crash" of the connection...:
> 
> `2025-02-20 12:02:20.326 WARNING (MainThread) [pymodbus.logging] Failed to connect [Errno 111] Connect call failed ('192.168.0.26', 502) 2025-02-20 12:04:08.670 WARNING (MainThread) [pymodbus.logging] Failed to connect [Errno 111] Connect call failed ('192.168.0.26', 502) 2025-02-20 12:07:20.328 WARNING (MainThread) [pymodbus.logging] Failed to connect [Errno 111] Connect call failed ('192.168.0.26', 502) 2025-02-20 12:09:08.677 WARNING (MainThread) [pymodbus.logging] Failed to connect [Errno 111] Connect call failed ('192.168.0.26', 502) 2025-02-20 12:12:20.332 WARNING (MainThread) [pymodbus.logging] Failed to connect [Errno 111] Connect call failed ('192.168.0.26', 502) 2025-02-20 12:14:08.684 WARNING (MainThread) [pymodbus.logging] Failed to connect [Errno 111] Connect call failed ('192.168.0.26', 502) 2025-02-20 12:17:20.334 WARNING (MainThread) [pymodbus.logging] Failed to connect [Errno 111] Connect call failed ('192.168.0.26', 502) 2025-02-20 12:19:08.723 WARNING (MainThread) [pymodbus.logging] Failed to connect [Errno 111] Connect call failed ('192.168.0.26', 502) 2025-02-20 12:22:20.337 WARNING (MainThread) [pymodbus.logging] Failed to connect [Errno 111] Connect call failed ('192.168.0.26', 502) 2025-02-20 12:24:08.737 WARNING (MainThread) [pymodbus.logging] Failed to connect [Errno 111] Connect call failed ('192.168.0.26', 502) 2025-02-20 12:27:20.340 WARNING (MainThread) [pymodbus.logging] Failed to connect [Errno 111] Connect call failed ('192.168.0.26', 502) 2025-02-20 12:29:08.774 WARNING (MainThread) [pymodbus.logging] Failed to connect [Errno 111] Connect call failed ('192.168.0.26', 502) 2025-02-20 12:32:20.341 WARNING (MainThread) [pymodbus.logging] Failed to connect [Errno 111] Connect call failed ('192.168.0.26', 502) 2025-02-20 12:34:08.777 WARNING (MainThread) [pymodbus.logging] Failed to connect [Errno 111] Connect call failed ('192.168.0.26', 502) 2025-02-20 12:37:20.345 WARNING (MainThread) [pymodbus.logging] Failed to connect [Errno 111] Connect call failed ('192.168.0.26', 502) 2025-02-20 12:39:08.797 WARNING (MainThread) [pymodbus.logging] Failed to connect [Errno 111] Connect call failed ('192.168.0.26', 502) 2025-02-20 12:42:20.347 WARNING (MainThread) [pymodbus.logging] Failed to connect [Errno 111] Connect call failed ('192.168.0.26', 502) 2025-02-20 12:44:08.819 WARNING (MainThread) [pymodbus.logging] Failed to connect [Errno 111] Connect call failed ('192.168.0.26', 502) 2025-02-20 12:47:20.349 WARNING (MainThread) [pymodbus.logging] Failed to connect [Errno 111] Connect call failed ('192.168.0.26', 502) 2025-02-20 12:49:08.833 WARNING (MainThread) [pymodbus.logging] Failed to connect [Errno 111] Connect call failed ('192.168.0.26', 502) 2025-02-20 12:52:20.352 WARNING (MainThread) [pymodbus.logging] Failed to connect [Errno 111] Connect call failed ('192.168.0.26', 502) 2025-02-20 12:54:08.856 WARNING (MainThread) [pymodbus.logging] Failed to connect [Errno 111] Connect call failed ('192.168.0.26', 502) 2025-02-20 12:57:20.354 WARNING (MainThread) [pymodbus.logging] Failed to connect [Errno 111] Connect call failed ('192.168.0.26', 502) 2025-02-20 12:59:08.858 WARNING (MainThread) [pymodbus.logging] Failed to connect [Errno 111] Connect call failed ('192.168.0.26', 502) 2025-02-20 13:02:20.357 WARNING (MainThread) [pymodbus.logging] Failed to connect [Errno 111] Connect call failed ('192.168.0.26', 502) 2025-02-20 13:04:08.860 WARNING (MainThread) [pymodbus.logging] Failed to connect [Errno 111] Connect call failed ('192.168.0.26', 502)`

The modbus connection in v3 is the most table it can get, so if the connection loss is due to an overly busy modbus, then it should resolve it.

hbpv, 2025-02-20:
@Pho3niX90 you got me too interessted - i will try now.

another offtopic question: Why do i have values with zero decimals (asspecially for statistics, with today, total etc..) in the home assistant energy dashboard? I other integrations i have at least 1 decimal for kW und up to 3 decimals for W. This makes the dashboard and value to inaccurate for statisticts, when i produced 1,9 kW it writes 1 till 2 is reached.

what i've already tried to was this beneath, but did'nt fix the problem, as value's still store at ,00 level.

[image]

hbpv, 2025-02-20:
First look is, that with 3.0.2 there are no connection problems for about 10 minutes - look great! only some Warnings:

2025-02-20 13:26:39.634 WARNING (MainThread) [homeassistant.helpers.frame] Detected that custom integration 'solis_modbus' sets option flow config_entry explicitly, which is deprecated at custom_components/solis_modbus/config_flow.py, line 118: self.config_entry = config_entry. This will stop working in Home Assistant 2025.12, please create a bug report at https://github.com/Pho3niX90/solis_modbus/issues

2025-02-20 13:27:09.172 WARNING (MainThread) [homeassistant.components.sensor] Entity sensor.solis_s6_solis_off_grid_overdischarge_soc (<class 'custom_components.solis_modbus.sensors.solis_sensor.SolisSensor'>) is using native unit of measurement '%' which is not a valid unit for the device class ('current') it is using; expected one of ['A', 'mA']; Please update your configuration if your entity is manually configured, otherwise create a bug report at https://github.com/Pho3niX90/solis_modbus/issues

2025-02-20 13:27:09.392 WARNING (MainThread) [homeassistant.components.sensor] Entity number.solis_s6_eh3p_off_grid_overdischarge_soc (<class 'custom_components.solis_modbus.sensors.solis_number_sensor.SolisNumberEntity'>) is using native unit of measurement '%' which is not a valid unit for the device class ('current') it is using; expected one of ['A', 'mA']; Please update your configuration if your entity is manually configured, otherwise create a bug report at https://github.com/Pho3niX90/solis_modbus/issues

hbpv, 2025-02-20:
battery charge power is not updated since 37 minutes.. so theres no new values coming in...

sensor.solis_s6_solis_battery_charge_power

debietim, 2025-02-20:
Now I lost connection as well...[image]

Pho3niX90, 2025-02-20:
in regards to 
`sensor.solis_s6_solis_battery_charge_power
sensor.solis_s6_solis_battery_discharge_power`

I will investigate these, they are derived sensors, and might have forgotten to add support back in v3 of the code.

Pho3niX90, 2025-02-20:
> Now I lost connection as well...[image]

do you have logs?

Pho3niX90, 2025-02-20:
> Yes please as soon as you able, because sensor.solis_s6_solis_battery_power is always positive value. If this sensor would be positive/negativ in terms of charge and discharge i could handle it.. but it is Not.
> 
> So we Need the charge/discharge values back 😅

Should fix it https://github.com/Pho3niX90/solis_modbus/releases/tag/3.0.0-beta-4

hbpv, 2025-02-20:
> Should fix it https://github.com/Pho3niX90/solis_modbus/releases/tag/3.0.0-beta-4

Nope… still no value Updates…

[image]

Pho3niX90, 2025-02-20:
@hbpv thanks for the logs, here is the fix https://github.com/Pho3niX90/solis_modbus/releases/tag/3.0.0-beta-5

hbpv, 2025-02-20:
> [@hbpv](https://github.com/hbpv) thanks for the logs, here is the fix https://github.com/Pho3niX90/solis_modbus/releases/tag/3.0.0-beta-5

2025-02-20 20:47:00.016 WARNING (SyncWorker_0) [homeassistant.loader] We found a custom integration dreame_vacuum which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you experience issues with Home Assistant
2025-02-20 20:47:00.017 WARNING (SyncWorker_0) [homeassistant.loader] We found a custom integration hacs which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you experience issues with Home Assistant
2025-02-20 20:47:00.018 WARNING (SyncWorker_0) [homeassistant.loader] We found a custom integration solis_modbus which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you experience issues with Home Assistant
2025-02-20 20:47:00.018 WARNING (SyncWorker_0) [homeassistant.loader] We found a custom integration frigate which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you experience issues with Home Assistant
2025-02-20 20:47:00.019 WARNING (SyncWorker_0) [homeassistant.loader] We found a custom integration network_scanner which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you experience issues with Home Assistant
2025-02-20 20:47:02.913 WARNING (MainThread) [homeassistant.components.sensor] Entity sensor.solis_s6_solis_off_grid_overdischarge_soc (<class 'custom_components.solis_modbus.sensors.solis_sensor.SolisSensor'>) is using native unit of measurement '%' which is not a valid unit for the device class ('current') it is using; expected one of ['A', 'mA']; Please update your configuration if your entity is manually configured, otherwise create a bug report at https://github.com/Pho3niX90/solis_modbus/issues
2025-02-20 20:47:02.926 ERROR (MainThread) [homeassistant.components.binary_sensor] rpi_power: Error on device update!
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/helpers/entity_platform.py", line 738, in _async_add_entity
    await entity.async_device_update(warning=False)
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 1320, in async_device_update
    await hass.async_add_executor_job(self.update)
  File "/usr/local/lib/python3.13/concurrent/futures/thread.py", line 59, in run
    result = self.fn(*self.args, **self.kwargs)
  File "/usr/src/homeassistant/homeassistant/components/rpi_power/binary_sensor.py", line 53, in update
    value = self._under_voltage.get()
            ^^^^^^^^^^^^^^^^^^^^^^^
AttributeError: 'NoneType' object has no attribute 'get'
2025-02-20 20:47:03.072 WARNING (MainThread) [homeassistant.components.sensor] Entity number.solis_s6_eh3p_off_grid_overdischarge_soc (<class 'custom_components.solis_modbus.sensors.solis_number_sensor.SolisNumberEntity'>) is using native unit of measurement '%' which is not a valid unit for the device class ('current') it is using; expected one of ['A', 'mA']; Please update your configuration if your entity is manually configured, otherwise create a bug report at https://github.com/Pho3niX90/solis_modbus/issues
2025-02-20 20:47:08.814 ERROR (MainThread) [homeassistant.helpers.template] Template variable error: 'state' is undefined when rendering '{{ (state('sensor.solis_s6_solis_pv_power_1') / 8 ) |round(2) }}"'
2025-02-20 20:47:08.815 ERROR (MainThread) [homeassistant.helpers.event] Error while processing template: Template<template=({{ (state('sensor.solis_s6_solis_pv_power_1') / 8 ) |round(2) }}") renders=2>
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/helpers/template.py", line 643, in async_render
    render_result = _render_with_context(self.template, compiled, **kwargs)
  File "/usr/src/homeassistant/homeassistant/helpers/template.py", line 2756, in _render_with_context
    return template.render(**kwargs)
           ~~~~~~~~~~~~~~~^^^^^^^^^^
  File "/usr/local/lib/python3.13/site-packages/jinja2/environment.py", line 1295, in render
    self.environment.handle_exception()
    ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^^
  File "/usr/local/lib/python3.13/site-packages/jinja2/environment.py", line 942, in handle_exception
    raise rewrite_traceback_stack(source=source)
  File "<template>", line 1, in top-level template code
  File "/usr/local/lib/python3.13/site-packages/jinja2/sandbox.py", line 399, in call
    if not __self.is_safe_callable(__obj):
           ~~~~~~~~~~~~~~~~~~~~~~~^^^^^^^
  File "/usr/src/homeassistant/homeassistant/helpers/template.py", line 3173, in is_safe_callable
    ) or super().is_safe_callable(obj)
         ~~~~~~~~~~~~~~~~~~~~~~~~^^^^^
  File "/usr/local/lib/python3.13/site-packages/jinja2/sandbox.py", line 265, in is_safe_callable
    getattr(obj, "unsafe_callable", False) or getattr(obj, "alters_data", False)
    ~~~~~~~^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/helpers/template.py", line 2788, in _fail_with_undefined_error
    return super()._fail_with_undefined_error(*args, **kwargs)
           ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^^^
jinja2.exceptions.UndefinedError: 'state' is undefined

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
[... cut ...]
2025-02-20 20:47:09.313 WARNING (MainThread) [pymodbus.logging] Failed to connect [Errno 111] Connect call failed ('192.168.0.26', 502)
2025-02-20 20:47:11.023 WARNING (MainThread) [pymodbus.logging] Failed to connect [Errno 111] Connect call failed ('192.168.0.26', 502)
2025-02-20 20:47:13.142 WARNING (MainThread) [pymodbus.logging] Failed to connect [Errno 111] Connect call failed ('192.168.0.26', 502)
2025-02-20 20:47:17.246 WARNING (MainThread) [pymodbus.logging] Failed to connect [Errno 111] Connect call failed ('192.168.0.26', 502)

hbpv, 2025-02-20:
Now After Dome initial (still Not solved) Connection erforscht, See Last Lines on the log, data ist pulled correctly again… will oversee it tomorrow

hbpv, 2025-02-21:
@Pho3niX90 

For my usecase it runs like charm. Since restarting in the morning there is not even 1 connection error.

Only problem that persists: After a restart (i have to do some for plugin updates etc.) this is always some lottery if:

- It needs 10-30 connections errors till it gets connected
- It will never connect, until you restart the system again

Maybe this is related to the soliscloud/modbus problem at a time, but maybe you could find a solution where we do not need to restart the system (maybe a pause in connection afford)

Best
Max

hbpv, 2025-02-23:
Initial connection Not solveable see #143 
Connection with 3.0.1 seems stable
Thanks - can be closed from my side..

closed 2025-02-23 by Pho3niX90

debietim, 2025-05-08:
Since the latest update this problem is back... after af few hours the connection with the inverter is broken and no new values are coming in.