Pho3niX90/solis_modbus #132: solis_modbus.write_holding_register fails to update the discharge current

ctepelea, 2025-02-15:
I am getting this error when doing the service call to solis_modbus.solis_write_holding_register. What could be the issue?

action: solis_modbus.solis_write_holding_register
metadata: {}
data:
  address: 43142
  value: >-
    {{ max(min(-(states("sensor.pm_net_power") | float /50 -
    states("number.solis_s6_solis_time_charging_discharge_current")|float) |
    round(1),100),4) * 10}}

[image]

Logger: homeassistant.components.automation.energy_solis_discharge_current_tracks_net_consumption
Source: helpers/script.py:526
integration: Automation (documentation, issues)
First occurred: 21:11:20 (294 occurrences)
Last logged: 22:51:20

Energy: Battery Discharge - Tracks Net Consumption: Error executing script. Unexpected error for call_service at pos 2: 'NodeDictClass' object has no attribute 'data'
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/helpers/script.py", line 526, in _async_step
    await getattr(self, handler)()
  File "/usr/src/homeassistant/homeassistant/helpers/script.py", line 762, in _async_call_service_step
    response_data = await self._async_run_long_action(
                    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
    ...<9 lines>...
    )
    ^
  File "/usr/src/homeassistant/homeassistant/helpers/script.py", line 727, in _async_run_long_action
    return await long_task
           ^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/core.py", line 2794, in async_call
    response_data = await coro
                    ^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/core.py", line 2844, in _execute_service
    return await self._hass.async_add_executor_job(target, service_call)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.13/concurrent/futures/thread.py", line 59, in run
    result = self.fn(*self.args, **self.kwargs)
  File "/config/custom_components/solis_modbus/__init__.py", line 29, in service_write_holding_register
    host = entry.data.get("host")
           ^^^^^^^^^^
AttributeError: 'NodeDictClass' object has no attribute 'data'

0rangutan, 2025-02-15:
Are you on the latest Solis firmware (6 rather than 3 charging slots)?

Pho3niX90, 2025-02-16:
This should resolve that https://github.com/Pho3niX90/solis_modbus/releases/tag/2.1.1

ctepelea, 2025-02-16:
Hello and thanks for the quick turnaround. However now the integration does not load at all, see below.

Logger: homeassistant.setup
Source: setup.py:409
First occurred: 10:26:50 (1 occurrences)
Last logged: 10:26:50

Just for reference if it makes a difference, I am using a Solis S5-EH1P6K-L-DC inverter with an S2 datalogger.

Error during setup of component solis_modbus: async_setup() takes 1 positional argument but 2 were given
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/setup.py", line 409, in _async_setup_component
    task = component.async_setup(hass, processed_config)
TypeError: async_setup() takes 1 positional argument but 2 were given

[image]

Pho3niX90, 2025-02-16:
I have released a fix for this as well now, an oversight from my part:
https://github.com/Pho3niX90/solis_modbus/releases/tag/2.1.2

ctepelea, 2025-02-16:
Thanks. That's really great speed. Unfortunately I still can't make it work.

Executed: 16 February 2025 at 16:54:50
Error: 'str' object has no attribute 'write_holding_register'
Result:
params:
  domain: solis_modbus
  service: solis_write_holding_register
  service_data:
    address: 43141
    value: 30
  target: {}
running_script: false

[image]

Logger: homeassistant.components.automation.energy_solis_charge_tracks_ac_total_output
Source: helpers/script.py:526
integration: Automation ([documentation](https://www.home-assistant.io/integrations/automation), [issues](https://github.com/home-assistant/core/issues?q=is%3Aissue+is%3Aopen+label%3A%22integration%3A+automation%22))
First occurred: 13:17:30 (1315 occurrences)
Last logged: 16:56:30

Energy: Solar Charge - Track AC Total Output: Error executing script. Unexpected error for call_service at pos 2: 'str' object has no attribute 'write_holding_register'
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/helpers/script.py", line 526, in _async_step
    await getattr(self, handler)()
  File "/usr/src/homeassistant/homeassistant/helpers/script.py", line 762, in _async_call_service_step
    response_data = await self._async_run_long_action(
                    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
    ...<9 lines>...
    )
    ^
  File "/usr/src/homeassistant/homeassistant/helpers/script.py", line 727, in _async_run_long_action
    return await long_task
           ^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/core.py", line 2794, in async_call
    response_data = await coro
                    ^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/core.py", line 2844, in _execute_service
    return await self._hass.async_add_executor_job(target, service_call)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.13/concurrent/futures/thread.py", line 59, in run
    result = self.fn(*self.args, **self.kwargs)
  File "/config/custom_components/solis_modbus/__init__.py", line 36, in service_write_holding_register
    hass.create_task(controller.write_holding_register(address, value))
                     ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
AttributeError: 'str' object has no attribute 'write_holding_register'

Pho3niX90, 2025-02-16:
@ctepelea can you use it from the developer view? I believe it's `script.solis_write_holding_register`

ctepelea, 2025-02-16:
[image]

Logger: homeassistant.helpers.script.websocket_api_script
Source: helpers/script.py:526
First occurred: 19:17:57 (6 occurrences)
Last logged: 19:21:07

websocket_api script: Error executing script. Unexpected error for call_service at pos 1: 'str' object has no attribute 'write_holding_register'
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/helpers/script.py", line 526, in _async_step
    await getattr(self, handler)()
  File "/usr/src/homeassistant/homeassistant/helpers/script.py", line 762, in _async_call_service_step
    response_data = await self._async_run_long_action(
                    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
    ...<9 lines>...
    )
    ^
  File "/usr/src/homeassistant/homeassistant/helpers/script.py", line 727, in _async_run_long_action
    return await long_task
           ^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/core.py", line 2794, in async_call
    response_data = await coro
                    ^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/core.py", line 2844, in _execute_service
    return await self._hass.async_add_executor_job(target, service_call)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.13/concurrent/futures/thread.py", line 59, in run
    result = self.fn(*self.args, **self.kwargs)
  File "/config/custom_components/solis_modbus/__init__.py", line 36, in service_write_holding_register
    hass.create_task(controller.write_holding_register(address, value))
                     ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
AttributeError: 'str' object has no attribute 'write_holding_register'

Pho3niX90, 2025-02-16:
I am investigating this, I believe this became an issue when I tried to better the multiple inverter support. 

In the meantime, it might be easier to set the actual `number.solis_inverter_time_charging_charge_current` for this example.

ctepelea, 2025-02-16:
Before I upgraded to 2.0, setting the number.solis_inverter_time_charging_charge_current was resulting in the inverters charge/discharge current being affecter in 8 out of 10 calls. And I was happy, it took maybe 10 seconds or so but it was working. After upgrading to the 2.0 version, setting the current via the input entity takes a very long time and 9 out of 10 the integration re-reads the currently set value in solis_s6_solis_time_charging_charge_current back to the number.solis_inverter_time_charging_charge_current before the newly set value is written to the inverter. So this is not working.

The workaround I thought about was to go back to the fbondy/modbus implementation and then it works very fast because I limited the number of registers being read/written. I cannot have also your integration and the modbus native HA working at the same time as you cannot have two connection simultaneously.

This is when I decided to try to write directly via solis_modbus.service_write_holding_register the charge current/discharge current register.

I hope this makes sense to you. If you want to try something is to prioritise writing the charge/discharge current registers when the respective input_number changes.

Pho3niX90, 2025-02-17:
Which inverter are you currently using?

Could you provide me in txt format an example that I can test my side to diagnose the issue?

ctepelea, 2025-02-17:
I am using a Solis S5-EH1P6K-L-DC inverter with an S2 datalogger.

"Could you provide me in txt format an example that I can test my side to diagnose the issue?" - what exactly should I be providing, sorry I don't understand.

I am on 2.1.2 but whatever values I may set for charge/discharte current, any EPM toggles, they revert to their previous values in about a minute or so.

ctepelea, 2025-02-21:
Hello, any ideas on how to solve this? I am not getting any values sent to the inverter. Reading works.

Pho3niX90, 2025-02-21:
> Hello, any ideas on how to solve this? I am not getting any values sent to the inverter. Reading works.

Have you tried the latest beta version as yet? There were a lot of changes in regards to the way things are written.

ctepelea, 2025-03-01:
With version 3.1.1 the issue is closed, writing works timely and flawlessly.

PR #133 (2025-02-16): Fix service call and enhance Modbus controller handling

### **User description**
fix: modbus switch
fix: service call

## 🔄 Related Issues
Closes #132

___

### **PR Type**
Bug fix, Enhancement

___

### **Description**
- Fixed service call issue in `solis_write_holding_register`.

- Enhanced handling for Modbus controller connection management.

- Updated logic to support multiple controllers in service calls.

- Improved code readability and maintainability in `switch.py`.

___

### **Changes walkthrough** 📝
<table><thead><tr><th></th><th align="left">Relevant files</th></tr></thead><tbody> [... cut]
commit 5ea890c7 (2025-02-16): fix: service call, fixes #132 (#133)

fix: modbus switch
closed 2025-03-01 by Pho3niX90