Pho3niX90/solis_modbus #135: [Bug]: After update to 2.1.1 on Ha, Integration is Not loading

hbpv, 2025-02-16:
[image]

2025-02-16 09:10:43.323 WARNING (SyncWorker_0) [homeassistant.loader] We found a custom integration dreame_vacuum which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you experience issues with Home Assistant
2025-02-16 09:10:43.324 WARNING (SyncWorker_0) [homeassistant.loader] We found a custom integration hacs which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you experience issues with Home Assistant
2025-02-16 09:10:43.325 WARNING (SyncWorker_0) [homeassistant.loader] We found a custom integration solis_modbus which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you experience issues with Home Assistant
2025-02-16 09:10:43.325 WARNING (SyncWorker_0) [homeassistant.loader] We found a custom integration frigate which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you experience issues with Home Assistant
2025-02-16 09:10:43.325 WARNING (SyncWorker_0) [homeassistant.loader] We found a custom integration network_scanner which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you experience issues with Home Assistant
2025-02-16 09:10:45.837 ERROR (MainThread) [homeassistant.setup] Error during setup of component solis_modbus: async_setup() takes 1 positional argument but 2 were given
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/setup.py", line 409, in _async_setup_component
    task = component.async_setup(hass, processed_config)
TypeError: async_setup() takes 1 positional argument but 2 were given
2025-02-16 09:10:46.125 ERROR (MainThread) [homeassistant.components.binary_sensor] rpi_power: Error on device update!
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/helpers/entity_platform.py", line 738, in _async_add_entity
    await entity.async_device_update(warning=False)
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 1320, in async_device_update
    await hass.async_add_executor_job(self.update)
  File "/usr/local/lib/python3.13/concurrent/futures/thread.py", line 59, in run
    result = self.fn(*self.args, **self.kwargs)
  File "/usr/src/homeassistant/homeassistant/components/rpi_power/binary_sensor.py", line 53, in update
    value = self._under_voltage.get()
            ^^^^^^^^^^^^^^^^^^^^^^^
AttributeError: 'NoneType' object has no attribute 'get'

Pho3niX90, 2025-02-16:
That isn't related to this plugin. Seems to be https://www.home-assistant.io/integrations/rpi_power/
```
File "/usr/src/homeassistant/homeassistant/components/rpi_power/binary_sensor.py", line 53, in update
value = self._under_voltage.get()
^^^^^^^^^^^^^^^^^^^^^^^
AttributeError: 'NoneType' object has no attribute 'get'
```

closed 2025-02-16 by Pho3niX90

hbpv, 2025-02-16:
But the other errors we’re Modus related… is this fixed now?

commit e0c34cf3 (2025-02-16): fix async setup closes #135