Pho3niX90/solis_modbus #136: Solis modbus fails to connect again

miketarg, 2025-02-16:
After an update this morning and after restarting it's failing to connect.

## 🔄 To Reproduce
It's not connecting, note this isn't just taking time to connect, it never does. Waiting for over 30 mins now.

## ✅ Expected Behavior
To connect 
## 📸 Screenshots / Logs

[image]

[image]

## 🔧 System Information
### 🔹 Home Assistant:
Core
2025.2.4
Supervisor
2025.02.1
Operating System
14.2
Frontend
20250214.0

### 🔹 Integration:
- **Plugin Version**: [e.g., 1.2.3]
- **Device Model**: [e.g., Solis 5G 6kW, Axitec 10kW]
- **Modbus Connection Type 

[image]

Ethernet
- **Custom Config Entries?**: [Yes/No]

## ➕ Additional Context
Any other details that might be helpful for debugging.

miketarg, 2025-02-16:
Update, it's finally connected all off a sudden 😐

What's causing this long delay in getting a successful connection?

Pho3niX90, 2025-02-16:
> Update, it's finally connected all off a sudden 😐
> 
> What's causing this long delay in getting a successful connection?

It's unfortunately because the dongle(solis cloud), and HA, battles for connection, which can cause delays in connection.

miketarg, 2025-02-16:
Is there a way to stop the cloud from trying to connect?