Pho3niX90/solis_modbus #138: [Bug]: All entity values jump full scale or zero randomly

Ian-Johnston, 2025-02-18:
## 🐛 Bug Description
Around 15 times in any 12 period all Solis entity values will spike to full scale or zero and back again.

## 🔄 To Reproduce
No user interaction required.

## ✅ Expected Behavior
Would rather no value was recorder, or retain last value.

## 📸 Screenshots / Logs
See below

## 🔧 System Information
See below

### 🔹 Integration:
See below

## ➕ Additional Context
See below

Every few hours or so it appears the connection to my inverter drops (Waveshare RS485 to Eth(B) issue - not sure) and my entities record a full scale value, or sometimes drop to zero value.
Example: The history graphs for all the entities, battery SOC, Solis Grid Frequency, any of the fixed settings entities etc all record the same thing.

Looking at one of the history graphs this happens about 15 times over a 12hr period.

All Solis_Modbus sensor entities are affected.
SolisCloud data does not experience the same issue.

Would prefer no value recorded or perhaps better, retain last known good value rather than escalating full scale or dropping to zero.

Core = 2025.2.4
Solis_Modbus = 1.6.2

Using Solis_Modbus S6 Waveshare integration type albeit mine is a slightly different model.

Running latest firmware 1.490 on my Waveshare devices.

Below is the condensed log which appears to coincide with one such event.

Logger: pymodbus.logging
Source: custom_components/solis_modbus/modbus_controller.py:52
integration: Solis Modbus ([documentation](https://solis-modbus.readthedocs.io/), [issues](https://github.com/Pho3niX90/solis_modbus/issues))
First occurred: 17 February 2025 at 23:56:30 (2 occurrences)
Last logged: 03:14:45
No response received after 3 retries, continue with next request

[image]

Pho3niX90, 2025-02-18:
This is strange behavior. Could you please upload debug logs so that I can see what the actual modbus is providing?

Pho3niX90, 2025-02-18:
@0rangutan have you seen the same behavior your side?

Ian-Johnston, 2025-02-18:
> This is strange behavior. Could you please upload debug logs so that I can see what the actual modbus is providing?

See attached partial debug log.
It incorporates a single spike from approx. 68% to Max at 10:16:20 and returns to normal 10:16:50 according to the history graph.
Battery SOC = sensor.solis_s6_solis_battery_soc

I haven't analyzed the log......I'm pretty much a HA newbie.

Thanks,

Ian.

[Partial_Debug_Log1.log](https://github.com/user-attachments/files/18845434/Partial_Debug_Log1.log)

Pho3niX90, 2025-02-18:
So reviewing the logs, I do see there are about 2 frames worth of 100%:

[image]

I did notice some other sensors also "reacting to this"

[image]

I am wondering, what is your BMS doing at this stage? Is it possible that your BMS is reporting incorrect information? Do you perhaps have 2 battery banks where the 2 values are getting confused, as I did notice the SOH also jumps between 97 and 100, for the same frame, the battery power also jumps from around ~200 to 1470

0rangutan, 2025-02-18:
> [@0rangutan](https://github.com/0rangutan) have you seen the same behavior your side?

No, I've not seen this on my system

Pho3niX90, 2025-02-18:
> I have just moved from my own Windows app that I wrote and using SolisCloud.com data retrieval via the API only, and it has been perfect for 2 years. Well, maybe the odd once per week spike to max only which others I know get also. Solis support said they couldn't find anything wrong albeit they did tweak stuff at their end.
> 
> It's only after I migrated to HA and achieving direct access data via the Waveshare devices that immediately out the box I get all these spikes. I do monitor SolisCloud.com in HA via the API also, and the sensor data there is perfect, it does NOT mirror the spikes at all.
> 
> There is only one Pylontech battery bank (10kWh) connected to the Solis inverter. I don't have any comms set up direct to the battery.
> 
> Where do I go from here?
> 
> Ian.

If you can leave the debug on for a while, and upload a larger log I can perhaps see what else is causing this. 

What are all the sensors that you observe this on?

Ian-Johnston, 2025-02-18:
Here is the full log data attached, I think I enabled it last thing last night and it ran right through the night until I pulled it today.
You could say from between 11pm last night and mid-day today the system ran unattended.
(Prior to 11pm last night I was coding and modifying HA non-stop).

I haven't paid too much attention to the other sensor, so can't say with any accuracy......I only noticed SOC has a problem because I have it on a history chart.

But looking at some of the sensors just now, maybe I could say as follows:
Solis S6 Solis Battery Power = At around 3am as the battery was overnight charging the power dropped to zero several times.
Solis S6 Solis A Phase Voltage = doesn't seem to spike at all.
Solis S6 Solis Grid Frequency = doesn't seem to spike at all.
Solis S6 Solis LLC Bus Voltage = Appears to drop right down in value 5:30am

I'm running some test code right now on sensor.py, I've hacked in some spike filtering logic on the modbus register for Battery SOC (I hope!). Been running for the past 45mins without issue albeit early days yet!
I also want to try replacing the Waveshare RS485 to ETH(B) TCP Server module with a Protoss PE11-M that I have and see if that makes any difference.

On my WIndows app I did NOT use the Waveshare modules, I got all my data from SolisCloud.com via the API.
As I said, I have the SolisCloud API in HA also, albeit disabled for now, and it's data does not spike including Battery SOC.
The issue seems to be through Waveshare only.......and if you google it, there have been reports with the Waveshare modules of spiking data, albeit the older formware. I have reported my issue to Waveshare.

[home-assistant.zip](https://github.com/user-attachments/files/18847706/home-assistant.zip)

Pho3niX90, 2025-02-18:
I know @0rangutan is also using the eth-b https://github.com/alienatedsec/solis-ha-modbus-cloud/discussions/24

Wondering if it's caused by EMI, I have seen this happen before on my Seplos bms modbus, I eventually replaced the ethernet cable, although the old one was twisted as well.

Ian-Johnston, 2025-02-18:
Interesting.....I'll take a look at the grounding etc make sure that's all intact around the inverter and the LAN components between it and my server cabinet/router.

PS. 2hrs now with the spike filtering logic in place and all good so far!

Ian.

Ian-Johnston, 2025-02-18:
I swapped out the TCP Server Waveshare for a Protoss PE11-M module and I still got spiking.
So I don't think it's it's a Waveshare issue, albeit I left the TCP Client module as a Waveshare.

Ian.

Ian-Johnston, 2025-02-19:
Until I get to the root problem, I have implemented spike detection on Battery SOC in sensor.py.
The spikes only ever last 1 or 2 cycles, so 15 or 30secs.
Green arrow in pic below is when I turned it on.

```
        # New attributes for spike filtering specific to battery SOC sensor (register 33139)
        self._spike_counter = 0
        self._last_valid_value = None
```

```
            # === Begin Spike Filtering Logic for battery SOC sensor (register 33139) ===
            if "33139" in [str(reg) for reg in self._register]:
                if n_value not in (0, 100):
                    self._spike_counter = 0
                    self._last_valid_value = n_value
                else:
                    self._spike_counter += 1
                    if self._spike_counter < 3:
                        _LOGGER.debug(
                            f"Ignoring short spike value {n_value} for battery SOC sensor; "
                            f"retaining previous value {self._last_valid_value} (counter={self._spike_counter})"
                        )
                        n_value = self._last_valid_value if self._last_valid_value is not None else n_value
                    else:
                        _LOGGER.debug(
                            f"Accepting persistent spike value {n_value} for battery SOC sensor after {self._spike_counter} cycles"
                        )
                        self._last_valid_value = n_value
                        self._spike_counter = 0
            # === End Spike Filtering Logic ===
```

[image]

Pho3niX90, 2025-02-20:
@Ian-Johnston regarding the waveshare config you are using. It was implemented with #75 , are you values correct for those sensors? If not, the normal hybrid config will fix that.

I am not sure as yet the reason behind the higher values, but seems not all waveshare users are experiencing the same

closed 2025-02-23 by Pho3niX90