Pho3niX90/solis_modbus #139: [Integration]: Solis RAI-3K-48ES-5G Battery Inverter

Ian-Johnston, 2025-02-18:
## 🔗 Inverter Model Details
Solis
RAI-3K-48ES-5G
Modbus port on inverter with S2-WL-ST, currently using dual Waveshare RS485 to ETH(B) modules for dual soliscloud.com/local access.

## 🔍 Why is this Important?
I am currently using the Solis S6 config in solis_modbus which I believe is almost the same. The RAI-3K-48ES-5G however is a battery inverter only, there is no PV functionality.

## 📄 Documentation / Modbus Registers
I did ask Solis for the Modbus register document for this particular inverter but they just gave me the generic one....which I am sure you probably have.

## ➕ Additional Context
The Solis S6 Waveshare config works after deleting all the PV sensors, and also modifying time.py per below.
Other than that so far I haven't seen anything else that doesn't work.

The RAI-3K-48ES-5G has only 3 Charge Start/End & Discharge Start/End slots.

```
    timeent = [
        {"type": "STE", "name": "Solis Time-Charging Charge Start (Slot 1)", "register": 43143, "enabled": True},
        {"type": "STE", "name": "Solis Time-Charging Charge End (Slot 1)", "register": 43145, "enabled": True},
        {"type": "STE", "name": "Solis Time-Charging Discharge Start (Slot 1)", "register": 43147, "enabled": True},
        {"type": "STE", "name": "Solis Time-Charging Discharge End (Slot 1)", "register": 43149, "enabled": True},

        {"type": "STE", "name": "Solis Time-Charging Charge Start (Slot 2)", "register": 43153, "enabled": True},
        {"type": "STE", "name": "Solis Time-Charging Charge End (Slot 2)", "register": 43155, "enabled": True},
        {"type": "STE", "name": "Solis Time-Charging Discharge Start (Slot 2)", "register": 43157, "enabled": True},
        {"type": "STE", "name": "Solis Time-Charging Discharge End (Slot 2)", "register": 43159, "enabled": True},

        {"type": "STE", "name": "Solis Time-Charging Charge Start (Slot 3)", "register": 43163, "enabled": True},
        {"type": "STE", "name": "Solis Time-Charging Charge End (Slot 3)", "register": 43165, "enabled": True},
        {"type": "STE", "name": "Solis Time-Charging Discharge Start (Slot 3)", "register": 43167, "enabled": True},
        {"type": "STE", "name": "Solis Time-Charging Discharge End (Slot 3)", "register": 43169, "enabled": True},
    ]
```

[Solis_RAI-3K-48ES-5G Datasheet.pdf](https://github.com/user-attachments/files/18845758/Solis_RAI-3K-48ES-5G.Datasheet.pdf)

[Solis AC Coupled 5G Inverter User Manual.pdf](https://github.com/user-attachments/files/18845769/Solis.AC.Coupled.5G.Inverter.User.Manual.pdf)

Pho3niX90, 2025-02-18:
I am busy working on v3 of the code, where the setup will take into consideration which functionalities exist for each model. I do have a couple of versions of their registrars, none really give good guidance (probably intentionally) in regards to detection of functionality via modbus.

Pho3niX90, 2025-02-18:
Out of curiosity, what was happening when the PV was part of the registrars?

Ian-Johnston, 2025-02-18:
I was getting HA errors saying the registers were not available when it tried to read them......which is why I modified the code per above to stop that.

Ian.

Pho3niX90, 2025-02-20:
This should now be resolved in https://github.com/Pho3niX90/solis_modbus/releases/tag/3.0.0-beta-1

closed 2025-02-20 by Pho3niX90