Pho3niX90/solis_modbus #142: Solis RHI-6K-48ES-5G  Support

lethux, 2025-02-20:
Hi,
I'm trying to have this integration works with my Solis RHI-6K-48ES-5G and Datalogger wifi.
Connecting to Solis Inverter IP and both 502 or 8899 port, I see zero values for all the information.
How can I help you integrate that model?
More than happy to help.

Pho3niX90, 2025-02-21:
@lethux please try https://github.com/Pho3niX90/solis_modbus/releases/tag/3.0.0-beta-6

It seems the RHI was incorrectly marked as an energy inverter instead of a Hybrid. 

Initial connection might take up to 10mins, as solis inverters do battle between solis cloud and modbus. I am still trying to see how a faster connection can be established overall. 

Which Dongle are you using? as they are the important part, S3 I believe dongles do not have a modbus host

lethux, 2025-02-21:
trying 3.0.0.-beta-6:

Error handling request from 10.0.5.36
Traceback (most recent call last):
  File "/usr/local/lib/python3.13/site-packages/aiohttp/web_protocol.py", line 480, in _handle_request
    resp = await request_handler(request)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.13/site-packages/aiohttp/web_app.py", line 569, in _handle
    return await handler(request)
           ^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.13/site-packages/aiohttp/web_middlewares.py", line 117, in impl
    return await handler(request)
           ^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/http/security_filter.py", line 92, in security_filter_middleware
    return await handler(request)
           ^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/http/forwarded.py", line 83, in forwarded_middleware
    return await handler(request)
           ^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/http/request_context.py", line 26, in request_context_middleware
    return await handler(request)
           ^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/http/ban.py", line 86, in ban_middleware
    return await handler(request)
           ^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/http/auth.py", line 242, in auth_middleware
    return await handler(request)
           ^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/http/headers.py", line 32, in headers_middleware
    response = await handler(request)
               ^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/helpers/http.py", line 73, in handle
    result = await handler(request, **request.match_info)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/http/decorators.py", line 81, in with_admin
    return await func(self, request, *args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/config/config_entries.py", line 222, in post
    return await super().post(request, flow_id)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/http/data_validator.py", line 74, in wrapper
    return await method(view, request, data, *args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/helpers/data_entry_flow.py", line 121, in post
    result = await self._flow_mgr.async_configure(flow_id, data)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/data_entry_flow.py", line 340, in async_configure
    result = await self._async_configure(flow_id, user_input)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/data_entry_flow.py", line 387, in _async_configure
    result = await self._async_handle_step(
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
        flow, cur_step["step_id"], user_input
        ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
    )
    ^
  File "/usr/src/homeassistant/homeassistant/data_entry_flow.py", line 490, in _async_handle_step
    result: _FlowResultT = await getattr(flow, method)(user_input)
                           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/solis_modbus/config_flow.py", line 54, in async_step_user
    if await self._validate_config(user_input):
       ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/solis_modbus/config_flow.py", line 68, in _validate_config
    inverter_config: InverterConfig = next(
                                      ~~~~^
        (inv for inv in SOLIS_INVERTERS if inv["model"] == inverter_model), None
        ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
    )
    ^
  File "/config/custom_components/solis_modbus/config_flow.py", line 69, in <genexpr>
    (inv for inv in SOLIS_INVERTERS if inv["model"] == inverter_model), None
                                       ~~~^^^^^^^^^
TypeError: 'InverterConfig' object is not subscriptable

lethux, 2025-02-21:
for inverter RHI-6K-48ES-5G, what is best Inverter Model in drop down menu between RHI-1P and RHI-3P?

lethux, 2025-02-21:
trying beta 7.
I'm trying also to use port 8899, but I see some logs still pointing to 502.
Did you have port 502 written in code?

"2025-02-21 15:32:53.915 WARNING (MainThread) [pymodbus.logging] Failed to connect [Errno 111] Connect call failed ('10.0.x.x', 502)
2025-02-21 15:33:14.596 ERROR (MainThread) [pymodbus.logging] No response received after 5 retries, continue with next request
2025-02-21 15:33:14.596 ERROR (MainThread) [custom_components.solis_modbus.modbus_controller] Failed to read input register 35000: Modbus Error: [Input/Output] No response received after 5 retries, continue with next request
2025-02-21 15:33:24.166 WARNING (MainThread) [pymodbus.logging] Failed to connect [Errno 111] Connect call failed ('10.0.x.x', 502)

lethux, 2025-02-21:
also:
 [homeassistant.components.sensor] Entity number.solis_rhi_off_grid_overdischarge_soc (<class 'custom_components.solis_modbus.sensors.solis_number_sensor.SolisNumberEntity'>) is using native unit of measurement '%' which is not a valid unit for the device class ('current') it is using; expected one of ['mA', 'A']; Please update your configuration if your entity is manually configured, otherwise create a bug report at https://github.com/Pho3niX90/solis_modbus/issues

Pho3niX90, 2025-02-21:
> I've manually changed the port in 2 files to match 8899.
> 
> getting this error now: [homeassistant.components.sensor] Entity number.solis_rhi_off_grid_overdischarge_soc (<class 'custom_components.solis_modbus.sensors.solis_number_sensor.SolisNumberEntity'>) is using native unit of measurement '%' which is not a valid unit for the device class ('current') it is using; expected one of ['mA', 'A']; Please update your configuration if your entity is manually configured, otherwise create a bug report at https://github.com/Pho3niX90/solis_modbus/issues

Is it reading the inverter values now?

Pho3niX90, 2025-02-21:
> trying beta 7. I'm trying also to use port 8899, but I see some logs still pointing to 502. Did you have port 502 written in code?
>
> "2025-02-21 15:32:53.915 WARNING (MainThread) [pymodbus.logging] [...]

Plugin needs a reload when the port changes, otherwise it will keep using the old port.

lethux, 2025-02-21:
no value after 30 mins or more.. I've a long list of this:
025-02-21 16:41:42.892 ERROR (MainThread) [pymodbus.logging] No response received after 5 retries, continue with next request
2025-02-21 16:41:42.892 ERROR (MainThread) [custom_components.solis_modbus.modbus_controller] Failed to read holding register 43750: Modbus Error: [Input/Output] No response received after 5 retries, continue with next

It's a better message rather then having "cannot connect".

lethux, 2025-02-21:
it seems a datalogger V2 (YingZhen)
serial numbere starts with 4xxxxxx

Pho3niX90, 2025-02-21:
These are the ones that are known to work
https://www.solisinverters.com/global/accessories9/S2_WL_ST_gl.html

These do not
https://www.solisinverters.com/global/accessories5/GPRS&WiFi_Data_Logging_Stick_gl.html

Pho3niX90, 2025-02-21:
If you are using S2_WK_ST, then you are suppose to get data. 

1. What I would try, delete the current integration settings, and set it up with port 502, leave it for 15mins to see if all is well. 
2. If step 1 did nothing, do the same, this time with 8899, leave for 15mins again. 

I am not sure what it's currently connected to, but `Modbus Error: [Input/Output] No response received after 5 retries, continue with next` is the same as no connection at this point.

lethux, 2025-02-22:
at this point, I think the dataloggeer is a GPRS&WiFi_Data_Logging_Stick_gl.
Not sure, as nothing is printed on the case and nothing is reported in the Solis Cloud web/app.
Thanks for now, at least we fixed some minor bugs.
btw, very good integration.

Pho3niX90, 2025-02-22:
> at this point, I think the dataloggeer is a GPRS&WiFi_Data_Logging_Stick_gl. Not sure, as nothing is printed on the case and nothing is reported in the Solis Cloud web/app. Thanks for now, at least we fixed some minor bugs. btw, very good integration.

Not an issue, this is a good solution https://github.com/alienatedsec/solis-ha-modbus-cloud

closed 2025-02-22 by Pho3niX90