Pho3niX90/solis_modbus #143: [Enhancement]: Add 2 decimals to total calculations

hbpv, 2025-02-21:
hi there,

We seriously Need 2 decimals for Daily, Yesterday and total calculations of consumption, feedin, imported, Charged and discharge Energy. See examples attached: 1 hour round about 200 Watts household load, makes 0,0 kWh between 13:00-14:00.
Values in entities are always xx,00 wich is incorrect and leads to massively wrong statistics over the day..weeks… etc

[image]
[image]
[image]

Pho3niX90, 2025-02-21:
Home assistant should be taking care of this under the hood. 

can you post the values from the developer view:

[image]

Pho3niX90, 2025-02-21:
another example, the precision doesn't go to 2 decimals for totals

[image]

hbpv, 2025-02-21:
> another example, the precision doesn't go to 2 decimals for totals
> 
> [image]

Okay but the Most interessting ones are the totals as HA Energy Dashboard Uses them … and there you have 0 decimals, 1 decimal is Minimum to get solid statistics 😣 so is this a Solis Problem? I have some Goodwe integrations, there is everything Fine 

[image]
[image]

Heres data from Goodwe Inverter Installation:

[image]

Pho3niX90, 2025-02-22:
try https://github.com/Pho3niX90/solis_modbus/compare/3.0.0-beta-9...3.0.0-beta-10

it suggests to HA what the decimals should be

Pho3niX90, 2025-02-22:
does v3 resolve the decimals now that they are suggested to HA from the plugin side?

hbpv, 2025-02-22:
Hey.. sorry.. i‘m still on beta 5 and changed the enegrydashboard sources to Daily, where it already was ok. Will have a try to Jump to released 3.0 tomorrow and Tell you more..

hbpv, 2025-02-22:
Did it…

Problems After 3.0 release and Restart:

- Connection Errors („fight against soliscloud“) - duration: maybe 5 Minutes, now its working
- values total and year still without decimals, when entity Set to 2 decimals, its then always ,00
- battery bms current looks strange..
- some other errors After Restart:

2025-02-22 20:03:09.502 WARNING (SyncWorker_0) [homeassistant.loader] We found a custom integration dreame_vacuum which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you experience issues with Home Assistant
2025-02-22 20:03:09.502 WARNING (SyncWorker_0) [homeassistant.loader] We found a custom integration hacs which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you experience issues with Home Assistant
2025-02-22 20:03:09.503 WARNING (SyncWorker_0) [homeassistant.loader] We found a custom integration solis_modbus which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you experience issues with Home Assistant
2025-02-22 20:03:09.503 WARNING (SyncWorker_0) [homeassistant.loader] We found a custom integration frigate which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you experience issues with Home Assistant
2025-02-22 20:03:09.504 WARNING (SyncWorker_0) [homeassistant.loader] We found a custom integration network_scanner which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you experience issues with Home Assistant
2025-02-22 20:03:12.185 ERROR (MainThread) [homeassistant.components.sensor] Error adding entity sensor.solis_s6_solis_serial_number for domain sensor with platform solis_modbus
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/components/sensor/__init__.py", line 635, in state
    numerical_value = int(value)
ValueError: invalid literal for int() with base 10: '143315024A120021\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/components/sensor/__init__.py", line 638, in state
    numerical_value = float(value)
ValueError: could not convert string to float: '143315024A120021\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00'

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/helpers/entity_platform.py", line 928, in _async_add_entity
    await entity.add_to_platform_finish()
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 1384, in add_to_platform_finish
    self.async_write_ha_state()
    ~~~~~~~~~~~~~~~~~~~~~~~~~^^
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 1023, in async_write_ha_state
    self._async_write_ha_state()
    ~~~~~~~~~~~~~~~~~~~~~~~~~~^^
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 1148, in _async_write_ha_state
    self.__async_calculate_state()
    ~~~~~~~~~~~~~~~~~~~~~~~~~~~~^^
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 1085, in __async_calculate_state
    state = self._stringify_state(available)
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 1029, in _stringify_state
    if (state := self.state) is None:
                 ^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/sensor/__init__.py", line 642, in state
    raise ValueError(
    ...<5 lines>...
    ) from err
ValueError: Sensor sensor.solis_s6_solis_serial_number has device class 'None', state class 'None' unit 'None' and suggested precision '0' thus indicating it has a numeric value; however, it has the non-numeric value: '143315024A120021                ' (<class 'str'>)
2025-02-22 20:03:19.214 WARNING (MainThread) [pymodbus.logging] Failed to connect [Errno 111] Connect call failed ('192.168.0.26', 502)
2025-02-22 20:03:20.857 WARNING (MainThread) [pymodbus.logging] Failed to connect [Errno 111] Connect call failed ('192.168.0.26', 502)
2025-02-22 20:03:23.137 WARNING (MainThread) [pymodbus.logging] Failed to connect [Errno 111] Connect call failed ('192.168.0.26', 502)
2025-02-22 20:03:27.241 WARNING (MainThread) [pymodbus.logging] Failed to connect [Errno 111] Connect call failed ('192.168.0.26', 502)
2025-02-22 20:03:35.344 WARNING (MainThread) [pymodbus.logging] Failed to connect [Errno 111] Connect call failed ('192.168.0.26', 502)
2025-02-22 20:03:42.234 ERROR (DbWorker_0) [homeassistant] Error doing job: Task exception was never retrieved (None)
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/components/sensor/__init__.py", line 635, in state
    numerical_value = int(value)
ValueError: invalid literal for int() with base 10: '143315024A120021\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/components/sensor/__init__.py", line 638, in state
    numerical_value = float(value)
ValueError: could not convert string to float: '143315024A120021\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00'

[... cut ...]
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 1029, in _stringify_state
    if (state := self.state) is None:
                 ^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/sensor/__init__.py", line 642, in state
    raise ValueError(
    ...<5 lines>...
    ) from err
ValueError: Sensor sensor.solis_s6_solis_serial_number has device class 'None', state class 'None' unit 'None' and suggested precision '0' thus indicating it has a numeric value; however, it has the non-numeric value: '143315024A120021                ' (<class 'str'>)

Pho3niX90, 2025-02-23:
Yes, total and year provide in kwh, no decimals provided by inverters. This will typically go into megawatts eventually and watts will mean nothing. 

Using the daily instead on the energy dashboard is the best, I am using the same to keep precision. 

The initial connection will never improve, only way is to use a waveshare device

The fix for the other errors:
https://github.com/Pho3niX90/solis_modbus/releases/tag/3.0.1

commit 202c754d (2025-02-21): closes #143
closed 2025-02-23 by Pho3niX90