Pho3niX90/solis_modbus #145: [Feature]:

indie89, 2025-02-23:
## 🚀 Feature Request
Add MPPT Multi Peak scanning on/Off switch and time span between Scans.

## 🔍 Use Case
Especially the s6 3ph 13-20 needs multiple minutes to scan and set a new maximum. Meanwhile there is no production on either string. It would be great to manually turn on the periodic scan, for it directly performs a scan on switching it on. Afterwards one could directly turn it off again to enforce a single scan (whenever a shade in one's individual situation appears).

## 🛠️ Implementation Ideas (if any)
There is already an option to Turn it on/Off over the App, so it should be feasible over modbus.

Thanks for the great Integration!!

indie89, 2025-02-24:
I'm not bery familiar with the modbus protocoll, but it might be, that these are the correct mappings:  

[image]

(Found Herr: https://usservice.solisinverters.com/support/solutions/articles/73000560403-solis-modbus-protocols-maps-)

Pho3niX90, 2025-02-24:
> I'm not bery familiar with the modbus protocoll, but it might be, that these are the correct mappings:
> 
> [image]
> 
> (Found here: https://usservice.solisinverters.com/support/solutions/articles/73000560403-solis-modbus-protocols-maps-)

These seem to be the string/grid inverter modbus settings. 

For the Scanning, the only setting I could find is the scanning interval, but no way to disable it:

[image]

closed 2025-02-24 by Pho3niX90

Pho3niX90, 2025-02-24:
https://github.com/Pho3niX90/solis_modbus/releases/tag/3.0.2

indie89, 2025-02-25:
@Pho3niX90 , the important setting in order to solve my use case would be the "turn on / off of mppt scanning" and I'm sure there is a way to implement it with modbus. I will ask the (mostly helpful) Solis Support for the correct mapping in order to enable the implementation. Will let you know if I receive information. Thank!

indie89, 2025-03-03:
@Pho3niX90 I received the current Modbus Mappings from Solis as a PDF but I had to sign an NDA. So I can tell that there is (currently) unfortunately indeed no switch to turn on/off mppt scanning. In fact I have to drop my request...

If you haven't yet received the current modbus mapping document (what you probably have), contact the support, sign an NDA an use the info for this integration. I hesitate to just send the doc to you...

Pho3niX90, 2025-03-03:
> @Pho3niX90 I received the current Modbus Mappings from Solis as a PDF but I had to sign an NDA. So I can tell that there is (currently) unfortunately indeed no switch to turn on/off mppt scanning. [...]

Thanks for letting me know. 

Yes, I have received it the same way, but a newer version made its way to me by means of support request

commit e138c9f0 (2025-02-24): feat: mppt scanning interval, fixes #145
feat: more status mappings