Pho3niX90/solis_modbus #146: Updating to 3.0.0 causes time of use switch to become unavailable

miketarg, 2025-02-23:
As above but also noticed that the modbus was connected so it's a bit odd.
## 🔄 To Reproduce

## 🔧 System Information
### 🔹 Home Assistant:
- **Version: 2025.2.4

### 🔹 Integration:
- **Device Model**: S5-EH1P5K-L
- **Modbus Connection Type: Ethernet

The following entities are unavailable 

[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

Please can someone assist?

miketarg, 2025-02-23:
[image]
[image]
[image]
[image]

Logs

Pho3niX90, 2025-02-23:
I would need the actial debug logs, wcreenshots doesn't really say what's gping on in the backend. 

That being said, have you tried the patch that was released earlier today? '3.0.1'

miketarg, 2025-02-24:
> I would need the actial debug logs, wcreenshots doesn't really say what's gping on in the backend.
> 
> That being said, have you tried the patch that was released earlier today? '3.0.1'

Yes unfortunately it had the same effect, how do I give you debug logs?

Pho3niX90, 2025-02-24:
I will post a how to later for the logs, currently on the phone. 

From the screenshots, it seems the inverter is supplying invalid data for those specific registrars.

What firmware is your inverter currently on? A big win would be to request an update from support to enable this functionality for you. But, I will quickly fix the bug tonight, which would mean these items would be disabled when invalid data is received.

miketarg, 2025-02-24:
[image]

Is this it?

miketarg, 2025-02-24:
> Screenshot_20250224-103640.png (view on web)
>
> Is this it?
>
> No this is the app version on your phone.
>
> Easiest is to contact support, and request a firmware update.

Ok thanks will do

miketarg, 2025-02-25:
Hello,

I have now been out on the latest firmware, but unfortunately the issue persists.

[image]

Interestingly enough, I don't know if it makes any difference, but see the below comment made by solis support 

[image]

I have also tried the 3.0.2 version with the same result (2.1.2 works as expected)

Also I note that my model number is missing from the configuration drop-down (not sure if this is why just thought I'd mention it)

I have the following 

[image]

Regards Mike.

Pho3niX90, 2025-02-25:
See if this helps with generating the logs, then I can temporarily disable those registrars for this model 
https://www.home-assistant.io/docs/configuration/troubleshooting/#debug-logs-and-diagnostics

miketarg, 2025-02-25:
Ok some good news (the time of use mode switch is ok now (must have just been Modbus fighting again) 

I have done as requested see the attached

[home-assistant_solis_modbus_2025-02-25T10-19-56.919Z.log](https://github.com/user-attachments/files/18962282/home-assistant_solis_modbus_2025-02-25T10-19-56.919Z.log)

 file

miketarg, 2025-02-25:
> yes those look perfect. if they aren't throwing errors, it can be left
> until support is added, in the meantime they will just be "disabled" until
> the modbus provides information for them.

Ok thanks