Pho3niX90/solis_modbus #147: [Bug]: Household Load energy today is always zero

richvs, 2025-02-24:
The state of sensor.inverter_solis_household_load_today_energy is always zero. The register appears to be read and updated on a reasonable schedule, it just always presents as zero.

I've found an alternative address to read from which reports the correct value. I've used the one from the hybrid protocol definition document here:

https://usservice.solisinverters.com/support/solutions/articles/73000560403-solis-modbus-protocols-maps-

In sensor_data/hybrid_sensors.py
I've exchanged 33586 for 33179 and it now gives me the expected readings.

Inverter is in the UK, a hybrid S5-EH1P6K-L running firmware updated this week (4A 0050). No battery fitted yet. Modbus over WiFi from an S2-WL stick; local access only.

HA2025-1.2, Docker on a Synology NAS.
Integration version 3.0.1, this has been necessary on the last few versions too but I'm only reporting it now.

I'm not sure if this is because I've missed something in the configuration or if anyone else has the same issue. It's configured as an S6-E1HP. 
Trying a different model number didn't seem to make a difference and upon digging it seems that there are only String and Hybrid definition tables.

Thanks very much for such a fantastic project. It's a capable piece of hardware hobbled by poor official software, this really unlocks its potential.

Pho3niX90, 2025-02-24:
33179 seems to be house load + backup load (named `Today Energy Consumption`)
33586 seems to be only house load (named `Household Load Today Energy`)

richvs, 2025-02-24:
Thanks for getting back to me! 
I have nothing connected to the Backup port.

I've just swapped this to read register 33590 (backup only) and I get the value that I was expecting for household consumption. Naturally with only one port connected the total should indeed read the household load which is what I was getting with 33179.

Perhaps there is a peculiar configuration inside my inverter. I wonder if anyone else has the issue. If so I'd love to see a  'Swap household and backup energy consumption' option in the configuration. Of course if it's only me then I can just make the tweak after you make updates.

[image]

How do I buy you a beer/coffee?

Pho3niX90, 2025-02-24:
So it seem the inverter is labelled correctly, according to the documentation

[image]

I do see there is another register 36041, that is labeled `Mapping grid-side`, although it states it's the same as 33586, which is labeled `Household load today energy`. So confusing to say the least 

Having a look at my inverters stats (S6), I have my load completely on the backup connector:

[image]

So wondering if the registrars are swopped between the S6 and S5, but the physical naming/location is the same:

[image]

Pho3niX90, 2025-02-24:
@ctepelea @miketarg @wawyed @danielTinkiel 
Are you seeing the same behavior on your S5?

wawyed, 2025-02-24:
Hey @Pho3niX90 yeah I noticed the same as @0rangutan (different values for consumption ofc)

> [@richvs](https://github.com/richvs) - I have the exact same model S5 inverter.
> 
> Here is what I see for mine (possibly only causing further confusion):
> 
> * Today Energy Consumption = 11.4kWh (looks right for today)
> * Household Load Today Energy = 0kWh
> * Backup Load Total Energy = 0kWh

Pho3niX90, 2025-02-24:
> Hey [@Pho3niX90](https://github.com/Pho3niX90) yeah I noticed the same as [@0rangutan](https://github.com/0rangutan) (different values for consumption ofc)

are your backup and household also 0, but today energy has your  correct value?

Pho3niX90, 2025-02-24:
> Yes, see attached photo:-
> […](#)

image missing :)

0rangutan, 2025-02-24:
> are your backup and household also 0, but today energy has your correct value?

Yes.  Backup and Household both show 0, but today energy has the correct value.

ctepelea, 2025-02-24:
> [@ctepelea](https://github.com/ctepelea) [@miketarg](https://github.com/miketarg) [@wawyed](https://github.com/wawyed) [@danielTinkiel](https://github.com/danielTinkiel) Are you seeing the same behavior on your S5?

Hello, sorry for being late to reply. For me version 3.0, 3.0.1 aren’t working well at all. Am missing half the entities, charge/discharge current aren’t available.

I am sticking to 2.1.2 as I am skiing with the kids at the moment. 🚠

richvs, 2025-02-24:
I've reverted my integration to the latest build after having mucked about with it. I'm now seeing what everyone else sees.

Total load from 33179 appears correct, albeit from my system which doesn't have anything connected to backup.
Zero from backup load total 33590 or household load today 33586.

Sorry for the red herring when I thought I'd swapped it (turns out I probably didn't).

miketarg, 2025-02-24:
> > Yes, see attached photo:-
> > […](#)
> 
> image missing :)

[image]

Whoops

wawyed, 2025-03-02:
I believe those registers don't exist on my inverter (S5-EH1P5K-L). I can't find them in the documentation (33580- 33596)

closed 2025-03-17 by github-actions[bot]