Pho3niX90/solis_modbus #148: Frequent errors in log

RadioactiveMan56, 2025-02-24:
I see frequent errors of the following type in the logs:

2025-02-24 18:09:11.753 ERROR (MainThread) [homeassistant.core] Error running job: <Job listen solis_modbus HassJobType.Callback <bound method SolisTimeEntity.handle_modbus_update of <entity time.solis_rai_solis_grid_time_of_use_charge_end_slot_1=unknown>>>
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/core.py", line 1577, in async_fire_internal
    self._hass.async_run_hass_job(job, event)
    ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/core.py", line 940, in async_run_hass_job
    hassjob.target(*args)
    ~~~~~~~~~~~~~~^^^^^^^
  File "/config/custom_components/solis_modbus/time.py", line 148, in handle_modbus_update
    self._attr_native_value = datetime.time(hour=int(hour), minute=int(minute))
                              ~~~~~~~~~~~~~^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
ValueError: hour must be in 0..23

Integration version 3.0.2
Inverter Solis RAI
Home assistant version
Core
2025.2.5
Supervisor
2025.02.1
Operating System
14.2
Frontend
20250221.0

Hardware
Raspberry Pi 3
rpi3-64

[home-assistant_solis_modbus_2025-02-24T17-14-39.510Z.log](https://github.com/user-attachments/files/18947490/home-assistant_solis_modbus_2025-02-24T17-14-39.510Z.log)

RadioactiveMan56, 2025-02-25:
Hi,

I installed 3.0.4, still seeing a few similar errors. See the debug log

[home-assistant_solis_modbus_2025-02-25T14-44-17.752Z.log](https://github.com/user-attachments/files/18966607/home-assistant_solis_modbus_2025-02-25T14-44-17.752Z.log)

Pho3niX90, 2025-02-26:
please see if this resolves the issue, there were logic issues in 3.0.4
https://github.com/Pho3niX90/solis_modbus/releases/tag/3.0.5

commit 5e133c6d (2025-02-25): fixes #148