Pho3niX90/solis_modbus #149: [Enhancement]: Switch to enable only read-only entities eg Sensors only. Power, temperature, SoC, voltage, current

veletron2, 2025-02-24:
## 🎯 Enhancement Request
Nice to see the recent work on this allowing exclusion of certain sensors for folks who don't have solar or don't have a battery.

Can this be taken further please in a future version categorising entities into sensors, switches and settings. In my personal use-case I am on a time-of-use tariff and only ever need access to the basic sensor values

## 🔍 Use Case
Read-only use where only basic sensor access is required: current, temperature, power, voltage, soc, grid in and grid out. Such a switch removes the possibility of accidentally changing a setting on an inrterter that shall just manage its-self.

## ✅ Expected Improvement
A switch when adding a new device alongside the other switches eg: 'Read Only entities'

## 🛠️ Implementation Suggestions
If you have an idea on how to improve it, add details.

## ➕ Additional Context
Related issues or documentation.

Pho3niX90, 2025-02-24:
So, if I understand correctly. You would prefer option to disable inputs and switches? 

I will have a look into categorization for this. Just for reference, which inverter/inverters are you using?

veletron2, 2025-02-24:
Thanks for response

Yes, but understand you cant please everyone!! Basically the sensors needed to populate the energy flow card and the energy dashboard. Could maybe term it 'minimalist' or something (!)

Also, I don't think there is much point in exposing entities for eg configured time of use or the current operation mode of the inverter if the switch/config entity is not exposed. So basically power, voltage, current, temperature

My inverter is an S5-EH1P5K-L but I am using the S6-EH1P in the config for the integration in HA.

[image]

Pho3niX90, 2025-02-24:
This does sound like a good suggestion. There are currently some advanced seitches exposed that could potentially do more harm than good if not used wisely. 

I do like your idea of categorizing them, and drill down from there. I think most people do have set and forget settings, and perhaps just want the statistics.

RadioactiveMan56, 2025-02-24:
I support the idea of a monitoring only version. I have an RAI which has a very basic HMI display, hiding the time of use settings seems a bit harsh. It takes a lot of button pushing to inspect them from the HMI.

veletron2, 2025-02-25:
Hi

The more I think about this, the more I wonder if its actually better handled with some kind of entity config/supression file that directs which entities should be created when the device is loaded at startup (or when reloaded)?

Something like

Solid_Mod/1/config.yaml for device 1
Solid_Mod/2/config.yaml for device 2 etc

If file does not exist, everything created, if file exists, parse file anything not listed in file is created by default, within file its possible to supress the creation of entities of certain types (sensors, switches, configurable fields) for certain inverter behaviour eg:

General: (covers basic inverter mode switches, sensors, inputs (basic power, current, voltage, temperature)
  - SENS: True
  - SW: False
  - CONFIG: False

Battery: (battery power, voltage, current, SoC)
  - SENS: True
  - SW: False
  - CONFIG: False

Backup: (related to the backup port)
  - SENS: True
  - SW: False
  - CONFIG: False

Solar: (Power, current, voltage)
  - SENS: False
  - SW: False
  - CONFIG: False

Generator:
  - SENS: False
  - SW: False
  - CONFIG: False

ToU: (Time of Use related sensors)
  - SENS: False
  - SW: False
  - CONFIG: False

etc

'General' would cover all entities that are either unrelated to the other categories, or which have cross-over between categories?

Maybe the above could be split slightly more into two SENSOR classes 'Sensors' and 'HistoricalSensors' - this because there is generally no need for inverter-provided historical sensors when using HA cause all the data is stored in HA at time of use.

You'll know way way more than me about the various entity classes that get exposed, the above should be generic to work with any inverter while not over-complicating existing setup, and not breaking existing installs cause the config.yaml wont exist for them and thus everything would be enabled. For folks that require even more granularity, the existing HA mech to disable certain entities can be used in combo with the above.

Sadly, I am not a python programmer currently, so am of zero help re the coding for the above. Its on my list for something to learn if I ever find the time!

Pho3niX90, 2025-02-25:
Some good points. 

I just need to figure out how to split the registrars. For the basic to allow read/write, it's easy as the holding and reading registrars are never read in the same block of registrars. 

But when it comes to things like solar/generator/backup, they are typically split between blocks of registrars. 

The whole idea of the blocks is to read 50 registrars with one request, eliminating the inter frame delay, and speeding up the process.  having them more granular in configuration would mean to potentially splitting the ~26 groups into perhaps 30-40 groups. 

I just need to make sure as I am labeling them, how I can move them around between groups.

veletron2, 2025-02-25:
Something else I noticed is the occasional dodgy sensor value. My HA is on the backup port so it keeps running if I kill the mains, I noticed a spike in the Solis temperature reading to over  800 degC when I killed the mains. This must just be nonsense coming from the Solis, but maybe some kind of filter on applicable sensors to eliminate single readings that are way outside the bounds of normality? Obv not appropriate for power/current sensors tho!!

Pho3niX90, 2025-02-25:
> Something else I noticed is the occasional dodgy sensor value. My HA is on the backup port so it keeps running if I kill the mains, I noticed a spike in the Solis temperature reading to over 800 [...]

thats intresting, grid frequency and temprature are in the same block. If you can replicate this with the debug logging on, I can implement something to take care of it. Just need to know how the modbus responds in that scenario

Dilbertdog, 2025-05-22:
Have just had an issue with my master and slave solis inverters, where they were added to a new version of the solis modbus software, and the settings on the inverters corrupted. This could be a Solis software issue, but also could be that some of the advanced switches accidently got modified.
Having the ability to set this as a "read only" for all the data, and also when making a change, bring up an "are you sure" message to confirm the change would be really useful. Alongside the ability to download the entities and their status, so that if anything happened you could see what had changed from the orginal set up.

closed 2026-10-11 by Pho3niX90